esp8266/Arduino #212: OSX release esptool broken or only part. functional

madias123, 2015-05-09:
Hello,
maybe it's a known problem, but I didn't find it as standalone topic, so that OSX user won't get mad as I getting things to work:
The ESPtool in the OSX version isn't working.
There is no "ESPtool" under "programmers", upload seemed to be ok, some silly blink/serial tests are working, but when using every WIFI stuff, the ESP hangs.
Tried the latest OSX download release.
Topics in the forum:
My topic:
http://www.esp8266.com/viewtopic.php?f=26&t=2988&p=17021#p17021
same here:
http://www.esp8266.com/viewtopic.php?f=26&t=2423
Maybe there is a possibility, until there is no fix, to sign the OSX release as broken, because it's really hard to find out the problem (without resarching, switching to another virtual OS to see, that your hardware is running fine.)

gerardwr, 2015-05-09:
I have no problems under OSX.

It looks like your power source is adequate for programming the ESP and running non-Wifi sketches, but seems underpowered for Wifi-sketches.

You need a good power supply, powering from a USB-SERIAL adapter is mostly inadequate.

madias123, 2015-05-09:
nope.I use an self build external power supply (far beyond 1A) plus tandal caps, elkos and tried even 18650 lithium ion batteries and as I also told, I'm not the only one with exactly this problems.
As I told in my thread it does work with the same laptop with the same setup and connection if I use the virtual windows machine, switching back without HW changes it doesn't. Played the one than more time forward and backward even with different USB/TTL/FTDI converters. So there must a problem with config/SW
Do you have the ESP in the "programmer" list of the IDE? What's your version number of OSX? (I know that from STM32Duino, where Roger and I worked on different OS versions and had big troubles in finding a compiled uploader that works on all versions).

rei-vilo, 2015-05-10:
On Mac OS X, `esptool` fails to upload to the NodeMCU board.

Using `esptool.py` solves the issue, as per #78 **SOLVED — Add Support for nodemcu — Use esptool.py**

Including `esptool.py` in the package would be nice.

gerardwr, 2015-05-10:
@madias123 

- on Mac OSX 10.0.3
- no "esptool" visible in "programmer" (in 1.6.4), but uploads fine
- The precompiled OSX IDE is seriously outdated. It worked OK initially but the latest version cause unreliable uploads due to a bug in esptool. I then switched to the latest updated.

Both the "sandeepmistry github + Arduino 1.6.3" and the "Arduino 1.6.4 + Boards Manager ESP8266" route works OK for me.

madias123, 2015-05-10:
Thank you to all!
sandeepmistry github (release 0.0.5)  on 1.6.3 fixed it!
So maybe the precompiled OSX can be set offline or recompiled?
The ESP is not an "easy" device and with a "partially" working IDE the frustrating level is very high...

regards
Matthias

igrr, 2015-05-19:
The latest release has esptool updated to 0.4.4, which should work better with CH340 usb-ttl chips.

rei-vilo, 2015-05-19:
Thank you! I'll test it and let you know.

sticilface, 2015-05-19:
Just tested this and omg!  it works.  this is the first time that the esptool has worked for me on mac.  Well done @igrr!  You're work on this is going to become the stuff of legend!  Can't wait for the SPIFFS.. which is now teased in the status bar.... all 64K of it.  

Is there an easy way to get the esptool to send a break signal?  I have some ATTINY85s doing zero wire programming... its easy to mod the python version, i thought that esptool did it automatically after nerdralph made a request.  Likely wrong though.

PS.. I can upload at 460800 now!  super fast. well done

zenmanenergy, 2015-05-19:
I was just writing up an issue about this. sticilface, you got it to work? I don't see esptool in the programmer menu. What programmer should I use? I'm getting this error:

esptool v0.4.4 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 460800
	setting port from /dev/tty.usbserial to /dev/cu.wch ch341 USB=>RS232 1410
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/rb/svhs1qg52vzfnvl9tkp7_0jh0000gn/T/build2317840705052365870.tmp/test1.cpp_00000.bin success
opening port /dev/cu.wch ch341 USB=>RS232 1410 at 460800
	tcgetattr
warning: serialport_set_baudrate: baud rate 460800 may not work
	tcsetattr
error: setattr stage 1 failed
error: espcomm_open failed

sticilface, 2015-05-19:
by default you do not need to select a programmer.  esptool is selected by default.  I did nothing, just connected my NODE-MCU devboard, and it worked.  connected my FTDI board and ESP and that worked as well.  None of these have ever worked before.  

The baud rate is going to be highly variable.  from what i;ve seen the ESPs are quite flexible... i'be got a test sketch and using one of my FTDI dongles got a baud of 2,000,000 to work....  but that is with this one adapter that seems to be pretty good.  can't flash at that speed though, at least not with the python esptool.  does yours work at sensible bauds of 115200?

zenmanenergy, 2015-05-19:
I don't have esptool in the list on my mac osx version. Did I not install something correctly?

On my Windows v1.6.1 it has esptool listed.

zenmanenergy, 2015-05-19:
I didn't fully answer your question... No. Mine doesn't work at any baud rate on OSX. Here's more debugging information if that helps anyone. My settings:

1) esp-01 on Mac OSX Yosemite 10.10.3
2) gpio0 is grounded
3) trying to upload the wifiscan.ino included example
4) IDE Board: Generic ESP8266 Module, I tried the other two
5) 80 Mhz (i tried 160 also, same error)
6) flash speed 512K
7) uploaded speed 115200 (I have tried them all)
8) port /dev/cu.wch ch34 USB =>RS232 1410
9) Programmer... esptool is NOT listed, this seems like the problem I've tried all of the AVR ISP through Arduino as ISP

Here's the output:

esptool v0.4.4 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.wch ch341 USB=>RS232 1410
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	stat /var/folders/rb/svhs1qg52vzfnvl9tkp7_0jh0000gn/T/build2317840705052365870.tmp/WiFiScan.cpp_00000.bin success
opening port /dev/cu.wch ch341 USB=>RS232 1410 at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed

madias123, 2015-05-19:
all I can say is that there is no "ESP-tool" in the programmer list under OSX. But as you can see on your output the esptool is under use.
I didn't tried the new repo, but the sandeepmistry "addon version" without problems. What usb/ttl dongle do you use? I tried it with FTDI and some no name dongles with success.

zenmanenergy, 2015-05-19:
I'm using a dongle that I bought ages ago. It says "AI" on the back, other than that I have no idea. I have the ftdi version 2.3 driver installed. Again, it does work on this exact same machine when I'm running Windows via vmware. So the hardware itself seems to work. Something about the mac side is the issue.

It's certainly possible something isn't installed on my mac that should be there. Or something is broken. I just don't know how to go about debugging that.

madias123, 2015-05-19:
Hm. I would at first test the dongle under OSX. Just try with an arduino board (or something else or just connect rx with tx on your dongle) a simple serial call/response connection with all baud rates to the dongle via serial monitor.

zenmanenergy, 2015-05-19:
Whoa. I would love to do that, but I'm not sure how. Can you send me a url that has a explanation and/or some google search terms to find out how to do that test?

madias123, 2015-05-19:
If I'm not completely wrong just wire the TX and rx on your dongle together, fire up serial monitor and type anything and you should get back everything you typed as loop back

zenmanenergy, 2015-05-19:
That just blew my mind. It makes 100% perfect sense. ha

I just tested it... That much works! Any idea of what to try next?

zenmanenergy, 2015-05-19:
I just took a step back and decided to see if I could get ./esptool.py to work on my mac... no luck. I copied the path of the compiled .bin file from the arduino IDE and pasted it into a command line. LIke this:

sudo ./esptool.py --port /dev/cu.wch\ ch341\ USB\=\>RS232\ 1410 write_flash 0x00000 /var/folders/rb/svhs1qg52vzfnvl9tkp7_0jh0000gn/T/build3531805817819405198.tmp/WiFiScan.cpp_00000.bin

It gives me this error:

Connecting...
Erasing flash...
Traceback (most recent call last):
  File "./esptool.py", line 533, in <module>
    esp.flash_begin(blocks*esp.ESP_FLASH_BLOCK, address)
  File "./esptool.py", line 196, in flash_begin
    struct.pack('<IIII', size, num_blocks, ESPROM.ESP_FLASH_BLOCK, offset))[1] != "\0\0":
  File "./esptool.py", line 110, in command
    raise Exception('Invalid response')
Exception: Invalid response

madias123, 2015-05-19:
Ok, you tried out all baud rates? I've several USB dongles and some work better on parallels windows on my macbook and some are working better native. ESP is really picky with that. Ok, I tried the new version  of esptool v0.4.4 with success, but I'm still under https://github.com/sandeepmistry/esp8266-Arduino
Maybe you can give this one a try?

madias123, 2015-05-19:
You can also try UECIDE the new beta release has the ESP as plugin, tried it out just yet and runs out of the box!

zenmanenergy, 2015-05-19:
So far, no luck with esptool v0.4.4 command line version.

./esptool -cp /dev/cu.wch\ ch341\ USB\=\>RS232\ 1410 -cf helloServer_00000.bin -ca 0x10000 -cf helloServer_10000.bin 

Uploading 33280 bytes from helloServer_00000.bin to flash at 0x00000000
warning: espcomm_send_command: wrong direction/command: 0x01 0x08, expected 0x01 0x02
warning: espcomm_send_command(FLASH_DOWNLOAD_BEGIN) failed

zenmanenergy, 2015-05-19:
I downloaded the UECIDE alpha release, but I don't see anything in there about the ESP8266.

madias123, 2015-05-20:
all boards are installed via Plugin manager in UECIDE

igrr, 2015-05-20:
@zenmanenergy do you have a logic analyzer or a scope? If you could capture the UART interaction during upload, that would really help. Also which usb-ttl adapter are you using?

madias123, 2015-05-20:
@igrr : He wrote "I'm using a dongle that I bought ages ago. It says "AI" on the back," I suggested a loopback test to him under OSX, which seems to run fine. So the best idea would to install UECIDE with ESP as plugin - this runs out of the box with 1-2 mouse clicks. But I fear that he has an driver problem with the dongle under osx.  "/dev/cu.wch ch341" looks for me it's a CH340/CH341 not FTDI.
So zenman: Maybe you should try to update your usb-dongle OSX driver with this version:
http://www.wch.cn/download/CH341SER_MAC_ZIP.html
and please read this for OSX 10.10: 
http://javacolors.blogspot.co.at/2014/08/dccduino-usb-drivers-ch340-ch341-chipset.html

Juppit, 2015-05-20:
From Matt Jenkins 3 day bevor: I have created a UECIDE core, compiler and board packages from the ESP8266 repository. It is available for installation in the current beta version of UECIDE.

igrr, 2015-05-20:
Oh, sorry, missed that.
I thought the CH340 issue was sorted out in this pull request: https://github.com/igrr/esptool-ck/pull/6
But i don't have the hardware myself to test/verify... one NodeMCU board is on its way, but it won't arrive for another couple of weeks. I'd really like to sort out the upload issues, but i need a logic analyzer capture or al least an strace log to understand what's wrong.

zenmanenergy, 2015-05-20:
I will try the UECIDE today, where do I download the plugin for esp8266?

Can any of you that has gotten this to work with a Mac show me which ftdi you are using? I would love to just order it and be done with this problem

madias123, 2015-05-20:
UECIDE -> plugin manager -> boards -> ESPxxx
FTDI: I use the red ones with male headers from aliexpress (about 3USD)
http://www.aliexpress.com/w/wholesale-ftdi.html?initiative_id=SB_20150520045017&site=glo&groupsort=1&SortType=total_tranpro_desc&shipCountry=at&SearchText=ftdi

sticilface, 2015-05-20:
Those are exactly the ones i am using, with OS X and having success.  but only with the latest release, installed via boards manager.  I've got a serial test sketch to work with a baud rate of 2,000,000 with these things.  uploading only the  one around 400K works, not 921,000.

ErikAndren, 2015-05-23:
Using mac os x and ESP-01.
I need to power cycle after each programming, otherwise I'm unable to connect.

zenmanenergy, 2015-05-26:
PERFECTO! Thanks guys. I bought one of those ftdi boards last week and it just arrived yesterday. It works great. Maybe on a wiki page we can add a link to the ftdi boards that work.

madias123, 2015-05-26:
zenman: congrats to your success!
but it's a lottery what FTDI device you get from most sellers (I only know that the one with the female headers is problematic) even they won't send you the one as shown on the picture :/
...and don't forget: Never ever install the newest official FTDI drivers for that china device!

closed 2015-10-28 by igrr