esp8266/Arduino #213: ESP object

Pablotis, 2015-05-10:
How can I use the ESP object for reset, sleep, etc. It sais:
sketch_may10a.ino:11:2: error: 'ESP' was not declared in this scope

Thanks

bprobbins, 2015-05-10:
I'm wondering the same thing. For now I'm using 
system_deep_sleep_set_option(1); //not yet sure what the constants translate to
system_deep_sleep(60000000); //sleep time in usecs. 60000000 = 60 secs.
and
system_restart(); 
for the corresponding ESP functions. Not sure if they are identical

Makuna, 2015-05-10:
This works for me.
In my NeoPixel library (https://github.com/Makuna/NeoPixelBus) you can see that I use the new ESP.getCycleCount().

I have found that in general it is required to include Arduino.h; which is not required for AVR simple apps.

igrr, 2015-05-12:
@Pablotis it's in the latest trunk and you are likely using the older release. Please hang on for the new release this week.

robertgregor, 2015-05-12:
Hello, would it be possible to add support for the wake pin interrupt? I see in the ESP8266 SDK include/gpio.h:
gpio_pin_wakeup_enable(GPIO_ID_PIN(4), GPIO_PIN_INTR_HILEVEL);

For me it looks like, tHAt it is possible to enable the pin interrupt and wake up from deep sleep with this.

igrr, 2015-05-12:
Not from deep sleep. Deep sleep is essentially a "power down the CPU, keep RTC running" mode. This is why wakeup is bound to GPIO16 — this is the only pin that is controlled from RTC domain.

robertgregor, 2015-05-13:
I see ok, thanks.

closed 2015-05-19 by igrr

koravadi, 2016-02-28:
In my application I want to make the module to go to light sleep till a MQTT packet received or IO pin become high. Is any one know how I can do this?? I don't want to keep pooling the IO since it will consume power..