esp8266/Arduino #215: timer interrupt

yakirmalka, 2015-05-10:
is ther any way to start timer interrupt?
in lua its "tmr.alarm(1, 200, 1,function()"

gerardwr, 2015-05-10:
Ticker!

gerardwr, 2015-05-10:
Ticker!?

yakirmalka, 2015-05-10:
10x alot.

closed 2015-05-12 by igrr

madsci1016, 2015-08-03:
Anything with uS precision? Looks like ticker is only mS.

igrr, 2015-08-03:
Yes we have two timers, please see Servo library for an example:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/Servo/src/esp8266/ServoTimers.h

No other documentation right now, unfortunately.

mralexgray, 2016-01-23:
@igrr link is broken..  really need clarification on how to use timers with this platform.  is it possible to just use `Timer1`, `Timer2`, etc?

GeorgeHahn, 2016-09-20:
@mralexgray https://github.com/esp8266/Arduino/blob/master/libraries/Servo/src/esp8266/ServoTimers.h