esp8266/Arduino #216: Support for hardware PWM?

Janjanus, 2015-05-10:
Hello,

Is there support for hardware PWM on the ESP8266? Since the README.md tell there is only software PWM available but this is not suitable for my application. 

The ESP8266 is capable of hardware PWM right?

igrr, 2015-05-12:
If it does have hardware PWM, it is not documented anywhere. So for now software PWM is the only option.

cherowley, 2015-05-13:
Hi iggr,

Just wondering how accurate will your software PWM be? 
Also, will it support 4 pwm at the same time - thinking rgbw led strips you see :)

Cheers!

Toshik, 2015-05-13:
I'd suppose to use special module for PWM based on PCA9685 chip.
It is nice and driven by I2C

http://www.ebay.com/itm/311188037855

pRoFlT, 2015-05-19:
Does software PWM work? when i try analogWrite(pin#,255) i don't get anything on that pin? I've tried several DIO and several levels from 0-255 with nothing. I'm using the nodeMCU V0.9 ESP8266 board. I've also tried on the Huzzah ESP8266 from adafruit. Same results. Am i missing something?
 
code is:
in setup
pinMode(4,OUTPUT);

in loop
analogWrite(4,127);

closed 2015-07-30 by igrr

mralexgray, 2016-01-02:
Why was this closed, with no resolution?

9H1LO, 2016-02-02:
up to 3 ch soft pwm is fine, also trying to do rgbw and get a flicker (when there is network activity) when introducing 4th channel i have tried pca9685 with i2c and it works great but overkill to use a 16ch chip just for 1 rgbw

seidler2547, 2018-06-24:
Sorry to ressurect this old topic. I don't  know if it is a recent change, but the [datasheet](https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf) clearly states (chapter 4.7) that the ESP8266 *has* 4 hardware PWMs. Any chance to implement support for that?

devyte, 2018-06-24:
@seidler2547 why do you think it has hw pwm? Section 4.7 clearly says that pwm is done by timer interrupt.