esp8266/Arduino #217: Add a Versioning File for ESP8266 Plug-In and Libraries

rei-vilo, 2015-05-10:
I compiled Arduino for ESP8266 based on the source code, and noticed many changes compared to the latest available binary [1.6.1-esp8266-1](https://github.com/esp8266/Arduino/releases/tag/1.6.1-esp8266-1).

It would be nice to mention the version of the ESP8266 plug-in and libraries in a text file, for example, `esp8266.txt` under the `/Applications/IDE/Arduino.app/Contents/Java/lib` folder. This is how Teensy manages the Teensyduino plug-in.

rei-vilo, 2015-05-10:
There is a file at `Espressif.app/Contents/Java/hardware/tools/esp8266/sdk/version` with the version of the SDK, but this version differs from the version of the plug-in and libraries.

Links2004, 2015-11-06:
we have a change log here:
https://github.com/esp8266/Arduino/blob/master/doc/changes.md

and SDK version here:
https://github.com/esp8266/Arduino/blob/master/tools/sdk/version

and SDK changes here:
https://github.com/esp8266/Arduino/blob/master/tools/sdk/changelog.txt

for more detail check the git commits

closed 2015-11-06 by Links2004