esp8266/Arduino #218: UDP sendto is still broken

jwatte, 2015-05-10:
The solution for udp beginPacket() makes the network stack filter out any packets that arrive from other sources while a packet is "begun."
It also does unnecessary amounts of work within the connect() and disconnect() functions.
Better is to remember the IP and port from the beginPacket() function, and use send(addr, port).
I will attempt to provide a patch, if I can get the raw clone to build/run.

igrr, 2015-05-10:
That's already done, will push the updated version next week along with
other network related fixes.

jwatte, 2015-05-10:
If by "that's already done" you mean the current HEAD on github, it has the bug that I already create. Using connect() temporarily in beginPacket() is not correct.

igrr, 2015-05-11:
Done in my local working copy, I just haven't pushed the changes to github
yet. Thanks for the pull though.

igrr, 2015-05-12:
fixed in ae2bc12

closed 2015-05-12 by igrr