esp8266/Arduino #221: Missing WiFi(ESP8266WiFi library)

asm7100, 2015-05-11:
The lib WiFi(ESP8266WiFi library) is missing????

frasch, 2015-05-11:
Please verify you have choosen "Generic ESP8266" board when building.
If selecting a 'normal' arduino board (e.g. yun, uno, etc.) there is no ESP8266Wifi...

asm7100, 2015-05-11:
Hi Yes it is set to "Generic ESP8266"
I get this error.

C:\Udvikling\ESP8266\Arduino ESP\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:58:9: note: int ESP8266WiFiClass::begin(const char*, const char*)
     int begin(const char* ssid, const char *passphrase);
         ^
C:\Udvikling\ESP8266\Arduino ESP\hardware\esp8266com\esp8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:58:9: note:   candidate expects 2 arguments, 3 provided
Fejl i kompilering.

And this is the program.

#include <ESP8266WiFi.h>
#include <Base64.h>
//AP definitions
#define AP_SSID "Aaron"
#define AP_PASSWORD "xxxxxxx","yyyyyyyyyyy"

// EasyIoT server definitions
#define EIOT_USERNAME    "admin"
#define EIOT_PASSWORD    "ggggggg"
#define EIOT_IP_ADDRESS  "192.168.0.194"
#define EIOT_PORT        80
#define EIOT_NODE        "N1S0"

#define INPUT_PIN         5
#define USER_PWD_LEN      40

char unameenc[USER_PWD_LEN];
bool oldInputState;

void setup() {
  Serial.begin(115200);
  pinMode(INPUT_PIN, INPUT);
  
  wifiConnect();
    
  char uname[USER_PWD_LEN];
  String str = String(EIOT_USERNAME)+":"+String(EIOT_PASSWORD);  
  str.toCharArray(uname, USER_PWD_LEN); 
  memset(unameenc,0,sizeof(unameenc));
  base64_encode(unameenc, uname, strlen(uname));
  
  oldInputState = !digitalRead(INPUT_PIN);
}

void loop() {
  int inputState = digitalRead(INPUT_PIN);;  
  
  if (inputState != oldInputState)
  {
    sendInputState(inputState);
    oldInputState = inputState;
  }
}

void wifiConnect()
{
    Serial.print("Connecting to AP");
    WiFi.begin(AP_SSID, AP_PASSWORD);
    while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.print(".");
  }
  
  Serial.println("");
  Serial.println("WiFi connected");  
}

void sendInputState(bool inputState)
{  
   WiFiClient client;
   
   while(!client.connect(EIOT_IP_ADDRESS, EIOT_PORT)) {
    Serial.println("connection failed");
    wifiConnect(); 
  }
 
  String url = "";
  String command = "";
  
  if (inputState)
    command = "ControlOn";
  else
    command = "ControlOff";
  
  url = "/Api/EasyIoT/Control/Module/Virtual/"+ String(EIOT_NODE) + "/"+command; // generate EasIoT server node URL

  Serial.print("POST data to URL: ");
  Serial.println(url);
  
  client.print(String("POST ") + url + " HTTP/1.1\r\n" +
               "Host: " + String(EIOT_IP_ADDRESS) + "\r\n" + 
               "Connection: close\r\n" + 
               "Authorization: Basic " + unameenc + " \r\n" + 
               "Content-Length: 0\r\n" + 
               "\r\n");

  delay(100);
    while(client.available()){
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }
  
  Serial.println();
  Serial.println("Connection closed");
}

frasch, 2015-05-11:
#define AP_PASSWORD "xxxxxxx","yyyyyyyyyyy"
is wrong. Should be:
#define AP_PASSWORD "mypassword"

asm7100, 2015-05-11:
Thanks :-) Now it compiles :-)

igrr, 2015-05-12:
@asm7100 do you think we can close this issue?

asm7100, 2015-05-12:
Yes thanks

closed 2015-05-12 by igrr