esp8266/Arduino #223: Launch4j - An error occurred while starting the application

DavidPilling, 2015-05-11:
The above is what I get running the arduino.exe from the latest alpha version (2015-05-11). 

Running the debug version I get:

Exception in thread "main" java.lang.UnsatisfiedLinkError: jnidispatch (/com/sun
/jna/win32-amd64/jnidispatch.dll) not found in resource path

This is on Windows 8, 64 bit.

You're doing a good job. Hope this helps.

frasch, 2015-05-11:
My IDE from alpha or private build is not starting up also.
If I replace the Java directory at "Arduino 1.6.1\Java" with Java-director from release of March, 27 (https://github.com/esp8266/Arduino/releases) everything is working fine.
My machine is Win 7, 64 bit with jdk1.8.0_45 (64bit)

DavidPilling, 2015-05-13:
That fixes it for me as well - thanks.

closed 2015-08-05 by igrr