esp8266/Arduino #224: ESP8266WebServer: adding additional headers

holgerlembke, 2015-05-11:
With current implementation, how do I add additional header lines when answering a request?

I found no solution, so I made some changes. I dislike my solution, I would prefer something with overloading. But that might break all the existing solution/samples/etc.

holgerlembke, 2015-05-11:
in ESP8266WebServer.h

  // send response to the client
  // code - HTTP response code, can be 200 or 404
  // content_type - HTTP content type, like "text/plain" or "image/png"
  // content - actual content body
  void send(int code, const char* content_type = NULL, String content = String(""));
  void sendh(int code, const char* content_type = NULL, String content = String(""), String headerlines = String(""));

and in ESP8266WebServer.cpp

void ESP8266WebServer::sendh(int code, const char* content_type, String content, String headerlines) {
  String response = "HTTP/1.1 ";
  response += String(code);
  response += " ";
  response += _responseCodeToString(code);
  response += "\r\n";

  if (!content_type)
    content_type = "text/html";
  _appendHeader(response, "Content-Type", content_type);
  
  // weitere kopfzeilen
  response += headerlines;
    
  response += "\r\n";
  response += content;
  _currentClient.print(response);
}

igrr, 2015-05-14:
8c1a40b adds a ```server.sendHeader(key, value)``` function which may be used before calling ```server.send```.

holgerlembke, 2015-05-14:
Nice. Looks fine to me.

Next thing I'm not entirely sure if it is a good solution:

I like the on(url,function)-idea. 

On the other hand: i have a precompiled binary blob of filenames+html with an index. So I don't use the on-method (because that would result in a double index, one in the on-list, one in my blob) but I go through the list in the "onNotFound"-event.

It works and I'm fine with it.

But it doesn't really "looks as designed" to me.

Just to vent my feelings. :-)

igrr, 2015-05-14:
Right now SD card webserver and Spiffs webserver samples also use
onNotFound handler. Actually I want to add a stack of handlers much in the
same way as it's done in node.js — that would allow for middleware
libraries that plug easily into the webserver.

So thanks for letting me know that you also need this :)

holgerlembke, 2015-05-14:
My current concern is that all the html pages/strings go into heap space. and as far as I can see, that is a major limit currently.  I run at about 25k free...

On the other hand, flash rom seems to be more than enough, new models coming with even more... that's why I go for the blob solution. So if I find some time, I will dig into some ideas about saving html pages into flash via web-server-upload.

igrr, 2015-05-14:
We have SPI flash filesystem working already, and a tool which will pack
your files and upload them along with the sketch. Should be completed
shortly.

holgerlembke, 2015-05-14:
I have seen the work. Hopefully it does not end in something tooo difficult. Fiddling around with make files and such. I like my "upload with web server"-idea... :-)

Meanwhile I file issues.... https://github.com/esp8266/Arduino/issues/240

closed 2015-05-23 by igrr

martin072, 2018-01-24:
Hi,

I noticed this issue, and was wondering if it will be implemented? I need to pass some extra headers with the  webServer.send command, but I noticed adding custom headers is not yet implemented?