esp8266/Arduino #225: system_get_vdd33 or ESP.getVcc() always 65535

tontito, 2015-05-11:
system_get_vdd33 always return 65535.

void setup() 
{
  Serial.begin(115200);
  Volt = system_get_vdd33();
}

igrr, 2015-05-12:
Perhaps your TOUT pin is connected?
See https://github.com/nekromant/esp8266-frankenstein/issues/66#issuecomment-86937980.

tontito, 2015-05-13:
I have install nodemcu (SDK 0.9.5) and send command 
=node.readvdd33
3614 return
Esp8266 for Arduino (SDK 1.0.1)
system_get_vdd33() return 65535

maybe sdk problem

robertgregor, 2015-05-13:
Hello, I have ESP-07 module, where the TOUT is labeled as ADC pin. I don't have it connected, but I am as well reading 2^16. I have SDK 1.0.1

tontito, 2015-05-13:
I tried esp-07 and esp-12 always return 65535.

esp-07 and esp-12 nodemcu return 3614 or 3616

creationix, 2015-05-13:
What include is required to `get system_get_vdd33()`?

tontito, 2015-05-14:
extern "C" {
  #include "user_interface.h"
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println(analogRead(0));
  Serial.println(system_get_vdd33());

}

return serial monitor: 
737
65535

ADC --> VCC return serial monitor:
1024
65535

igrr, 2015-05-14:
ADC has 0 – 1.0 V range, so don't connect it to VCC...

tontito, 2015-05-14:
I thought so TOUT = ADC (for ESP-07) i tried return VCC  maybe.

Sorry my english very bad. google translate :)

creationix, 2015-05-14:
@tontito Hmm, adding user_interface.h doesn't help.  I'm in the main .ino file if that helps any.

I can see the definition is in that file.

```c
extern "C" {
#include "user_interface.h"
}

void setup() {
 Serial.begin(115200);
}

void loop() {
  Serial.println(analogRead(0));
  Serial.println(system_get_vdd33());
  delay(1000);
}
```

And the output I get is:

```
Arduino: 1.6.1 (Linux), Board: "Generic ESP8266 board"

Build options changed, rebuilding all

/home/tim/Downloads/arduino-1.6.1/hardware/tools/esp8266/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -I/home/tim/Downloads/arduino-1.6.1/hardware/tools/esp8266/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10601 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/home/tim/Downloads/arduino-1.6.1/hardware/esp8266com/esp8266/cores/esp8266 -I/home/tim/Downloads/arduino-1.6.1/hardware/esp8266com/esp8266/variants/esp01 /tmp/build7372776056876877767.tmp/scanner.cpp -o /tmp/build7372776056876877767.tmp/scanner.cpp.o 
scanner.ino: In function 'void loop()':
scanner.ino:11:35: error: 'system_get_vdd33' was not declared in this scope
Error compiling.
```

creationix, 2015-05-14:
Also if I manually define it by inserting the header line:

```c
uint16 system_get_vdd33(void);
```

It compiles, but fails in linking with:

```
/tmp/build7372776056876877767.tmp/scanner.cpp.o: In function `setup':
scanner.cpp:(.text+0x20): undefined reference to `system_get_vdd33()'
scanner.cpp:(.text+0x3c): undefined reference to `system_get_vdd33()'
collect2: error: ld returned 1 exit status
Error compiling.
```

chadouming, 2015-05-14:
In your extern c declaration, add :

uint16 readvdd33(void);

And use that instead. It works fine here.

creationix, 2015-05-14:
@chaduoming Thanks, but that's not working either.  (Also didn't you mean `uint16_t` for the return type?)

If I put it in the extern block, the compiler complains it's not defined.

If I put it outside, the linker complains:

```
scanner.cpp:(.text+0x160): undefined reference to `readvdd33()'
```

chadouming, 2015-05-14:
see that : https://github.com/chadouming/hardware/commit/06e0f00fda03101dab036f34252f1b1fc5bc4e96

creationix, 2015-05-14:
@chadouming nice, how do I use your fork?  Do I just dump it in the hardware folder of arduino?

I also have local modifications to use esptool.py with custom `-fm dio` in the flash command (Using the black nodemcu devkit).

tontito, 2015-05-14:
nice job :)
insert 

ESP.cpp
extern "C" uint16_t readvdd33(void);
uint16_t EspClass::getVdd33(void)
{
    return readvdd33();
}

and ESP.h
uint16_t getVdd33(void);

Volt = ESP.getVdd33();
return 3614 

:)) very very nice.

for me it's a beautiful day

insert SpifFs file system work and readvdd33 work :)

only one for me OTA (air update) :))))))

tontito, 2015-05-15:
sorry two pull request. 

I don't know merge pull request.

igrr, 2015-05-15:
That should just replace the implementation of getVcc because it's the same
thing...

chadouming, 2015-05-15:
System_get_vdd33 just work differently, they added documentation in their appendix.

Testato, 2015-06-18:
This problema is related to this, because i have your same result
https://github.com/esp8266/Arduino/issues/338#issuecomment-113099775

closed 2015-07-12 by tontito

rumbero71, 2015-07-18:
Hi,
there seems to be another problem with the ESP.getVdd33() function. 
Setup: 
-ESP-01 with XPD_DCDC soldered to RESET
-Code Initializing Serial, Initializing WIFI, connecting to an MQTT broker, Reading out a DHT22, Reading out getVdd33, pushing values to mqtt broker.
Without the getVdd33, everything is fine. Wifi is connected, DHT22 read, Values pushed out, deep sleep invoked, waking up...ad infinitum.
However, when measuring getVdd33, sometimes voltage gets read correctly, but most times the chip wdt resets. Putting in wdt_disable(), program stays indefinitely at getVdd33.
Any hints on that ?
Thanks + Regards
Markus

Testato, 2015-07-18:
Are you tried the Staging .json link ?
Found it in the documentation branch

ardyesp, 2015-07-22:
Another one experiencing occasional watchdog resets when reading VDD. The call I am using is uint16 readvdd33(void); from "user_interface.h"
It is ESP07 operating in STA mode with ADC pin floating.

Testato, 2015-07-22:
The old way to read Vdd wos completely wrong.
It is corrected now but you must use the Staging version of the core.
After the staging installation (please remove your core and install the development version by the staging .json link) you can use the new function in this way:
-----------------
ESP.getVcc() may be used to measure supply voltage. ESP needs to reconfigure the ADC at startup in order for this feature to be available. Add the following line to the top of your sketch to use getVcc:
ADC_MODE(ADC_VCC);
TOUT pin has to be disconnected in this mode.
----------

murix, 2016-05-30:
The problem is not solved with release 2.2.0

extern "C" {
#include "user_interface.h"
  ADC_MODE(ADC_VCC);
}

double get_vbat(){
double vcc = ESP.getVcc();
vcc = vcc / 1000.0;
return vcc;
}

always return 65,535

chaeplin, 2016-05-30:
@murix test this https://gist.github.com/chaeplin/d47c2de3c6d92513996cf0163aed531b

murix, 2016-05-30:
Hi, I solved the problem with version 2.2.0. I put it in separate file.
Seems a linkage problem with SDK and Arduino IDE.
Mixing C and C++ lead to buggy behaivor.

------------------------------- vbat.cpp ------------
#include <Arduino.h>
#include "user_interface.h"
#include "vbat.h"

ADC_MODE(ADC_VCC);

double get_vbat() {
  double vcc = ESP.getVcc();
  vcc = vcc / 1000.0;
  return vcc;
}
------------------------------- vbat.h ------------
double get_vbat();
--------------------------- vbat.ino --------
void setup(){
   Serial.begin(115200);
}
void loop(){
  Serial.println(get_vbat());
 delay(1000);
}
------------------------------------------

chaeplin, 2016-05-30:
@murix don't need user_interface.h.
```
ADC_MODE(ADC_VCC);

int vdd;

void setup() {
  Serial.begin(115200);
  while (!Serial) {
    ;
  }
  Serial.println();
  vdd = ESP.getVcc();
  Serial.println(vdd);
}

void loop() {
  vdd = ESP.getVcc();
  Serial.println(vdd);
  delay(5000);
}
```

arlaor, 2016-05-31:
Hi.
I'm having constant problems with my ESP8266 V1 to work with the library IRremoteESP8266.h.
I'm working with Websockets and add the library IRremoteESP8266.h, I generated the following
`Exception (0):
epc1=0x4020a3b0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffc60 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe00:  40213302 3fff175c 3ffeac32 3ffedc44  
3ffffe10:  4000437d 00000030 00000016 ffffffff  
3ffffe20:  60000200 00000008 24042403 80000000  
3ffffe30:  20000000 3fff2790 80000000 203fc120  
3ffffe40:  80000000 3fffc6fc 3ffee828 3fff2794  
3ffffe50:  000001d4 003fc120 60000600 00000030  
3ffffe60:  3ffee850 3fff03f0 3ffe9ee8 00000001  
3ffffe70:  00000000 007e4b45 4020a468 3fff03f0  
3ffffe80:  3fff0250 01948dc0 0000001f 00000000  
3ffffe90:  00000000 00000000 0000001f 40105755  
3ffffea0:  4000050c 4020a3b0 3fffc258 4000050c  
3ffffeb0:  40000f68 00000030 0000001a ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  4010634c 4020fec0 00000000 00000190  
3ffffee0:  00000113 3fff0470 3ffee828 3fffdab0  
3ffffef0:  00000000 3fffdcb0 3ffee878 00000030  
3fffff00:  00000000 400042db 3ffeabfa 60000600  
3fffff10:  40004b31 3fff2674 000002f4 003fc000  
3fffff20:  40105b3e 3ffee850 3ffed8f0 4010706c  
3fffff30:  4020f991 3ffed8f0 3ffee850 0195072d  
3fffff40:  3fff2674 00001000 4020fe26 00000008  
3fffff50:  4022e4a8 00000000 4020fed3 3ffed9a4  
3fffff60:  3ffee850 007e7134 3ffee828 3ffee850  
3fffff70:  402215a5 3ffed9a4 3ffee850 0194e82a  
3fffff80:  402215ea 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffee860 3fffdad0 3fff05f0 4020992b  
3fffffa0:  40000f49 000184c7 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld`
After that you disconnect from internet and I must reflash the pages losing orders sent

I read on this link [Stack](https://github.com/esp8266/Arduino/issues/756) was by readvdd33 and ESP.getVcc (), according @igrr   but not in that place of my sketch I implement these commands to improve the stability of my ESP8266 or if in the end it really does not help me much.

You know of any solution for this error? @igrr @murix @Testato @robertgregor @creationix @rumbero71 @ardyesp @chaeplin @	gordonendersby

arlaor, 2016-05-31:
I made the change from the library [IrRemoteESP8266.h](https://github.com/markszabo/IRremoteESP8266) and the problem has stopped. The new library works correctly with other libraries my project.Please excuse me

murix, 2016-05-31:
@chaeplin I tested again, and work with #include "user_interface.h".
I'm using release 2.2.0 with Arduino IDE 1.6.9 and PlatformIO.
In release 2.1.0 without #include "user_interface.h" , I get build error.

chaeplin, 2016-05-31:
@murix 
Ah, I haven't check the version of IDE, Thanks. 
I will add 'user_interface.h' to be sure.

arlaor, 2016-05-31:
apparently it is generated by how some libraries are written