esp8266/Arduino #226: softAP problem

Picatek, 2015-05-11:
Try using softAp by calling WiFi.softAP(ssid); 
AP will not show up on a wifi client (IPad, or windows 7 PC) most of the time. sometimes if I change the ssid it will show up for one time and disconnect. and will not let connecting a gain U"
Can't see a system in it. Tried many combinations of AUTH types with/without passwords,  etc.
also conf.max_connection = 25;
I found a patched library which is newer then the one you distribute  with arduino-1.6.1-p1-windows
However could not make it works with your distribution
The lib is from there:
http://bbs.espressif.com/viewtopic.php?f=5&t=426

The esphttpd comes with the eclipse distribution works fine in ap+sta mode. But not the arduino like distribution mentioned above

 My home router alway shows up, but not ESP.

robertgregor, 2015-05-11:
Hello, I am using this regulary and it is working perfectly. Maybe the power supply problem? (Should be able to load at least 350 mA )

luc-github, 2015-05-12:
ESP8266 softAP only support bg - be sure you are not in n

Picatek, 2015-05-12:
Regarding power: The same hardware is working with the eclips SDK
Regarding protocol (bgn) It uses the g protocol. It show on my pc. (but I can't connect to it)
Which sdk are (all those that got the AP mode to work) are using ?
It works fine for me when I use the eclips bundled sdk but got the  problem  with the Arduino arduino-1.6.1-p1-windows.zip budled SDK... and they are not the same.
For example the libnet80211.a are different. Now I can see that this lib was updated 6 days ago. May be I should just grab the new libs. I will giv it a try
 If someone got it to work relibly with the Arduino SDK I could use a simple skatch to try.
Thanks

closed 2015-11-06 by Links2004