esp8266/Arduino #227: WDT with Pubsubclient

BJvDL, 2015-05-11:
I'm using Pubsubclient with Arduino and Ethernet shield with no problems. Now with the ESP-07 I got surprised with the WDT firing.
I'm having 2 problems. When I subscribe to more than 3 items the WDT is fired. I tried putting in some delay() after the client.subscribe with no help. Then I tried wificlient.flush() behind it and after that no WDT anymore. Now I can subscribe to more than 5 items (I did not try more). Happy as I was.... I got surprised by another problem.
When the client is connected to the broker and the wifi goes down it fires the WDT. This is a bigger problem. Sometimes after a WDT reset the ESP tries to connect to the wifi and after that it comes online with the broker again, but sometimes it just keeps hanging. My application should not reset when the wifi goes down because in the mean time it is counting pulses in a running process and would mis the pulses. When the wifi is restored it could send the remaining counter data.
I already tried to put ESP. WdtDisable() in setup() but then it hangs completely after losing the wifi.

What can I do about this problem?

BJvDL, 2015-05-14:
Switched to Arduino 1.6.4 and installed ESP8266 v1.6.2 from Adafruit via Board manager.
No WDT anymore. but the loop() is not working anymore after a reconnect
I 'm using PubsubClient from Imroy https://github.com/Imroy/pubsubclient and test with the provided mqtt_subscriber example. (using another PubsubClient library has the same result)
The only thing I changed besides the ssid,pass and server is the loop() 

void loop()
{
  Serial.println("loop"); -> see if it runs
  client.loop();
  delay(1000);
}

this is what happens:

scandone
scandone
.add 0
aid 2
pm open phy_2,type:2 0 0
cnt 

connected with Zonder_draad, channel 3
dhcp client start...
..ip:192.168.12.108,mask:255.255.255.0,gw:192.168.12.254
.
WiFi connected
loop
loop
loop
loop
loop
loop
loop
beacon timeout                                -> wifi switched off
rm match
pm close 7 0 0/7959458
loop
scandone
no Zonder_draad found, reconnect after 1s
reconnect
scandone
no Zonder_draad found, reconnect after 1s
reconnect
scandone
no Zonder_draad found, reconnect after 1s
reconnect
scandone
add 0                                                          -> wifi switched on again
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with Zonder_draad, channel 3
dhcp client start...
ip:192.168.12.108,mask:255.255.255.0,gw:192.168.12.254

-> this is where it stops; nothing happens anymore

eelcor, 2015-05-16:
Did you get any further? I have a WDT problem when I want to periodically read out some sensors and have the module deep sleeping in between.

Kind regards, 

Eelco

BJvDL, 2015-05-16:
Yes, I've got it running somehow but with a lot of questions. 

'No heap available, failed to malloc 0' comes every 2 seconds

The long delay() at the end is very important otherwise it gets stuck after the wifi is restored.
I've put all my other code, which is time critical behind a Ticker.

I changed my loop() and it works

void loop()
{
  client.loop();

  if (WiFi.status() == WL_CONNECTED)
  {
    if (client.connected())
    {
      mqtt_publish(); //call to function where I publish once per 10 seconds
    }
    else
    {
      //Connect to the MQTT server
      Serial.println("Connecting to MQTT broker....");
      if (client.connect((char *)CLIENTID))
      {
        mqtt_subscribe();
      }
    }
  }
  else
  {
    //Serial.println("WiFi lost");
  }
delay(500);
}

BJvDL, 2015-05-19:
Today I changed the version of ESP8266 with Arduino 1.6.4 in Board Manager to the latest from this site and now it is getting worse.
The ESP keeps running while the WiFi drops out but resets with a WDT after the restoration of the WiFi.
After that everything is comming online again.

scandone
add 0
aid 2
pm open phy_2,type:2 0 0
cnt 

connected with Zonder_draad, channel 3
dhcp client start...
ip:192.168.12.100,mask:255.255.255.0,gw:192.168.12.254
Connecting to MQTT broker (192.168.12.3) as 18:fe:34:a0:c5:77
Connected to 192.168.12.3
Publishing...
No heap available, failed to malloc 0
Publishing...
beacon timeout
rm match
pm close 7 0 0/23994827
reconnect
scandone
no Zonder_draad found, reconnect after 1s
reconnect
scandone
add 0
aid 1
pm open phy_2,type:2 0 0
cnt 

connected with Zonder_draad, channel 3
dhcp client start...
ip:192.168.12.100,mask:255.255.255.0,gw:192.168.12.254
Connecting to MQTT broker (192.168.12.3) as 18:fe:34:a0:c5:77
Fatal exception (9):
epc1=0x401019c2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x075b1006, depc=0x00000000
Fatal exception (9):
epc1=0x401019c2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x075b1006, depc=0x00000000
Fatal exception (9):
epc1=0x401019c2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x075b1006, depc=0x00000000
Fatal exception (9):
epc1=0x401019c2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x075b1006, depc=0x00000000
Fatal exception (9):
epc1=0x401019c2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x075b1006, depc=0x00000000
Fatal exception (9):
epc1=0x401019c2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x075b1006, depc=0x00000000
Fatal exception (9):
epc1=0x401019c2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x075b1006, depc=0x00000000
Fatal exception (9):
epc1=0x401019c2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x075b1006, depc=0x00000000
Fatal exception (9):
epc1=0x401019c2, epc2=0x00000000, epc3=0x00000000, excvaddr=0x075b1006, Fataepc1=0x%L‘þ?L‘þ?[=0x%
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x40100000, len 28780, room 16 
tail 12
chksum 0xa5
ho 0 tail 12 room 4
load 0x3ffe8000, len 1580, room 12 
tail 0
chksum 0x74
load 0x3ffe8630, len 3372, room 8 
tail 4
chksum 0xf2
csum 0xf2
r‚scandone
scandone
add 0
aid 1
pm open phy_2,type:2 0 0
cnt 

connected with Zonder_draad, channel 3
dhcp client start...
ip:192.168.12.100,mask:255.255.255.0,gw:192.168.12.254
Connecting to MQTT broker (192.168.12.3) as 18:fe:34:a0:c5:77
Connected to 192.168.12.3
Publishing...
No heap available, failed to malloc 0
Publishing...
Publishing...

sticilface, 2015-05-19:
I'm getting a lot of WDT resets with this latest release and boards manager.  Some things are quite stable.. but my main sketch, with WS2812... does a lot of rebooting...  I'm back tracking commits to work out where exactly!  but it is resetting a lot!

closed 2015-06-24 by BJvDL