esp8266/Arduino #228: libiconv-2.dll missing

frasch, 2015-05-11:
When compiling windows from latest source with "ant dist" no compile with Arduino IDE is possible after installation of the package.

IDE Compile command just show the command for calling the compiler and states "Error during compile".

When excecuting same command from a dos-box there is a message box "... libiconv-2.dll missing...".

Copying the dll from the Release 1.6.1-esp8266-1.zip  from March, 27 solves the problem - but this dll is missing with the current distribution.

igrr, 2015-05-14:
Toolchains were updated yesterday in 0d376cb. Now libiconv-2.dll should appear in the right places.

Toshik, 2015-05-14:
@igrr I confirm, now it is in it's place

closed 2015-05-19 by igrr