esp8266/Arduino #229: ESP.deepSleep(10000000);

robertgregor, 2015-05-11:
Hello, I wrote simple sketch:

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(5000);
}

void loop() {
  Serial.println("Sleep....");
  ESP.deepSleep(10000000);
  Serial.println("Woked up....");
}

I have connected GPIO16 and RST pin together, I have tried this on the ESP-03 and also on ESP-07, but seems, the module doesn't sleep at all. Any idea?

ipernjakovic, 2015-05-12:
Hi,
How do you know it doesnt sleep at all?

ESP doesn't continue after the deepSleep command when it wakes up, it resets. So, your sketch will never actually come to the Serial.println("Woked up...."); line.

chaeplin, 2015-05-12:
Without delay(), module doesn't sleep at all.

Following sketch works.

void loop() {
Serial.println("Sleep.... 2");
delay(100);
Serial.println("Sleep.... 1");
delay(100);
ESP.deepSleep(20000000);
delay(100);
Serial.println("Woked up....");
}

tail 0 
chb��d䄆����������ć䦦FC����������䤆���Ǆ�b�ƄbŧĄ�FÇ�ĥ����������ǄD�ƄbŧĄ�DÇ�ĥ�ƅ���������D�ƄbŧĄ�D���ĄbbSleep.... 2 
Sleep.... 1 
b��d䄆����������ć䦦FC����������䤆���Ǆ�b�ƄbŧĄ�FÇ�ĥ����������ǄD�ƄbŧĄ�DÇ�ĥ������������D�ƄbŧĄ�D���Ąb�bSleep.... 2 
Sleep.... 1 
b��d䄆����������ć䦦FC����������䤆���Ǆ�b�ƄbŧĄ�FÇ�ĥ����������ǄD�ƄbŧĄ�DÇ�ĥ�ƅ���������D�ƄbŧĄ�D���Ąb�bSleep.... 2 
Sleep.... 1 
b��d䄆����������ć䦦FC����������䤆���Ǆ�b�ƄbŧĄ�FÇ�ĥ����������ǄD�ƄbŧĄ�DÇ�ĥ������������D�ƄbŧĄ�D���Ąb�bSleep.... 2 
Sleep.... 1

robertgregor, 2015-05-12:
Wow!!! Cool, works now. Thanks very much for the help!

closed 2015-05-12 by robertgregor

duncan-a, 2015-05-12:
It will still *never* get to:

*delay(100); Serial.println("Woked up....");*

That's why it's not appearing in the Serial output - nothing *after *
*ESP.deepSleep();* is actioned if it resets when it awakes....