esp8266/Arduino #230: Memory leak ESP8266WebServer

robertgregor, 2015-05-12:
Hello, I have a sketch, which is using the webserver. Seems there is memory leak, thus using timer, I am printing each 2 seconds:
 Serial.print("Free heap:");
 Serial.println(ESP.getFreeHeap(),DEC); 

I did the regular access to the Webserver port - and it was in debug mode and see the output:

 Free heap:6288
Free heap:6288
Free heap:6288
Free heap:6288
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:6056
Free heap:6056
Free heap:6056
Free heap:6056
Free heap:6056
New client
method: GET url: /cwd search: 
args: 
Request: /cwd
 Arguments: 
Free heap:5864
Free heap:5864
Free heap:5864
Free heap:5864
New client
method: GET url: /cww search: 
args: 
Request: /cww
 Arguments: 
Free heap:5672
Free heap:5672
Free heap:5672
Free heap:5672
Free heap:5672
Free heap:5672
Free heap:5672
Free heap:5672
Free heap:5672
Free heap:5672
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:5480
Free heap:5480
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:5288
Free heap:5288
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:5096
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:4904
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:4712
Free heap:4712
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:4520
Free heap:4520
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:4328
Free heap:4528
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:4336
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:4144
Free heap:4144
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:3952
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:3952
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:3760
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:3568
New client
method: GET url: / search: 
args: 
Request: /
 Arguments: 
Free heap:3376
Free heap:3376
Free heap:3568

It seems, that with each request, there is lost of the 200 bytes...

igrr, 2015-05-12:
> Free heap:3376
Free heap:3376
Free heap:3568

As you see, memory gets reclaimed, so it is not lost.
Closed connections go into TIME_WAIT state where they must be kept by the TCP stack for 2xMSL interval. We might though add client.abort() method to just send RST segment and be done with the connection fast...

robertgregor, 2015-05-12:
OK, thanks, yes, I see now. I did several tests and yes, you are correct.

closed 2015-05-12 by robertgregor

Lukaspp, 2015-06-27:
Thanks for your job.
 
I would like to know if there is a solution for the leak memory when server has  1 or 2 connections per second over a period of time?

Every cycle the memory is going down until it is not enough. But if the cycle stop (no connections) for a while, memory is recovering again.

Thanks for your help and best regards,

Lukaspp, 2015-06-29:
This could be a Solution: 

ReinholdHiller wrote:
In clientcontext.h I changed line 70 and following (the following if ()... is now useless, but for a test.) to:

err = tcp_close(_pcb);
tcp_abort(_pcb);
if(err != ERR_OK) {

This fixes the "memory leak" and does not seem to harm the Webserver example.

But the file is a little different now, so: I could figure it out. Modification is made for the current file release at line 64.

ClientContext.h

64 err = tcp_close(_pcb);
65	tcp_abort(_pcb); // Modification 28-06-2015
66 if(err != ERR_OK) {
67 DEBUGV(":tc err %d\r\n", err);
68 tcp_abort(_pcb);
69 err = ERR_ABRT; 

Very good Jop.

Best regards

porcao, 2015-08-24:
Lukaspp, I was the same mistake here. Exception ( 9). Practically unfeasible my project.
Your solution seems to have worked perfectly! I am at a few hours testing and not getting over the error.
Thank you for sharing!

dun46, 2017-01-24:
Hi, thanks to this post I have been able to solve a similar problem
thank you very much

laercionit, 2020-03-15:
Hello, good night, I have this problem currently using CORE 2.6.3 with WEBServer and NTP.
Anyone going through this now?