esp8266/Arduino #231: WiFi client hangs on reconnect/wifi lost

Toshik, 2015-05-12:
If ESP tries to send something over WiFi while WiFi network suddenly switches of, there is 99% chance to hang ESP.
I use following code to debug with Socket Test Tool (http://sourceforge.net/projects/sockettest/):

```
void loop() {
    delay(10);

    if (WiFi.status() == WL_CONNECTED)
    {
        Serial.print("test wf: ");
        Serial.println(counter++);
        int connected = wifiClient.connected();
        if (connected)
        {
            Serial.println("test wf: sending");
            wifiClient.write("0123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789\r\n", 132);
            wifiClient.flush();
            Serial.println("test wf: sent");
        }
        else
        {
            Serial.println("test wf: connecting");
            wifiClient.connect(SOCKET_SERVER_IP_ADDRESS, 21);
        }

        Serial.println("test wf: done");
    }
}
```

I got following output:

```
test wf: 240
test wf: sending
test wf: sent
test wf: done
test wf: 241
test wf: sending
beacon timeout
rm match
pm close 7 0 0/21552041

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

```

After that ESP can be reset only by hardware button.

Toshik, 2015-05-14:
Why it at least does not reboots? It is very bad behavior...

Links2004, 2015-05-15:
check you hardware (GPIO) setup. 
https://github.com/Links2004/Arduino#minimal-hardware-setup-for-bootloading-only

```boot mode:(1,6)``` indicates there is something wrong with GPIO0

tontito, 2015-05-15:
My usb ttl adaptor 
http://www.ebay.com/itm/291297582006?_trksid=p2057872.m2749.l2649&ssPageName=STRK%3AMEBIDX%3AIT

don't DTR and RST pin

GPIO0 button to GND and manuel firmware mode resetting push button

Links2004, 2015-05-15:
for normal boot GPIO0 need to be High. 
the watchdog get triggerd the chip reboots and check the GPIO0:
low --> bootloader mode (no start of application) 
high --> normal boot to ```setup()```

GPIO0 shut only be low (GND) if you want to upload new software in all other cases it need an pullup

Blastersky, 2015-05-24:
I have the same problem. The code I'm showing below wokr's great until I do not reboot router.

I have tried many ways to jump over this issiue, like rebooting the ESP8266, but no luck so far. :(

Any ideas?
```
      if (client.connected() != 1 || WiFi.status() != WL_CONNECTED)
      {
         ESP.reset(); 
      }
```

Complete code:
```
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include "DHT.h"

#define DHTPIN 2     // what pin we're connected 
#define DHTTYPE DHT22   // DHT 22  (AM2302)
DHT dht(DHTPIN, DHTTYPE, 15);
float t_old = 22.00;
String tempTopic;

const char *ssid =	"zzz";		// cannot be longer than 32 characters!
const char *pass =	"zzz";		//

// Update these with values suitable for your network.
IPAddress server(10, 0, 0, 9);

void callback(const MQTT::Publish& pub) {
  // handle message arrived
}

PubSubClient client(server, 1883);

String macToStr(const uint8_t* mac)
{
  String result;
  for (int i = 0; i < 6; ++i) {
    result += String(mac[i], 16);
    if (i < 5)
      result += ':';
  }
  return result;
}

void setup()
{  

  dht.begin();

  // Setup console
  Serial.begin(115200);
  delay(10);

  client.set_callback(callback);

  WiFi.begin(ssid, pass);

  int retries = 0;
  while ((WiFi.status() != WL_CONNECTED) && (retries < 10)) {
    retries++;
    delay(500);
    Serial.print(".");
  }

  delay(500);
  
  uint8_t mac[6];
  WiFi.macAddress(mac);
  tempTopic += "/";
  tempTopic += macToStr(mac);
  tempTopic += "/";
  tempTopic += "temperature/";
  if (client.connect(MQTT::Connect("arduinoClient")
	             .set_auth("testeruser", "testpass"))) {
  }
}

void loop()
{
    client.loop();  
    delay(2000);

        float t = dht.readTemperature();
        if (isnan(t)){
          Serial.println("Failed to read from DHT sensor!");
        }
        else {
        t_old = t;
        }
        
        
        String tempValue = "";
        tempValue += t_old;
        client.publish(tempTopic,tempValue);

      if (client.connected() != 1 || WiFi.status() != WL_CONNECTED)
      {
         ESP.reset(); 
      }

}

```

xdanik, 2015-06-02:
I have similar problem,

when PubSubClient was publishing message and I unplug wifi AP from power, it caused a garbage line in serial monitor and ESP restart itself...

Is there any way to protect itself from reboots?

I tried something like this, but it did not help...

if(WiFi.status() == WL_CONNECTED){
client.publish(topic, msg);
}

kdschlosser, 2015-06-05:
i guess there are alot of people with this problem, I had already launched a report about it in april. didn't get a response until today. stated wl_connected only checks to see if you have an IP address that is dynamically assigned, didn't explain anything further. my esp keeps on boot cycling until an IP address is given either static or dynamic don't matter. but is non functional until it has one, doesn't make this piece a viable micro controller because of this problem

closed 2015-09-30 by igrr

Opiron, 2016-03-31:
Hi,

Has somebody found a solution for this? 

Regards,

danoveb, 2017-01-20:
I think the best workaround until now is to detect WiFi.Status != WL_CONNECTED, and do a short deepsleep so it reboot hardwise. Easy on a Wemos, but not so easy on a SonOff when the XPD_DCDC pin isn't connected to something.

kdschlosser, 2017-01-20:
why can't the ESP just run instead of rebooting... because it can still check pin states or whatever else and store the data. but attempt a reconnect every so often. this is a micro controller.. it does many things... it shouldn't be hinged on having an IP address. what if the sole purpose to it was for channing of AP and signal strengths and datalogging the information?? it couldn't. And having an IP address or not does not mean it is not connected to wif. it just means that it couldn't communicate with a DHCP server. which at that time you could have it connect to a different AP and then perform tasks to notify of a missing DHCP or whatever. but because of how not having an IP means it has no wifi connection and causes a reboot.. you can't achieve this.

I know this is not a limitation of the ESP because micropython works in this manner. this is a limitation of this IDE

igrr, 2017-01-23:
@kdschlosser Can you share a sketch which causes ESP to boot loop if fails to get an IP address? Thanks.

kdschlosser, 2017-01-23:
This was some time ago... I changed to doing something else and I do not know if i have the sketch wandering about.

But i am assuming this is still an issue if there are people posting in a closed topic.

There are examples in this topic.. I do not know if this has been fixed or not.. i don't understand why people would be posting if it has been. so I would guess that any of the above examples would show what the issue is.

I know I posted in another thread about this same issue as i stated many moons ago. in this post. 
and I am sure that i would have put in a code example of what was going on.. again this was so long ago i can't remember.

i mean try doing this to see if the ESP boot loops
    int retries = 0;
    while ((WiFi.status() != WL_CONNECTED) && (retries < 10)) {
        retries++;
        delay(500);
        Serial.print(".");
      }
 this should spit out a bunch of dots to the serial...

I remember doing something just like this and it never printed a single dot.. it kept on rebooting.

and I am going to be completely honest with ya.. the answer I got which was something to the following "wl_connected only checks to see if you have an IP address that is dynamically assigned" and the length of time it took to get it. as well as another issue I had and the lack of a response to that one as well. and when i did get one it was some crazy workaround because using Serial.println had incredibly slow sends. like 2-3 CPS because of some issue with it only sending one byte per packet. or some nonsense. and there was no documentation explaining how to send because of this. all kind of drove me away from using this library. 

and i am sure that it has developed and become a great library.. but in the end those kinds of responses to people that are having a problem makes them not want to use it. it took 4 months and having the door closed on me once about it (i believe i was the first to point this out...) and then as more people started having the same problem.. then all of a sudden there was a problem with it..

danoveb, 2017-01-28:
This morning I tried to make a simple sketch to see more exactly if the problem was during sending or receiving.  But with this sketch it just will not reproduce the error. My ESP.restart() after loosing wifi connection reboots the esp correctly, BUT what I also noticed is that the well known(?) problem with ESP.Restart() after serial download also was gone. Can this 2 issues be related? I need some more time (that I don't have right now...) to sort out what i have in my other sketches that can lead me to the problem. The httpupdate for example is something that I use in all my sketches.

danoveb, 2017-01-29:
One step forward and two step backwards.... It quite sure now that this crash:

**Soft WDT reset

ctx: cont 
sp: 3ffefb50 end: 3ffefd70 offset: 01b0

>>>stack>>>
3ffefd00:  3ffeeb60 3ffeeb30 3ffeeac0 40202b98  
3ffefd10:  3fffdad0 00000000 3ffeeac0 40201e72  
3ffefd20:  4021d736 00000001 00000000 3fffdad0  
3ffefd30:  3ffeead8 0000000a 0000000a 402010a3  
3ffefd40:  00000000 00000000 00000001 3ffeed44  
3ffefd50:  3fffdad0 00000000 3ffeed3c 40205160  
3ffefd60:  feefeffe feefeffe 3ffeed50 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld**

happens when i'm reading the receiving buffer, and loosing Wifi connection at the same time. My test setup:
The ESP program is continuous sending 10 byte to my PC where i have VB6 application that receives the 10 bytes and then sends back 1000 bytes.
Then the ESP read outs the byte one by one with a 1 ms delay between every byte to ensure that I'm always in this routine when I disable the active SSID (Using Openwrt with a cron script that repeats this every minute). BUT it always boots in mode(3,x) now as it should.... Frustrating when I am 100% sure I could repeat the other behavior last week. Another thing about the softreboot (ESP.restart()) after serial download seems to be related to which IDE I'm working in. (See my prev. post) If I'm using Visual micro, I have that problem, with Arduino not....

Nakul93, 2018-04-17:
I am on the latest board version i.e 2.4.1 and also on latest commit but still this problem persists.

kdschlosser, 2018-04-20:
I think there is a misunderstanding of what WL_CONNECTED is. WL_CONNECTED or "Wireless LAN Connected" as it would have you believe has absolutely nothing to do with being connected to a wireless lan. it has everything to do with having an IP address. which is not the same thing. I believe what folks are wanting is a way to test to see if their WiFi dropped them. because you can still have an IP address and not a WiFi connection. this is called a static IP address. This specific feature is useless in detecting a connection loss even with DHCP because if you have your lease time set to say 1 hour (which i would never do do to the extra traffic) you would not know until the lease expired in this case it could be up to an hour later. 

As to why it causes a reboot of the ESP if there is no IP address is beyond me it should not do this. It should allow you to continue to do other tasks in the event there is no IP address at the very least. Like allow you to try and connect to a different network. But the WL_CONNECTED name needs to be gone and changed out for HAS_IP_ADDRESS or something of the sort. 

This issue has been here 3 years. I do not think it will ever get corrected and i do not think that the developers care. or maybe they simply do not want to say we have no clue why it is doing it and we have no clue how to fix it. Which I would be OK with instead of no response. Maybe a new topic needs to be created because this one is closed??? (why i have no idea problem has not been solved).