esp8266/Arduino #232: Undefined reference to `qsort'

marcoschwartz, 2015-05-12:
Hello,

Using a sketch based on the Web Server example I have the following problem since the latest commit (947188e8ac06a1f4406b4a667a0f83ec61f7bdf8):

core_esp8266_wiring_pwm.c:(.text+0x12c): undefined reference to `qsort'
core_esp8266_wiring_pwm.c:(.text+0x146): undefined reference to `qsort'

I can't pinpoint exactly what causes the error, but it was not here before this commit.

igrr, 2015-05-12:
please pull the latest commit, it's just been fixed

igrr, 2015-05-12:
It would be also be great if you could drop into our chat at https://gitter.im/esp8266/Arduino for "latest-commit-doesn't-compile" kind of issues.

marcoschwartz, 2015-05-12:
Sure, I will do that in the future! Thanks

closed 2015-05-12 by marcoschwartz