esp8266/Arduino #233: sqrt() causing reboot

leandrogs, 2015-05-12:
I have those three values `-83.00 | -181.00 | -243.00` and i'm trying to do this: 

    float amplitude = 0.0;
    amplitude = sqrt(val1*val1 + val2*val2 + val3*val3);

Every time my code reach this line, ESP8266 reboots...

chadouming, 2015-05-13:
Make sure val1*val1 doesnt return a negative.

igrr, 2015-05-13:
sqrt really doesn't work with the current version of toolchain.
I'm rebuilding with this change — https://github.com/jcmvbkbc/crosstool-NG/commit/b404fb9ec3a5c59e6b4d90741d39f7d1937431cd to see if this solves the issue.

leandrogs, 2015-05-13:
@chadouming when you multiply two negative values, the result is positive. Therefore, there is no way of multiplications are negative.

@igrr How can I use this change?

chadouming, 2015-05-13:
I know basic law of maths :P I saw strange things happen on esp8266 so this
would not surprise me. Anyway, you can use this commit if you want an
updated toolchain :

https://github.com/chadouming/hardware/commit/f1db56c4f97d1876d4311cd20ea5b5522cc47cfb

leandrogs, 2015-05-13:
@chadouming sorry if I was rude :| I'll take a look at your link...

igrr, 2015-05-14:
I have updated the toolchain yesterday but unfortunately the issue is not fixed.
In qemu this function it works fine, but when I run it on the real hardware, it hangs inside some loop.
Need to investigate further.

chadouming, 2015-05-14:
Meanwhile, you could try pow(value,0.5). Ugly hack, but it could work.

leandrogs, 2015-05-14:
Something weird just happened... I made a scketch from zero, and it worked. But on my original project it continues rebooting ESP8266
 ```
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(10);
}

void loop() {
  Serial.println(sqrt(4.3));

}
```

igrr, 2015-05-14:
Compiler optimized away that sqrt call because the argument is a constant.

quezadaminter, 2015-05-16:
I'm not sure how the sqrt function is implemented but I replaced my calls to it with:

#define sqrt(x)  exp(log(x) / 2.f)

and the call still hung causing the watchdog timer to end and reboot the chip.

I then tested calls to exp(x) and log(x) separately with the same result. Hopefully this information assists the debugging effort.

tomn46037, 2015-05-25:
Any news?

Just ran into the same issue.  Interestingly, I can do a sqrt in the setup but it fails in the loop.  See https://github.com/tomn46037/SQRTTesting  I also had it crash when I tried to to an atan2.

Yveaux, 2015-05-25:
Same here just using the log(x) function...

leandrogs, 2015-05-30:
Any progress related to this issue? @igrr, please tell me where I can start looking to help you guys.

igrr, 2015-05-31:
No progress here (working on other issues). 
Do log() and sqrt() work in NodeMCU or other ESP firmwares? If so, it is worth looking at what are the differences in libraries and toolchain. Try replacing ```libhal``` and/or ```libm``` from a working firmware and see if this fixes the issue. Looking at the generated code (objdump -S) may also give a hint at what might be happening...

quezadaminter, 2015-06-05:
I just tested by replacing the libm.a file from NodeMCU with the one in:

Arduino\hardware\esp8266com\esp8266\tools\windows\xtensa-lx106-elf\xtensa-lx106-elf\lib

There was no difference in the behavior and the ESP locked and then reset itself. I noticed that the libm.a file for linux and windows are very different sizes (about 1 meg). Not sure if that is an issue, just something I saw. I did not see a libhal.a in NodeMCU.

Can someone run this test but in Linux? I don't have access to that development environment at the moment.

quezadaminter, 2015-06-08:
I can't read the assembler from objdump -S so not much help there. However it looks to me like the implementation of sqrt(), log(), etc. are iterative algorithms. I am wondering if the watchdog times out before the function returns and causes the reset.

I did find a separate implementation of a square root algorithm. I wrote up a quick sketch with it and it provides results without crashing the ESP:

```
//http://www.codeproject.com/Articles/570700/SquareplusRootplusalgorithmplusforplusC

double powerOfTen(int num){
    double rst = 1.0;
    if(num >= 0){
        for(int i = 0; i < num ; i++){
            rst *= 10.0;
        }
    }else{
        for(int i = 0; i < (0 - num ); i++){
            rst *= 0.1;
        }
    }

    return rst;
}

double squareRoot(double a)
{
    /*
          find more detail of this method on wiki methods_of_computing_square_roots

          *** Babylonian method cannot get exact zero but approximately value of the square_root
     */
     double z = a; 
     double rst = 0.0;
     int max = 8;     // to define maximum digit 
     int i;
     double j = 1.0;
     for(i = max ; i > 0 ; i--){
         // value must be bigger then 0
         if(z - (( 2 * rst ) + ( j * powerOfTen(i)))*( j * powerOfTen(i)) >= 0)
         {
             while( z - (( 2 * rst ) + ( j * powerOfTen(i)))*( j * powerOfTen(i)) >= 0)
             {
                 j++;
                 if(j >= 10) break;

             }
             j--; //correct the extra value by minus one to j
             z -= (( 2 * rst ) + ( j * powerOfTen(i)))*( j * powerOfTen(i)); //find value of z
 
             rst += j * powerOfTen(i);     // find sum of a

             j = 1.0;
 
 
           }
 
      }
 
      for(i = 0 ; i >= 0 - max ; i--){
          if(z - (( 2 * rst ) + ( j * powerOfTen(i)))*( j * powerOfTen(i)) >= 0)
          {
              while( z - (( 2 * rst ) + ( j * powerOfTen(i)))*( j * powerOfTen(i)) >= 0)
              {
                  j++;

              }
              j--;
              z -= (( 2 * rst ) + ( j * powerOfTen(i)))*( j * powerOfTen(i)); //find value of z
 
              rst += j * powerOfTen(i);     // find sum of a
              j = 1.0;
           }
      }
      // find the number on each digit
      return rst;
 }

void setup() {
  // put your setup code here, to run once:
   Serial.begin(115200);
   
}

unsigned long MAX = 0xFFFFFF;
unsigned long i = 1;
void loop() {
  // put your main code here, to run repeatedly:
   Serial.print("i = ");Serial.print(i);Serial.print(", sqrt = ");Serial.println(squareRoot(i));
   i = i << 1;
   if(i > MAX)
   {
     i = 1;
   }
   delay(100);
}
```
Does anyone have something like this for log()? :)

robotiko, 2015-06-17:
I'm also stuck since all the math libs stopped working. 

There are many basic functions and algorithms that make use of that lib (graphics, servo , stepper, pid, etc).   As a fast patch, embedding an adhoc function might work, but I don't think is a good idea for long term for such basic need. 
Anyone had a chance to test with latest toolchain?

Links2004, 2015-06-17:
the math libs are from the compiler and not from the SDK.
latest SDK has same problem. WDT reset.

```cpp
#include <Arduino.h>

#define DEBUG_SERIAL Serial1

void setup() {
    DEBUG_SERIAL.begin(115200);

    DEBUG_SERIAL.println();
    DEBUG_SERIAL.println();
    DEBUG_SERIAL.println();

    DEBUG_SERIAL.print("[SETUP] Reset Info: ");
    DEBUG_SERIAL.println(ESP.getResetInfo());
    
    delay(100);
}

void loop() {
    DEBUG_SERIAL.println("[LOOP] sqrt:");
    DEBUG_SERIAL.flush();
    double test = sqrt(random(100));
    DEBUG_SERIAL.println(test);
}
```
not even the reset info is filled.

robotiko, 2015-06-18:
@Links2004 According to your point, then the problem is related to device overhead with other tasks?
this could be the reason why same lib doesn't work after some versions and SDKs?
FAST_MATH lib can solve this?

If not, seems that the only way to go are the adhoc solutions.

Other optimizations:
http://www.codeproject.com/Articles/69941/Best-Square-Root-Method-Algorithm-Function-Precisi

any function  using precomputed tables 

https://forums.adafruit.com/viewtopic.php?f=25&t=62598

 @quezadaminter this should help .. but other option 
http://sourceware.org/ml/libc-alpha/2007-06/msg00124.html

igrr, 2015-06-19:
Fixed in the esp8266 branch.

leandrogs, 2015-06-19:
@igrr I just tested it and ESP keeps rebooting.

igrr, 2015-06-19:
could you please share your sketch? with the latest version in the esp8266
branch I'm not able to reproduce this.

leandrogs, 2015-06-21:
```
#include <Wire.h>
#include <HMC5883L.h>

HMC5883L magnetometer;
MagnetometerScaled scaledVals;

void setup()
{ 
    Serial.begin(115200);
    Wire.begin(0,2);
    magnetometer = HMC5883L();
    magnetometer.SetScale(1.3);
    magnetometer.SetMeasurementMode(Measurement_Continuous);
}

void loop()
{
    scaledVals = magnetometer.ReadScaledAxis();
    printScaled(scaledVals);
    calcAmplitude(scaledVals);
    delay(200);
}

double calcAmplitude(MagnetometerScaled scaled)
{
    double amplitude = 0.0;
    printScaled(scaled);
    double sum = squareRoot(scaled.XAxis) + squareRoot(scaled.YAxis) + squareRoot(scaled.YAxis);
    if(sum != 0) amplitude = squareRoot(sum);
    return amplitude;
}
```

I noticed this version only uploads my sketch. Maybe I'm using an old firmware...

igrr, 2015-06-21:
Are you building the IDE from git, or using a boards manager package? In
that case, what is the version shown in the boards manager?

quezadaminter, 2015-06-23:
@igrr I am trying to build the IDE from the esp8266 branch but it fails when running the checksum for win32-xtensa-lx106-elf-gb404fb9.tgz.

The file downloads and opens in winzip ok but fails the checksum in ant. Any chance you forgot to update this value in the build file in the esp8266 branch?

quezadaminter, 2015-06-23:
Good news! The log() function is working and so is the sqrtf() function. However the sqrt() function does not link and fails with the following error:

```
Arduino: 1.6.5 (Windows 8.1), Board: "Generic ESP8266 Module, 80 MHz, 115200, 512K (64K SPIFFS)"

Build options changed, rebuilding all

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk//include -c -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\variants\generic C:\Users\mquezada\AppData\Local\Temp\build6566894843851933060.tmp\sketch_jun22a.cpp -o C:\Users\mquezada\AppData\Local\Temp\build6566894843851933060.tmp\sketch_jun22a.cpp.o 

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk//include -c -g -x assembler-with-cpp -MMD -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\variants\generic C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\cont.S -o C:\Users\mquezada\AppData\Local\Temp\build6566894843851933060.tmp\cont.S.o 

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk//include -c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\variants\generic C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\cont_util.c -o C:\Users\mquezada\AppData\Local\Temp\build6566894843851933060.tmp\cont_util.c.o 

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk//include -c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\variants\generic C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_eboot_command.c -o C:\Users\mquezada\AppData\Local\Temp\build6566894843851933060.tmp\core_esp8266_eboot_command.c.o 

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk//include -c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\variants\generic C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_flash_utils.c -o C:\Users\mquezada\AppData\Local\Temp\build6566894843851933060.tmp\core_esp8266_flash_utils.c.o 

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk//include -c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266 -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\variants\generic C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_noniso.c -o C:\Users\mquezada\AppData\Local\Temp\build6566894843851933060.tmp\core_esp8266_noniso.c.o 

C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -IC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk//include -c -Os -g -Wpointer-arith -Wno-implicit-function-declaration -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -std=gnu99 -DF_CPU=80000000L -DARDUINO= [... cut ...]
}

unsigned long MAX = 0xFFFFFF;
unsigned long i = 1;
void loop() {
  // put your main code here, to run repeatedly:
  double x = 15 + i;
   Serial.print("i = ");Serial.print(i);Serial.print(", sqrt = ");Serial.println(sqrtf(x));
   Serial.println(sqrt(16));
   i = i << 1;
   if(i > MAX)
   {
     i = 1;
   }
   delay(100);
}
```
I ran out of time before I could test other functions, such as the trig and exp() calls. Can someone else try those? Else I'll try them later this week.

@igrr Thanks!

igrr, 2015-06-23:
The remaining issue is that on Windows, libm.a was not build with those functions enabled. I will update the windows toolchain to fix this.

quezadaminter, 2015-06-23:
Sounds good. I look forward to the update. Will you be meging these changes into the SDK branch?

igrr, 2015-06-23:
sdk-1.0 branch will just be removed because sdk 1.1 is stable enough now.

nicjohnston, 2015-06-23:
Where is sdk 1.1 located? Is it the esp8266 branch?

igrr, 2015-06-23:
@nicjohnston yes, esp8266 branch is currently on 1.1.something.

nicjohnston, 2015-06-23:
Thanks.
Has anyone tested this on Linux?
If not, I should be able to test it later today.

leandrogs, 2015-06-25:
Is there any way to upload a new SDK using IDE? I'm building it from git source and only my sketch is been uploaded to the board.

nicjohnston, 2015-06-26:
This issue appears to be fixed when compiling on Linux (Ubuntu 14.04).
I am still unclear about how to install the latest version of the library using the board manager, so I simply used the executable in build/linux/arduino-1.6.4/.

Links2004, 2015-06-27:
the sdk is part of the sketch upload.

mtnbrit, 2015-06-28:
the log() function seems to be still causing the wdt reset issue on OS X build 1.6.6

closed 2015-11-06 by Links2004