esp8266/Arduino #238: ESP8266Webserver unable to send large amount of data (html)

rvbcrs, 2015-05-13:
Hi, I have the problem that I try to send a nice looking html page with _currentClient.print(response); but html is always getting capped at 2,876 bytes. Is there a way to send the whole page in chuncks? or another way to send the whole page? Anyone who can provide me with an example? Or alter the webserver code in a way it will accept a larger amount of data?

Thanks!

igrr, 2015-05-14:
8c1a40b2 hopefully fixes this: ```server.sendContent(string);``` knows how to split the payload into chunks and send them one by one.

rvbcrs, 2015-05-15:
Hi, thanks for your response! I downloaded the latests source and comiled my project again. It seems to work but when I connect to the ESP8266 I now see nothing anymore! In chrome in developer tools I get this message: (failed)net::ERR_EMPTY_RESPONSE

this is my current code:

    Serial.println("*** START HTML ***");
    Serial.println(homePage);
    Serial.println("*** END HTML ***");
    server.send(200, "text/html", homePage);

i see the complete HTML displayed in the serial console, but nothing in the browser!
Any advise?

robertgregor, 2015-05-15:
This could be problem of the stack - if you have a lot of Strings in the sketch, the stack is full.

rvbcrs, 2015-05-15:
Hmm I'm pretty new to all this! Can you tell me what to do to prevent this? Use F() around the string? Or use PROGMEM? I thoughtI read somewhere that it was not working.. 

Thanks!

rvbcrs, 2015-05-16:
Got it working! I've replaced all my files again with a freshly downloaded one from yesterday, put my string inside F() and now it works perfectly and fast!

closed 2015-05-19 by igrr