esp8266/Arduino #244: Reset or Restart alwayf wdt reset

tontito, 2015-05-15:
i have save eeprom ssid and password html and reset esp.

ESP.reset() or
ESP.restart() or 
system_reset() or 
system_restart() or

i tried code same error
wdtDisable()
ESP.restart()

Error:
 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

not boot i have hardware reset and working esp.

Links2004, 2015-05-15:
add a pullup resistor to GPIO0 (10k). 
for normal boot you need a high (3V3) signal on GPIO0.

tontito, 2015-05-15:
Yes work.

GPIO0 to 3.3v resistor and 
ESP.restart()

ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 29504, room 16 
tail 0
chksum 0x9c
load 0x3ffe8000, len 1596, room 8 
tail 4
chksum 0xf6
load 0x3ffe8640, len 3276, room 4 
tail 8
chksum 0xc9
csum 0xc9

boot working normaly.

closed 2015-05-23 by tontito

phamtankhai2202, 2015-08-11:
hi @tontito 
i have a problem like you and i don't know resolve... Can you help me?

phamtankhai2202, 2015-08-11:
i tried add  pullup resistors to GPIO0 (10k) and GPIO2(10k), but it not work.
 
ets Jan  8 2013,rst cause:4, boot mode:(1,7)
wdt reset

asetyde, 2015-08-11:
Also my board not always Reset correctly , I think is strong important that esp.restart() do a reset, if not I will try to use a pin to rst

igrr, 2015-08-11:
Make sure you also have GPIO15 pulled low, and both RST and CH_PD pins pulled high.

phamtankhai2202, 2015-08-12:
yes,ESP.restart() working . when you Upload Code to ESP8266. ESP8266 in mode UART so when restart esp will in mode UART. you should turn off power after use ESP.restart()

lnsriravichandran22, 2015-09-01:
Sorry for asking this Question here. 
Is there any option in esp8266 for remembering SSID and Password without storing it in EEPROM?

Thank in Advance!!

Links2004, 2015-09-01:
take a look here:
https://github.com/esp8266/Arduino/issues/724#issuecomment-135520586

lnsriravichandran22, 2015-09-02:
@Links2004 Thanks a lot for your reference!!. But I could not understand , how it could remember the SSID and PASSWORD, even after the unit is restarted??

igrr, 2015-09-02:
SSID, password, and WiFi mode (AP/STA) are stored in flash memory.
Therefore device will attempt to connect to the last AP, even after reset.

lnsriravichandran22, 2015-09-02:
@igrr Thanks for that!!. I have a question now. 
If so, then where should we mention the WiFi.begin () in the code? If it is in the void setup(), the code will anyways run it too and this applies for void loop() too right? As @Links2004 said, unless "WiFi.begin()" is called, the unit retains the SSID and PASSWORD set by the user.

Links2004, 2015-09-02:
using ```WiFi.begin()``` in the setup is no problem, but you not need to do it.
the cases where you not doing it are for example is you have a configuration process where the you setup the connection once by Web interface, a APP or WPS.
Its also possible to flash one time an a code with the ```WiFi.begin()``` begin in setup and after this you can flash other code where you not need to do the begin any more. 
to delete the local stored information you can use ```WiFi.disconnect()```

lnsriravichandran22, 2015-09-03:
@Links2004 Thanks a lot for the information!!.  Sorry for my Consecutive questions. It would be of a great help if you could post here an example for setting up the connection only once and to use that connection further , unless and until there is a change with the network configuration!!

Thanks in Advance!!

Links2004, 2015-09-03:
a simple SSID check will do it for example:
```cpp
#define SSID "test123"
#define PW "123456789"

if(String(SSID) != WiFi.SSID()) {
     WiFi.begin(SSID, PW);
}
```
can be called in setup or loop.

lnsriravichandran22, 2015-09-08:
@Links2004 For the First time, I would use this "WiFi.begin(ssid,password)" to connect with the specified network. Let's say that I am restarting the WiFi Module after successfully connecting with the specified network. As @igrr says https://github.com/esp8266/Arduino/issues/244#issuecomment-136947601, how could I reconnect with the network , without using WiFi.begin() after restarting it. In short , my question is if WiFi.begin () is to be used , whenever there is a change with the network settings, so which command should I use to connect with the network using the ssid and password that are stored already there in the flash memory i.e when there is no change with the ssid and password?

mtnbrit, 2015-09-08:
> @Links2004 <https://github.com/Links2004> For the First time, I would use this "WiFi.begin(ssid,password)" to connect with the specified network. Let's say that I am restarting the WiFi Module after successfully connecting with the specified network. As @igrr <https://github.com/igrr> says #244 (comment) <https://github.com/esp8266/Arduino/issues/244#issuecomment-136947601>, how could I reconnect with the network , without using WiFi.begin() after restarting it. In short , my question is if WiFi.begin () is to be used , whenever there is a change with the network settings, so which command should I use to connect with the network using the ssid and password that are stored already there in the flash memory i.e when there is no change with the ssid and password?

WiFI.begin(“”,””)

Links2004, 2015-09-08:
@lnsriravichandran22 you need to do nothing the SDK handle it itself no need for any peace of code.
@mtnbrit  the code will Overdrive the flash stored informations

mtnbrit, 2015-09-08:
its working for me. 

What IS the right command to start wifi using stored credentials then?

Links2004, 2015-09-08:
no command is needed this mean nothing too do. eaven with a code like this:

```cpp
void setup(void) {
}

void loop(void) {
}
```
the SDK will connect to the last wifi configuration :)

the ```WiFI.begin(“”,””);``` is only working be course the code check for wrong configurations like an empty SSID.
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp#L102
so it do not change the current Wifi settings.

Humancell, 2015-09-08:
I'm curious ... is there a way to have the module only connect when I explicitly request it to?  If you are saying it will always connect to the last stored credentials, is there a way - before I go to DeepSleep - to "disconnect" the Wifi and NOT have it reconnect when it wakes ... until I explicitly tell it to?

Links2004, 2015-09-08:
yes ```WiFi.disconnect();``` will do it.
https://github.com/esp8266/Arduino/blob/d108a6ec30193a8e44eb9c72efc7a55853b54f09/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp#L208

hallard, 2015-11-10:
Sorry guys, just discovered this very interesting point and I'm wondering if I can have more details?
If SDK memorize AP credential, when do we know when it start trying to connect? and when do we have hand in our code? 
IE does I need to wait connection in setup or is it already done when called setup() ?

I'm asking this because I want to manage connexion and I think it's not very optimized to disconnect (because SDK is connected) to plug my connection routine. By the way is it safe calling wifi.begin(bla,bla) when SDK is still trying to connect ?
Does setting SDK autoconnect to 0 (seen calling `WiFi.printDiag(Serial);`) prevent it doing anything ?

```
========== SDK Saved parameters Start
Mode: STA
PHY mode: N
Channel: 1
AP id: 0
Status: 0
Auto connect: 0
SSID (12): HOME-HOTSPOT
Passphrase (13): blablabla
BSSID set: 0
========== SDK Saved parameters End
```

Links2004, 2015-11-10:
sdk docs says:
[image]

but I have never seen that the value is to be respected by the SDK, it looks like it allways connect to the last settings after boot up.
```wifi_station_set_auto_connect``` is never used in the core.

the easiest way is may simply compare the wifi config with stored one from SDK and if no changes,
no need to call ```WiFi.begin```.

I think its possible change the begin function to do this kind of check internally so that there is no need for the application to keep track of it.

hallard, 2015-11-11:
@Links2004 
thank for these explanation ;-)

calozano, 2015-11-29:
I got the same issue showing the reset was for watchdog. I just Insert "ESP.wdtDisable();" inside loop and it doesn´t reset again.

Mewiss, 2016-01-18:
ESP.wdtDisable(); worked for me but my code had some while's and I had to put this line in each while

JitenMhatre, 2016-03-02:
ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld

please help!!!

Links2004, 2016-03-04:
please open a new Issue for you problem and provide more information's,
like core version, your hardware, your code.

aaszabo, 2016-04-10:
Hi,
I had the same reboot issue (wdt reset rst cause:4)
After trying many things and ran out of ideas I realized that I changed the ESP panel settings about the  FLASH by mistake. Even the upload went well the module was not able to operate. The origin settings was 512k, but I set it to 1M. As the 512k was set back everything went well again.

I hope this helps You too!

amirebr, 2016-05-17:
Hi everybody.

To prevent reset, go to:
Tools -> Flash Size: -> 1M(512K SPIFSS)

It solved my reset problem in login to network.

webliya, 2016-05-28:
Thank you amirebr it saved my day.
 
amirebr:
To prevent reset, go to:
Tools -> Flash Size: -> 1M(512K SPIFSS)

It solved my reset problem in login to network.

NicAshby, 2017-04-03:
I'm having this issue.
What "solved it" for me, was to touch my finger to the antenna edge of the esp board...and keep holding it.

as soon as I let go, the connection drops.

this wifi could be awful as i've had other connectivity issues with other devices here.
The router is less than 20 feet away on the other side of a thin wall, so I don't know.

KamaJogi, 2017-08-11:
NodeMCU v1.0 ESP-12E -> This solved my problem:

_void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
...
}_

They conect LED to GPIO16 - this is the case

LED_BUILTIN = D0 = GPIO16 and GPIO16 has Deep-sleep/wake function

look this: https://www.losant.com/blog/making-the-esp8266-low-powered-with-deep-sleep

**In this case GPIO16 can't stay floating pin, use:  "pinMode(LED_BUILTIN, OUTPUT);" to set pin GPIO16.**

For me, this is work even for Adafruit NeoPixel - no "wdt reset" ;)

**I don't see this message any more:**
_wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld_

rodrigormelo, 2018-07-01:
@KamaJogi Thanks a lot for your tip!!

asdrubalfuentes, 2018-07-01:
Good

rashey04ec, 2019-07-19:
I have solved it !!,
Please try to comment all pin you used in programming and it's related function )like pinMode , digitalWrite etc.
 - Again save and upload it 
- check the serial monitor 
- if you not getting error that means this is because of pin declaration
-  to solve this issue : dont use #define declaration for initialize pin.
-Direct use pin in pinMode definition and read /write function
- Use GPIO no in if you are doing it in ARDUINO IDE

May be helpfull 

Thanks : 
Jaydeep Shah
E-mail : radhey04ec@gmail.com