esp8266/Arduino #249: Error when compiling PubSub example

marcoschwartz, 2015-05-15:
Hello,

With the PubSub example I am having an error:

```c
In file included from esp8266_mqtt.ino:2:0:
/Users/marco/Dropbox/Arduino/libraries/PubSubClient/PubSubClient.h:72:46: error: section attribute not allowed for '<anonymous>'
    boolean publish_P(char *, uint8_t PROGMEM *, unsigned int, boolean);
                                              ^
```

Seems to be related to the PROGMEM thing, but as I didn't test this example for quite a while I don't know which commit caused the issue.

Makuna, 2015-05-16:
That line is incorrectly defined.  const unit8_t* or PGM_P would be correct.

Imroy, 2015-05-16:
Sorry about that. It's been fixed.

marcoschwartz, 2015-05-16:
Is this already fixed? I am still getting the error with the latest commit

Imroy, 2015-05-16:
Oh, you're using Nick's original version. My version definitely works with ESP8266 and has extra features (and is mostly compatible).
https://github.com/Imroy/pubsubclient

marcoschwartz, 2015-05-16:
Oh got it. I'll use yours from now on then. Thanks!

BJvDL, 2015-05-16:
@Imroy I'm using your's. but did you test what happens when the Wifi drops out?
The loop() isn't scanned anymore after a Wifi restore.
This problem happens to all the Pubsubclient versions

Imroy, 2015-05-20:
@BJvDL When Wifi drops out I presume WiFiClient reports so, and loop() returns false. You'll need a more complex loop in your sketch that checks the Wifi and reconnects it and the pubsubclient.

BJvDL, 2015-05-20:
@Imroy I mean the Arduino main loop() isn't working anymore. The ESP hangs.

closed 2016-02-29 by igrr