esp8266/Arduino #253: Wire library documentation update

sanchosk, 2015-05-16:
Can someone, please, update the front-page readme info to reflect the real i2c communication start?
First of all, Wire.begin(0,2) as described in the example will not compile.
There is other function, called Wire.pins, that should be used for setting the SDA SCL pins.
And one needs to issue also Wire.speed command, otherwise my i2c devices did not work.

TheKitty, 2015-06-14:
I second this, the only way I can compile is
  Wire.pins(0,2);         // SDA, SCL
  Wire.begin();       

as Wire.begin(0,2) throws an argument error.  It looks like there are issues in the Wire.cpp but I cannot provide you with specifics.

Lstt2005, 2015-06-17:
Please, update the front-page readme info to reflect the real i2c communication start...

igrr, 2015-08-05:
`Wire.begin(sda, scl)` works fine now in the latest version.

closed 2015-08-05 by igrr

AdamMiltonBarker, 2016-05-25:
@igrr can you direct me to some example code please I am struggling to find any docs on connecting the elechouse module 3

chaeplin, 2016-05-25:
@AdamMiltonBarker Is 'elechouse module 3' this one http://www.elechouse.com/elechouse/index.php?main_page=product_info&cPath=168_170&products_id=2254

AdamMiltonBarker, 2016-05-25:
No mate it is this, I have already developed the device on an Arduino, then moved to my second stage of project which included standalone ATMEGA but docs were thin and realised it could be done with ESP but same situation docs are thin and cannot get it working

https://dangerousthings.com/wp-content/uploads/2013/12/PN532_-Manual_V3.pdf

chaeplin, 2016-05-25:
@AdamMiltonBarker 
Elechouse has lib for PN532.
example) https://github.com/elechouse/PN532/blob/PN532_HSU/NDEF/examples/ReadTag/ReadTag.ino
I think one more line needed.
```
void setup(void) {
    Serial.begin(9600);
    Serial.println("NDEF Reader");
    //Wire.begin(sda, sol)
    Wire.begin(0, 2l) // <------ this
    nfc.begin();
}
```

AdamMiltonBarker, 2016-05-25:
I am asking about Esp8266, I have already done this in Arduino I need to get it working with an ESP8266 01 using i2c. Like the guy in this post did.

https://github.com/esp8266/Arduino/issues/501

AdamMiltonBarker, 2016-05-25:
One sec I didnt think about that. will try run my old code on there.

AdamMiltonBarker, 2016-05-25:
No it is not finding it connected to esp, this topic I am replying to is what I need to do, somehow ;)

chaeplin, 2016-05-25:
@AdamMiltonBarker Sorry for not working.
Have you checked this line ? https://github.com/elechouse/PN532/blob/PN532_HSU/PN532_I2C/PN532_I2C.cpp#L17

igrr, 2016-05-25:
Also make sure you have pull up resistors on SDA and SCL lines. They aren't required on Arduino boards but are necessary on the ESP, in most cases.

AdamMiltonBarker, 2016-05-25:
Thanks for the replies guys, yes have added pullups, current se up is SDA out of the module to GPIO 0 and SCL of module to GPIO 2 (ESP8266 01), 10K pull up resistors to 3.3v, and the basics of the code is exactly as provided in the example, I tried my original code then tried the example which is basically my original with everything else stripped out and I get the following:

`System Initiating...
Didn't find PN53x board
Soft WDT reset

ctx: cont 
sp: 3fff26a0 end: 3fff28b0 offset: 01b0

>>>stack>>>
3fff2850:  3fff1670 00000000 00000001 4020d32c  
3fff2860:  3fffdad0 00000000 3fff185c 40207904  
3fff2870:  feefeffe feefeffe feefeffe feefeffe  
3fff2880:  feefeffe feefeffe feefeffe 3fff1888  
3fff2890:  3fffdad0 00000000 3fff1880 4020eb10  
3fff28a0:  feefeffe feefeffe 3fff1890 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld

`

chaeplin, 2016-05-25:
@AdamMiltonBarker found this https://github.com/adafruit/Adafruit-PN532
ESP8266 : SPI only, I2C clock stretching not supported
hmm..

AdamMiltonBarker, 2016-05-25:
Thanks just realised you are same person from ESP8266 forum. For some reason I could never get SPI working with the chip in my hand either so have stuck with i2c exactly as it seems that guy has done, will try that solution thanks will get back to you buddy.

AdamMiltonBarker, 2016-05-25:
Based on set up description above does everything look on par, IE I the set up is correct ?

AdamMiltonBarker, 2016-05-25:
Nope adding the code in that topic to my existing set up I got: 

exit status 1
'class NfcAdapter' has no member named 'setPassiveActivationRetries'

Totally changing the includes to match exactly what he had done I got:

exit status 1
'class PN532' has no member named 'tagPresent'

chaeplin, 2016-05-25:
How about just adding Wire.setClockStretchLimit(2000); ?

AdamMiltonBarker, 2016-05-25:
Compiling now, two mins

AdamMiltonBarker, 2016-05-25:
No same as above,

System Initiating...
Didn't find PN53x board
Soft WDT reset

ctx: cont 
sp: 3fff26a0 end: 3fff28b0 offset: 01b0

>>>stack>>>
3fff2850:  3fff1670 00000000 00000001 4020d358  
3fff2860:  3fffdad0 00000000 3fff185c 40207914  
3fff2870:  feefeffe feefeffe feefeffe feefeffe  
3fff2880:  feefeffe feefeffe feefeffe 3fff1888  
3fff2890:  3fffdad0 00000000 3fff1880 4020eb3c  
3fff28a0:  feefeffe feefeffe 3fff1890 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld

AdamMiltonBarker, 2016-05-25:
Also the library used for elechouse is not the same as the one used by Seeed which is probably why it is not working, hmm I am still coming back to what I need to do is the origin of this post somehow, I know the guy here had it working with the same module hoping he will show up ;) 

https://github.com/esp8266/Arduino/issues/505

chaeplin, 2016-05-25:
@AdamMiltonBarker I can compile https://github.com/elechouse/PN532/tree/PN532_HSU/PN532/examples/readMifare
As elechouse forked Seeed-Studio/PN532.

And I have changed _wire->begin(); to _wire->begin(0, 2); in PN532_I2C.cpp

https://gist.github.com/chaeplin/407213f992deada54d9123e44774e394

Can you try the sketch ?

Without lib, raw i2 command can be issued.

#define PN532_I2C_ADDRESS       (0x48 >> 1)
#define PN532_COMMAND_GETFIRMWAREVERSION    (0x02)

AdamMiltonBarker, 2016-05-25:
Cool appreciate the help will check it out now. By the way if you have FB etc drop me a message have a few platforms you may be interested. adammiltonbarker@gmail.com

AdamMiltonBarker, 2016-05-25:
Sorry forgot to mention mayfare will not work with my chip  have only ever been able to get NDEF working, will try with your code though now but do not think it will work after past experience.

AdamMiltonBarker, 2016-05-25:
Hello!
Didn't find PN53x board
Soft WDT reset

ctx: cont 
sp: 3ffefba0 end: 3ffefd80 offset: 01b0

stack>>>
3ffefd50: 3fffdad0 3ffeeb98 3ffeed34 402020b9

3ffefd60: 3fffdad0 00000000 3ffeed50 40202f6c

3ffefd70: feefeffe feefeffe 3ffeed60 40100718

<<<stack<<<
ets Jan 8 2013,rst cause:2, boot mode:(1,2)

ets Jan 8 2013,rst cause:4, boot mode:(1,2)

wdt reset

chaeplin, 2016-05-26:
@AdamMiltonBarker https://github.com/ESP8266nu/ESPEasy/blob/master/_P017_PN532.ino i2c is used.

supersjimmie, 2016-05-26:
Please see the last few days from this discussion: https://github.com/esp8266/Arduino/issues/1541
I have trouble with the PN532 too.
Firstly, I added the stretchlimit AFTER the .begin(), and I  have made things just a bit easier by avoiding the continues reboots, by doing this:
```
  nfc.begin();
  Wire.setClockStretchLimit(2000);
  do {
    versiondata = nfc.getFirmwareVersion();
    if (! versiondata) {
      Serial.println("Didn't find PN53x board");
      delay(100);
    }
  }  while (! versiondata); // retry
```
That will cause a retry meganism instead or rebooting all the time.

Then I noticed some possible hints, as described in that topic.

AdamMiltonBarker, 2016-06-07:
@supersjimmie thanks with your code it will not compile it says 

'versiondata' was not declared in this scope

MulajEgzon, 2016-06-22:
@AdamMiltonBarker Did you get to work PN532 using i2c ?

AdamMiltonBarker, 2016-06-22:
Yes I did mate but it keeps crashing, I have ordered some different scanners to see if they are any better

MulajEgzon, 2016-06-22:
Cool :). Could you please send me the code or tell me what was the problem ?

MulajEgzon, 2016-06-29:
Hi Everyone i have fixed the problem ! Does anyone needs the example code  ?

AdamMiltonBarker, 2016-06-29:
@MulajEgzon  Hi sorry hadn't seen your last reply, if it solves the crashing issue I was having it would be definitely welcome, very thinly spread at the moment so anything that can save time would be very helpful.

AdamMiltonBarker, 2016-07-17:
@MulajEgzon  I have started playing around with this again, I have it working but it crashes sometimes and after a while of innactivity it just freezes, what did you do to solve the issues?

MulajEgzon, 2016-07-20:
Are you working it on Arduino or in Native C ?

AdamMiltonBarker, 2016-07-20:
Arduino IDE mate

MulajEgzon, 2016-07-20:
I found this https://github.com/pasko-zh/brzo_i2c it is for Arduino but, i made it on C . If you want my code to get UID from Mifare cards with pn532 modul tell me i will post the link.

Good Luck

AdamMiltonBarker, 2016-07-20:
I have it all working but it crashes regularly

MulajEgzon, 2016-07-20:
Did it passed a day without crashing , and did you reset it after it crashed ?

AdamMiltonBarker, 2016-07-20:
no not a day it works for a while then it crashes, constantly getting disconnected from MQTT as well, yes I reset after crash and sometimes it crashes straight after reset

AdamMiltonBarker, 2016-07-20:
After a bit of innactivity it fails to read with no crashing also

MulajEgzon, 2017-05-08:
Here is the code for reading mifare cards using pn532 module with esp8266 module.
https://github.com/MulajEgzon/ESP8266-PN532

MulajEgzon, 2017-06-01:
@Bharathgoku 
Here is an example .
Don't  forget to add two pull up resistors !
https://github.com/MulajEgzon/ESP8266-PN532

marcunzueta, 2017-06-15:
@MulajEgzon The example you mention, is using the NFC reader from Adafruit with the ESP8266 via I2C? I've not been able to find it out on the link you provided, just the whole library. Thanks! :)