esp8266/Arduino #255: Where Dns lookup timeout settings?

tontito, 2015-05-17:
Hi,

i have NTP Server to RTC Clock synchronization and get webpage value.

const int httpPort = 80;
client.connect("www.tredcafe.com", httpPort)); //no internet 5000 millis for time out
WiFi.hostByName("time.windows.com", timeServer); //no internet 5000 millis for time out

if internet normaly no problem but no internet 5 second freeze esp8266 for timeout error.

how setting DNS time out?

igrr, 2015-05-17:
DHCP client timeouts are set somewhere inside Espressif libraries.
We may be able to change these timeouts using lwIP APIs: 
```C++
/* set the DHCP timers */
static void dhcp_timeout(struct netif *netif);
static void dhcp_t1_timeout(struct netif *netif);
static void dhcp_t2_timeout(struct netif *netif);
```

The TCP client timeout is currently hardcoded in our code:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/include/ClientContext.h#L206

tontito, 2015-05-17:
The TCP client timeout is currently hardcoded in our code:
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/include/ClientContext.h#L206

it didn't work.

I asked how can i change dns timeout espconn_gethostbyname 
bbs.espressif.com forum 
as for the answer I'll try

closed 2015-05-22 by igrr

Msprg, 2023-10-18:
I know this is ancient but, for (likely my own) future reference, `WiFi.hostByName();` has a timeout parameter, and can be used as such:

```
IPAddress *IP = new IPAddress;
WiFi.hostByName(NTPserver, *IP, 50); // DNS query with timeout
    if (IP->isSet())
      NTP = new NTPtime(IP->toString());
delete IP;
```