esp8266/Arduino #256: UART1 using as DEBUG terminal

4refr0nt, 2015-05-17:
ESP8266 support 2 UART: UART0 and UART1. Usually we use only UART0. 
How I can use UART1 as second debug terminal (GPIO2=TX1, RX not needed, but usefull, GND - common) ?

Links2004, 2015-05-17:
yes 
```
    Serial1.begin(921600);
    Serial1.setDebugOutput(true);
```

closed 2015-05-18 by 4refr0nt

akkaman007, 2018-11-25:
Serial1.begin(921600);
    Serial1.setDebugOutput(true);

I tried these commands but its not working.
Can you provide a example code for this ?

akkaman007, 2018-11-25:
I cant get it to work. I have my second USB to UART's Rx pin connected to the ESP's Rx pin on D4 (which is GPIO2 on a NodeMCU from what I found) and the USB to UART's GND connected to one of the GND pins on the NodeMCU.

And yet when I connect to the second tty, I'm getting no output but if I use the built-in tty (the one used to flash the chip), I am getting the debug output I programmed.

I'm stumped.

devyte, 2018-11-25:
In the IDE set the debug port, rebuild, reflash.