esp8266/Arduino #257: lose connection without broadcast SSID

Juppit, 2015-05-17:
Immediately after the start of the web server application I get a connection from my browser, but no more after about half a minute.
This depends on the fact that the network SSID is hidden. As soon as I broadcast it, connections are ok.

ipernjakovic, 2015-05-19:
I've had the same issue, and it turned out to be the problem with the router, or, more specifically some timeout settings with idle connections.  As soon as i switched to a different router, everything worked just as expected. So, if you can, try it with a different router maybe it will help.

Juppit, 2015-05-21:
Meanwhile, I was able to test at two other routers. And I had stable connections with hidden SSID.
Thus, this matter should have done.
Thank you for your support.

closed 2015-05-21 by Juppit