esp8266/Arduino #258: softAP mode udp broadcast problem

tontito, 2015-05-17:
hi, 
i think the problem is SDK.

1 PC
1 Android
1 ESP8266 
All device connect to my modem no problem udp broadcast send and receive all device but
ESP8266 softAP mode and connect to Android and PC send broadcast modem (which does not matter)
1 client not receive broadcast message (sometimes all client receive)

tontito, 2015-05-19:
Espressif answer this question:

station A <-> ESP8266 softAP <-> station B

station B can not receive UDP broadcast from station A now，because ESP8266 softAP won‘t re-post the packet.

But we are working on this new feature now, it will be supported in the future.

Sorry for the inconvenience.

Espressif_Faye 281 Mon Oct 27, 2014 11:08 am

igrr, 2015-05-19:
Packet routing is not enabled in Espressif's lwIP build, so this applies not only to broadcast but to unicast and multicast as well.

closed 2015-05-22 by igrr