esp8266/Arduino #263: 'min' was not declared in this scope

blurfl, 2015-05-18:
A trivial sketch raises a compile error "'min' was not declared in this scope".
I don't know enough to dig further, but one clue might be that commenting out the line:
#include "FileSystem.h"
in Arduino.h allows the trivial sketch to compile.

--------------------------
Arduino: 1.6.5 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 115200, 512K (64K SPIFFS)"

junk.ino: In function 'void setup()':
junk:3: error: 'min' was not declared in this scope
junk.ino:3:16: note: suggested alternative:
In file included from /Users/username/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/memory:62:0,
                 from /Users/username/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/FileSystem.h:24,
                 from /Users/username/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:214,
                 from junk.ino:1:
/Users/username/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:239:5: note:   'std::min'
     min(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
junk:4: error: 'max' was not declared in this scope
junk.ino:4:16: note: suggested alternative:
In file included from /Users/username/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/memory:62:0,
                 from /Users/username/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/FileSystem.h:24,
                 from /Users/username/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:214,
                 from junk.ino:1:
/Users/username/Desktop/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_algobase.h:260:5: note:   'std::max'
     max(const _Tp& __a, const _Tp& __b, _Compare __comp)
     ^
'min' was not declared in this scope

--------------------------
[code]
// junk.ino
void setup() {
 int x = min(3,4);
int y = max(3,4);
}
void loop() {
}
[/code]

blurfl, 2015-05-18:
The most recent commit, 'fix tools path', seems to have fixed this, so closing....

blurfl, 2015-05-18:
It appears I didn't test enough; the issue still occurs :(

closed 2015-05-18 by igrr

redflag2k, 2015-07-09:
I just download 1.6.5 via http://arduino.esp8266.com/staging/package_esp8266com_index.json .

I have same issue and same error message with blurfl. 

This error is not occurred when I use 1.64 version.

How can I fix it? 

Thanks in advance.

gamecompiler, 2015-09-04:
Id fixed this while adding 

#define min(a,b) ((a)<(b)?(a):(b)) 

to my sketch.

NickWaterton, 2016-03-12:
Min/max seems broken again in latest git version as of today.

Links2004, 2016-03-13:
```min``` and ```max``` are not working in all cases since the define can brake normal c++ features like ```std::vector``` they are disabled in some cases. 
you can try ```_min``` and ```_max``` they are allways defined and not brake c++

https://github.com/esp8266/Arduino/blob/master/cores/esp8266/Arduino.h#L251-L258

peteroid, 2016-05-03:
Same issue of "'min' was not found in this scope". Just install the ver 2.2.0

fabifiess, 2016-07-25:
Same here with 1.6.9.
However only if the sketch contains other libraries for the ESP8266 (ESP8266WiFi.h, WiFiUdp.h, OSCMessage.h).
Is there a solution?

colonialT, 2016-08-15:
Had the same issue, min() worked before adding the following libraries:

`#include <ESP8266WiFi.h>
#include <WiFiUdp.h>
#include <OSCMessage.h>
#include <OSCData.h>`

I replaced `min()` with `_min()` which compiled and worked with a simple print sketch I used to test it.

commit 07ecca19 (2015-05-18): Fix issue with min/max

fix #263