esp8266/Arduino #265: _BV was not declared

Duality4Y, 2015-05-18:
Hi, tried compiling a lcd example that I downloaded from here:
https://bitbucket.org/fmalpartida/new-liquidcrystal/downloads
and got the error:
Build options changed, rebuilding all
/home/robert/sketchbook/libraries/LiquidCrystal/FastIO.cpp: In function 'void fio_shiftOut1(fio_register, fio_bit, uint8_t, boolean)':
/home/robert/sketchbook/libraries/LiquidCrystal/FastIO.cpp:214:19: error: '_BV' was not declared in this scope
   if(value & _BV(i))
                   ^
Error compiling.

if I remember correctly all _BV does is 1 << i
like bitshift.

Duality4Y, 2015-05-18:
ok looked at https://github.com/esp8266/Arduino/issues/125
i saw the same issue and i saw a commit for fixing it ? but that isn't in the link to the zip for linux.
because if i look in Arduino.h there is no #define _BV(b) (1UL << (b)) in it.
so i added it myself

Links2004, 2015-05-18:
the current release packages are behind the latest git version. you need to build the latest version yourself. 
or at least copying all the hardware/esp8266com/esp8266 and libraries stuff from the current git.

Duality4Y, 2015-05-19:
that worked ! copying the folder :)

Duality4Y, 2015-05-24:
Ok so i downloaded the git repo and ran ant dist, that gave me a archive i unpacked that, and ...  when i tried to compile my code it gave me that error ? i though it was supposed to be the newest code :)? also i can't find the Arduino.h anymore that has this thing build into it ..

closed 2015-05-24 by Duality4Y