esp8266/Arduino #268: Remote firmware update?

zenmanenergy, 2015-05-18:
Is it possible to upgrade the codebase so we can remotely upload new firmware?

lukecyca, 2015-05-19:
It should already be possible, but you would need to do it by calling the underlying functions directly. The IoT SDK has a series of `system_upgrade_*` functions, and Sming has a module called 
[HttpFirmwareUpdate](https://github.com/anakod/Sming/blob/71a5a23849b30ae292de51f3fdd8d511f1703f19/Sming/SmingCore/Network/HttpFirmwareUpdate.h) and [an example](https://github.com/anakod/Sming/blob/master/Basic_AirUpdate/app/application.cpp) which might be good starting points.

It would great to eventually expose this through the ESP object with a more user-friendly API.

zenmanenergy, 2015-05-19:
Holy crud. That looks fucking awesome!! Thank you!

igrr, 2015-06-11:
We now have parts of OTA update infrastructure ready. Uploads of sketch from the IDE over WiFi are not ready yet, but here is an example sketch which can be updated using a python script:

sketch: https://gist.github.com/igrr/43d5c52328e955bb6b09

python script: https://gist.github.com/igrr/d35ab8446922179dc58c

Pablotis, 2015-06-15:
I'm interested on that

Pablotis, 2015-06-15:
I can't upload:
C:\Users\pablo_000\Downloads>python ota_server.py 192.168.1.25 Wifi_temp_post.cp
p_00000.bin
Traceback (most recent call last):
  File "ota_server.py", line 64, in <module>
    sys.exit(main(sys.argv))
  File "ota_server.py", line 59, in main
    return serve(args[1], args[2])
  File "ota_server.py", line 19, in serve
    print >>sys.stderr, 'starting up on %s port %s' % server_address
TypeError: unsupported operand type(s) for >>: 'builtin_function_or_method' and
'_io.TextIOWrapper'

igrr, 2015-06-15:
I think this print statement stuff will not work with python 3. I've written this script with 2.7.

Pablotis, 2015-06-15:
OK thanks, on Ubuntu iḿ using 2.7.9, I'll see on Windows

igrr, 2015-06-15:
Regarding the bin file, there is now only one .bin file generated (like ```Blink.cpp.bin```)
You need to use either git version or this package:
http://arduino.esp8266.com/staging/package_esp8266com_index.json

Pablotis, 2015-06-16:
I cannot update:
socket.error: [Errno 10054] Se ha forzado la interrupci¾n de una conexi¾n existente por el host remoto

igrr, 2015-06-16:
You have the OTATest.ino sketch loaded?

Pablotis, 2015-06-16:
Yes, it shows Update failed

igrr, 2015-06-16:
Can you please post complete output from both sides?
Also it would be helpful if you would mention what kind of board you are using, what is the flash size, what is the size of the sketch you are trying to load via OTA. Thanks.

olivernz, 2015-06-16:
This is way cool! I have ESP's that are built into walls behind switches
and upgrading is a....
So this'd be great!

Duality4Y, 2015-06-17:
this seems to work here :)
that it resets the module (wdt) is what is supposed to happen after it is finished uploading ?

Pablotis, 2015-06-17:
Here is the log and the bin file:
https://drive.google.com/folderview?id=0BxPwW0n9dYaFfkVFTy1tUG96QkVCQi1xSC1YdEJ2d2o3Xy1KZ09wbFpoRjdPQnBHVWhVSG8&usp=sharing

Pablotis, 2015-06-18:
Any new?

igrr, 2015-06-18:
From your log:
```
Free size: 184320
...
upload size: 262416
```
It fails because free space size is less than the size of the sketch you are trying to upload.

Pablotis, 2015-06-18:
But I tried to do it with Blink example and still not working

igrr, 2015-06-18:
In that case please attach the log for the case when you are downloading the blink example. It's pretty hard trying to guess what's happening on your side :)

chadouming, 2015-06-18:
I'm pretty sure that even using a blank sketch won't work with 512KB space,
since blank sketch and library takes roughly 200K

thewiep, 2015-06-18:
I was having, what seems to be, the same issue and I suspect it has something to do with my ESP module or flakey wifi signal.
I frequently have issues getting the ESP into flashing mode.

this is the output in my terminal:
```
root@desktop:/tmp# python ota.py 192.168.1.119 ESP8266_Remote_Sketch_Upload_OTA.cpp.bin 
starting up on 0.0.0.0 port 48266
upload size: 253280
sending invitation
waiting for a connection
connection from ('192.168.1.119', 4097)
opening file ESP8266_Remote_Sketch_Upload_OTA.cpp.bin
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
Traceback (most recent call last):
  File "ota.py", line 64, in <module>
    sys.exit(main(sys.argv))
  File "ota.py", line 59, in main
    return serve(args[1], args[2])
  File "ota.py", line 48, in serve
    connection.sendall(chunk)
  File "/usr/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 32] Broken pipe
```
```
root@desktop:/tmp# ls -lah | grep ESP8266_Remote_Sketch_Upload_OTA.cpp.bin 
-rw-r--r--  1 user   user   248K jun 18 17:50 ESP8266_Remote_Sketch_Upload_OTA.cpp.bin
```
```
root@desktop:/tmp# python --version
Python 2.7.6
```
this is the output on the ESP:
```
OTA test 3
scandone
f 0, ....scandone
.add 0
aid 4
pm open phy_2,type:2 0 0
cnt 

connected with cybertron_main_g, channel 6
dhcp client start...
.ip:192.168.1.119,mask:255.255.255.0,gw:192.168.1.1
.
Connected to cybertron_main_g
IP address: 192.168.1.119
Sketch size: 253268
Free size: 794624
Got packet
0 48266 253280
Update failed
```
when it works I get following output in terminal:
```
...
sending 4096
sending 4096
sending 4096
sending 3424
done!
```
and this from the ESP:
```
OTA test 3
scandone
f 0, scandone
.add 0
aid 4
pm open phy_2,type:2 0 0
cnt 

connected with cybertron_main_g, channel 6
dhcp client start...
.ip:192.168.1.119,mask:255.255.255.0,gw:192.168.1.1
.
Connected to cybertron_main_g
IP address: 192.168.1.119
Sketch size: 253268
Free size: 794624
Got packet
0 48266 253280
rm 0
pm close 7 0 0/7924696
Fatal exception (28):
epc1=0x402029ce, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1464, room 16 
tail 8
chksum 0x7a
csum 0x7a
@
cp
��ՁMEM CHECK FAIL!!!
�rl
OTA test 3
...
```
the Fatal exception seems scary but seems to indicated the update succeeded..

This is a really nice feature!

CptanPanic, 2015-06-20:
Wow this could be a great feature that I will need to try. Does this utilize http://www.esp8266.com/viewtopic.php?f=12&t=3115&start=10 ?

igrr, 2015-06-20:
@CptanPanic i wrote my own bootloader, called eboot. It's a bit more efficient because it requires only one segment for code and data, plus it avoids an extra copying step which both rboot and Espressif's original bootloader choose to do for some reason. Also the code is slimmer and a bit easier to digest. You can check the source code in hardware/esp8266com/esp8266/bootloaders/eboot.

prussiap, 2015-06-21:
@igrr I saw it in the git repo but how do you make use of it? Was there an example or samples in the wiki? I saw another post you had about using the ota_server.py. Do you flash the eboot bootloader then use the ota_server.py ipaddress any_arduino_sketch  method ?

igrr, 2015-06-21:
The binary which is generated by IDE contains the bootloader in the first
4kb sector, so you don't need anything extra — the bootloader will be
flashed along with the sketch.
To use OTA you can flash OTATest.ino linked above, and then use
ota_server.py to upload the new sketch.

In a short while this will be integrated into the IDE so you will just have
to hit "upload" button for OTA to work.

duncan-a, 2015-06-21:
This is a wonderful project - and it just keeps getting better and better...

Thank you everyone who is involved..................

probonopd, 2015-06-21:
I get on the ESP-01:
```
OTA test
scandone
f 0, ....scandone
.add 0
aid 3
pm open phy_2,type:2 0 0
cnt 

connected with h48, channel 6
dhcp client start...
...ip:192.168.0.16,mask:255.255.255.0,gw:192.168.0.1
.
Connected to h48
IP address: 192.168.0.16
Sketch size: 248580
Free size: 188416
Got packet
0 48266 248592
Update failed
```
And on the Mac:
```
python /Users/user/Downloads/ota_server.py 192.168.0.16 OTAtest.cpp.bin
starting up on 0.0.0.0 port 48266
upload size: 248592
sending invitation
waiting for a connection
connection from ('192.168.0.16', 4097)
opening file OTAtest.cpp.bin
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
Traceback (most recent call last):
  File "/Users/user/Downloads/ota_server.py", line 64, in <module>
    sys.exit(main(sys.argv))
  File "/Users/user/Downloads/ota_server.py", line 59, in main
    return serve(args[1], args[2])
  File "/Users/user/Downloads/ota_server.py", line 48, in serve
    connection.sendall(chunk)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 32] Broken pipe
```

igrr, 2015-06-21:
Free size: 188416
upload size: 248592

You need to have at least 1M flash for the OTA to fit.

probonopd, 2015-06-21:
This means I need a different hardware module, correct? So OTA will not work on ESP-01? (If so it would be good to document that fact or put in a check). Also, what is not clear to me: When I upload OTAtest, does that mean I can upload blink.cpp.bin OTA and it will still be possible to load another sketch via OTA afterwards, or does every sketch that I upload via OTA need to contain the OTAtest functionality (in order to allow for subsequent OTA updates)?

igrr, 2015-06-21:
This is a very much work-in-progress feature so please forgive the lack of
documentation. You need at least 1M flash for this to work.
Many ESP12/ESP07 modules right now ship with 4M flash.
For now the OTA function is implemented in the sketch, so if you upload
Blink sketch, it won't be able to update. I'm open to suggestions how to
handle this better though.

One more thing that you will certainly need in a sketch in order for OTA to
be possible is ESP8266WiFi library.

probonopd, 2015-06-21:
Thanks. That makes it clear. So what'd be really cool is if the OTA updating could be abstracted away inside the libraries and added "behind the scenes" so that it doesn't need to "live" in each sketch.

prussiap, 2015-06-22:
@igrr thanks for all this info. Is there a discussion group somewhere like on gitter to learn more and participate? I'd like to see if I can help with Arduino Ide or getting OTA built into a bootloader/firmware?

For now to recap it seems like the OTA will only work with bigger modules with 1M plus flash like thE esp-12 and potentially the esp-201. Also you need to upload a sketch with ip address (wifi info) etc on the device. Then you can use the ota-server.py script with whatever sketch you want to OTA. You can only really use it once since after the upload you've overwritten the initial sketch?

It seems like maybe at some point we'll be able to put a firmware on the esp-12 that will act like a wifi AP (have the equivalent of the OTA sketch in it) or have a way to connect to your wifi but then be able to change the ip info through the ide. The Arduino IDE will either have an ip set for the esp so it knows where to send the sketch or perhaps you'll be able to connect to the esp AP and update the sketch?

I guess i'm grasping a bit but i'd like to find out more about he firmware/bootloaders to see if we can't put some of these features into them instead of on a sketch?

igrr, 2015-06-22:
@prussiap there is a gitter chat at https://gitter.im/esp8266/Arduino

The idea is to do no changes on the IDE side, that is, integrate with the existing code written for Arduino Yun. IDE will find all Yun's (or any other network-enabled Arduino boards for that matter) on the LAN using DNS-SD protocol, and add their IP addresses into the Ports menu. When we select an IP address and invoke upload, Arduino will run upload tool and pass it the port name (which in this case will look like "192.168.1.100"). So our esptool will have to take this "port name", recognize that this is an IP address, and start upload over TCP/IP instead of serial. On the sketch side, some always-on part of ESP8266WiFi library will listen for incoming uploads and flash new sketch when IDE sends it.
The largest piece which is missing is actually DNS-SD. Everything else is more or less ready.

prussiap, 2015-06-22:
Cool I'll try to join as soon as I can..

Thank you for the added info. Does this mean though that OTA will always be just a one time affair then since it will need a special  ESP8266WiFi sketch on the esp? Is it impossible to put a firmware/bootload something on the esp that can always accept a sketch?

Toshik, 2015-06-22:
Hello. I have ran into issue: OTAServer sketch uploades and works. When I use Python script to upload main sketch - it uploads and ends with success on PC but at ESP side:

```
Sketch size: 248712
Free size: 712704
Got packet
0 48266 268544
rm 0
pm close 7 0 0/24263998
Fatal exception (28):
epc1=0x4020442a, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (28):
epc1=0x4020442a epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffea170 end: 3ffea500 offset: 01a0

>>>stack>>>
3ffea310:  00000000 3fff3a58 3ffe9210 4020439d  
3ffea320:  3fff26c4 3ffeb1bc 3fff4030 40202546  
3ffea330:  40228b47 000000a0 3fff43e8 40228b57  
3ffea340:  00000001 676e6173 7365725f 74636570  
3ffea350:  00000000 00000000 00000000 00000000  
3ffea360:  00000000 b022000c 088d695a 3ffe94e0  
3ffea370:  402108c2 402113fe 3fff4030 3fff24cc  
3ffea380:  0003d000 402113c4 3fff4030 3fff4dd8  
3ffea390:  40229538 3fff44d8 00000001 0003d000  
3ffea3a0:  4020fd7f 3fff44d8 3fffc718 0007e900  
3ffea3b0:  00000900 00000008 40204af2 3fff44d8  
3ffea3c0:  00041900 0003d000 00000900 40204b25  
3ffea3d0:  00000001 3fff44d8 3fff44d8 40204b22  
3ffea3e0:  eb001000 00000001 0003d000 00000000  
3ffea3f0:  00041900 00000005 3ffea45f 40205e54  
3ffea400:  3fff3a58 3fff3ce0 000000ff 00000000  
3ffea410:  00000006 40101425 3fff3ce0 00000018  
3ffea420:  40106080 3fff3ce0 3fff43e8 0000ff00  
3ffea430:  3fff1268 40214f03 00000000 00000000  
3ffea440:  4020fd29 3fff3a30 3fff3ce0 3ffea490  
3ffea450:  402044ec 3fff39b0 402017fe 068592cf  
3ffea460:  3ffea490 00000900 0007e000 00000900  
3ffea470:  5b6410ac 3ffea560 3ffe9210 0000bc8a  
3ffea480:  00041900 3ffea560 3ffe9210 40202418  
3ffea490:  3ffe8c80 00000000 000003e8 00006db5  
3ffea4a0:  00000000 00000000 3ffe8e70 5b6410ac  
3ffea4b0:  3ffe8e70 5b6410ac 3ffea560 40202364  
3ffea4c0:  00000000 00000000 00000016 40101f05  
3ffea4d0:  4020181d 00000000 00000000 3ffea52c  
3ffea4e0:  3fffdc20 00000000 3ffea524 4020187a  
3ffea4f0:  00000000 00000000 3ffe94e0 40100450  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1464, room 16 
tail 8
chksum 0x7a
csum 0x7a
@
cp

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
ets_main.c 
```

after that - only normal reflash helps

Links2004, 2015-06-22:
there are some problems with the restart function.
patch: https://github.com/Links2004/Arduino/commit/46b53e84fea443cd13dfeb0426de71d2dad60bbd
at my branch there is a HTTP update variant for updating.
https://github.com/Links2004/Arduino/tree/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266httpUpdate/src
there also is a patch for checking the magic byte at the bin file so you not upload a wrong file.
https://github.com/Links2004/Arduino/commit/c0e5b5407398610210afc41367705199a2377764

Toshik, 2015-06-23:
@Links2004 that did not helps

sticilface, 2015-06-25:
Is there a way of enabling more detailed debug information from the OTA update process.  

Mine works, it is just very unreliable.  Working maybe 1 in 5 attempts.  

I'm using 4M esp 12
```
Sketch size: 309892
Free size: 737280
```

I get a very similar error to the guys above.. it reached the end and fails... (i've put a reset in after so that I can attempt is again.  Im not sure if the LmacRx bit is because of the reset or why it has failed. 
```
Got packet
0 48266 309904
Update failed
LmacRxBlk:1
```  

on the computer end 

```
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
sending 4096
Traceback (most recent call last):
  File "ota_server.py", line 64, in <module>
    sys.exit(main(sys.argv))
  File "ota_server.py", line 59, in main
    return serve(args[1], args[2])
  File "ota_server.py", line 48, in serve
    connection.sendall(chunk)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
socket.error: [Errno 32] Broken pipe
```

It just doesn't report back with done. 

This is using my sketch, not the OTA update sketch...

sticilface, 2015-06-25:
I was also thinking about the update abstraction... using this way with some code in the sketch you've got some control over monitoring the process.  for example, i turn the LEDs green when the packet is received... they go red if it fails, then it reboots, so one success they go green, then off... fail = green--> red --> reboot.  

if it is all abstracted then you won't be able to do this.  also helps make sure you are flashing the right ESP.

sticilface, 2015-06-25:
I had a successful upload, followed by 5 failures and thought I'd look at where it is failing.  Every time, it is with the 35th packet....  when it succeeded it went through to the 76th.

sticilface, 2015-06-25:
enabling debug output
```
Got packet
0 48266 310464
usedSize=310452 freeSpaceStart=311296 freeSpaceEnd=1048576
erase @0x4c000 size=0x4c000
erase done
writing
.stream read failed
Update failed

 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
```

sticilface, 2015-06-25:
by chance i put a delay(1000) in after the UDP close and before the update is called... and it has worked 10 times in a row...

```
    listener.stop();

    delay(1000);

    if (!ESP.updateSketch(cl, sz)) {..
```

hallard, 2015-06-29:
@Links2004 
just saw you httpupdate has been merged, great !!!!! you debugged on Serial1, love this I'm always using this one also ;-)

Okay, now, would it be possible to have a sample code or documentation to do on esp8266 side ?

Tell me if I'm wrong, I'm just trying to understand, esp client is responsible to call myhttpupdate.update(blah blah, ...) whenever it would like an upgrade. Until there I think I got it. But now, how the client knows there is a new version ? where do I deal with the version in my sketch (or .bin) ?

Thanks for your nice job, was waiting for OTA from Arduino, you're so close.

PS : I also tried @igrr update with python, it works but TCP is less stressfull for me ;-)

igrr, 2015-06-29:
@hallard i used UDP only for the "advertisement" stage. firmware was transferred over TCP as well.
DNS-SD support from @me-no-dev has also been merged in, so we're just one step away from integrating everything well into the IDE, with the upload button uploading the sketch over OTA.

hallard, 2015-06-29:
@igrr  
great news for TCP, does a check is done to be sure data is not corrupted before flashing ?
Yeah cool for IDE, in fact you know I can deal without, I click on compile button then I do an up arrow on my command line (after typed the flash command the 1st time) and it's done, and it's so much faster than dealing with upload speed  that works when it wants at speed above 115200 ;-)

what is your vision to select which ESP module to flash in Arduino IDE ? like this
[image]

Charles

igrr, 2015-06-29:
No check is done at the moment (although adding such a check is trivial).
Right, the esp will be listed next to the Yuns in ports menu.

Links2004, 2015-06-29:
@hallard all checks are optional and be done on server side.
the simplest use case is:
```cpp
ESPhttpUpdate.update("serverip/DNS", 80, "/esp.bin");
```
if you call the line it will update you ESP.
you can host the file at any HTTP server you like.

in a more extended case you can do:
```cpp
ESPhttpUpdate.update("serverip/DNS", 80, "/esp.bin","VERSION");
```
then the server get the Version in the HTTP header.
If the URL point to a script you can now handle the "Update request".
a HTTP answer with "200" will start the update with the data you script sends (the bin file) and "304" will mean no update need. all other codes are handled as error.

hallard, 2015-06-29:
@Links2004 
thank for the information, I'm not sure to be able to put in place the version option, may be I'll do some test with a basic PHP file or nodejs script when I'll gee time. Luckilly for now @igrr python script rocks for me ;-)
thx you both

EUA, 2015-07-08:
It works! No, it doesn't. 
At least with big sketchs.
OTATest.ino works. I can burn it multiple times.
But if I burn my another sketch which is 256,368 bytes (device has 4Mflash) it doesn't working.
Flashing look OK, at least server thing it's OK but device not working...

```
Got packet
0 48266 260512
rm 0
pm close 7 0 0/38947060
del if0
usl
sul 0 0

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1464, room 16 
tail 8
chksum 0x7a
csum 0x7a
@
cp

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
```

OTA is really too important feature for most of us.
Do you have any idea about whats gonna wrong? :(

igrr, 2015-07-08:
There was a bug which caused new sketch to be corrupted when it was larger
than the existing one. It was fixed couple of days ago but the boards
manager package has not been updated yet.

EUA, 2015-07-08:
When will it be updated in package manager? Also do I need to do anything to have that update?
I am suspicious if it will fix the problem. Because exactly same (the big one) sketch doesn't work If I flash it from OTA.
Any more details:
Big sketch -> flash small -> Works!
Small sketch -> flash big -> Doesn't. -> Stuck at boot
Big sketch -> flash same -> Doesn't. -> Gives boot loop (error log on bottom)
```
load 0x4010f000, len 1464, room 16 
tail 8
chksum 0x7a
csum 0x7a
esystem param error
4d 0a 16 76 02 72 0d 04 70 00 64...(too many of them)
```
Thanks

igrr, 2015-07-08:
There are a few major issues left currently (one with OTA, another one with heap corruption), i can't really give you an estimate when these will be resolved.

Regarding the boot loop, can you provide the sketch which triggers this issue?

EUA, 2015-07-08:
I think it's false positive. Now big->same flashing works proper (changed module & breadboard). Just small->big OTA issue remaining (which is already solved :+1 ). Just waiting the boards manager package update (or if available any patch file that can apply immediately, I like to try.)

hallard, 2015-07-08:
@EUA 
what version are you trying ?
Yesterday I've tested the lasted (which change the way to handle OTA in main loop), got some minors problems but it works with modified python uploader. check #517

Toshik, 2015-07-08:
Using latest I cannot upload by OTA any of sketches, even the same OTA test sketch:

```
rduino OTA Test
Sketch size: 304316
Free size: 741376
IP address: 172.16.100.234
Update Start: ip:172.16.100.91, port:48266, size:304320
Fatal exception (0):
epc1=0x40235b3e, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (0):
epc1=0x40235b3e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffdd0 end: 3fffffb0 offset: 01a0

>>>stack>>>
3fffff70:  40216f81 3fff1828 3ffed9bc 000000f5  
3fffff80:  3ffebed0 00000000 00000002 000002a8  
3fffff90:  40216968 3fffdab0 00000000 3fffdcc0  
3fffffa0:  3ffebed0 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)
```

hallard, 2015-07-08:
@Toshik 
Yeah, OTA management has changed since fex days, not sure all is compatible, so may be you need to update also your sketch (the one currently running onto ESP) to be compliant to new version. 
If you can't have serial access to your ESP (like me) what I've done so far:

* Modify old sketch from IDE to incorporate new OTA 
* OTA the new compiled sketch with old python script (because your ESP target has old version)
* verify OTA was well performed
* then try OTA with new python script version (because your uploaded sketch is now updated to new OTA)

Toshik, 2015-07-08:
@hallard Thanks, but I have used all "brand new" :)

* Sketch from: 
```hardware/esp8266com/esp8266/libraries/ESP8266mDNS/examples/DNS_SD_Arduino_OTA/DNS_SD_Arduino_OTA.ino ```
* And uploader from: 
```hardware/esp8266com/esp8266/tools/espota.py ```

hallard, 2015-07-08:
@Toshik 
Great, so it should be working !!
That's strange, never saw the fatal exception one (and I saw much), just to be sure, do you have a 4Mb Flash ESP board ? Because OTA won't work with 512Kb !!
Do you have a good Wifi con between ESP and your AP ?

Toshik, 2015-07-08:
@hallard 
Yes, I have 4Mb.
It breaks before all data is transferred into ESP, at different stages ~70Kb transferred

Previous version of OTA at least worked for the same OTA sketch

hallard, 2015-07-08:
Did you tried to reduce chunk size and add delay in python uploader like in #517 ?
this solved 75% of my failures, and I'm uploading 310Kb sketch

[118 more replies]

closed 2015-12-13 by igrr