esp8266/Arduino #269: Linux 32-bit and ARM

majenkotech, 2015-05-18:
I have compiled a Linux 32-bit toolchain (since I couldn't find it in the build.xml, nor guess the URL).  

Download: http://cdn.majenko.co.uk/esp/xtensa-lx106-elf-linux32.tgz

ARMhf version is compiling at the moment.  I'll provide a link for that too when it's done and packaged.

Linux 32-bit esptool is up as well: http://cdn.majenko.co.uk/esp/esptool-0.4.3-linux32.zip

Again I'll get an ARM version of that up there soon as well.

igrr, 2015-05-19:
Huge thanks for the toolchain! Did you build it using crosstool-NG? If so, which git revision was it?

igrr, 2015-05-19:
Also, why is esptool in a zip archive? I thought I changed ```make dist``` to generate .tar.gz archive for Linux and Mac OS...

igrr, 2015-05-19:
I have added the 32-bit tools to the board manager package.

majenkotech, 2015-05-19:
Yes, crosstool-NG from https://github.com/jcmvbkbc/crosstool-NG - the HEAD of the lx106-g++ branch

I manually packaged esptool as .zip because the 0.4.3 version is packaged as .zip, and my UECIDE package build scripts expect it to be .zip.  I can do a .tgz for you as well if you prefer.

By the way, the ARM compiler is nearly done - 871 minutes compiling so far and it's on cross-gdb.

majenkotech, 2015-05-19:
Actually, "make dist" makes a zip file.  The repo I am using is https://github.com/igrr/esptool-ck

igrr, 2015-05-19:
Actually it should be tar.gz (I made that change yesterday): 
https://github.com/igrr/esptool-ck/commit/d564b01c62c458c3a22f3268270485235d15ea7c

I switched to tar.gz because Arduino Boards Manager didn't preserve permissions for executable files inside .zip files, switching to tar.gz did the trick.

Regarding ARM — do you know if there is a version of Arduino for ARM?

majenkotech, 2015-05-19:
Ah - my linux32 repo pre-dated that change (pulled, and built now - http://cdn.majenko.co.uk/esp/esptool-0.4.4-linux32.tar.gz ).  My ARM copy of the repo was new enough, but it can't identify that it's linux, so it builds it as a .zip.

    matt@armadillian:~/esptool-ck$ uname -a
    Linux armadillian 3.18.11+ #15 PREEMPT Sun May 3 17:18:11 BST 2015 armv6l GNU/Linux
    matt@armadillian:~/esptool-ck$ uname -p
    unknown 
    matt@armadillian:~/esptool-ck$ uname -m
    armv6l

I'll manually package it for now.
 
And yes, Arduino is available for ARM.  Only seems to be 1.0.5 in the default repository though, but no reason why you can't use any other Linux download of Arduino - it's Java after all.

igrr, 2015-05-19:
does ```uname -s``` tell that it's Linux?

majenkotech, 2015-05-19:
Better still, there's a fix for ARM.  Also a package: http://cdn.majenko.co.uk/esp/esptool-0.4.4-linux-armhf.tar.gz

majenkotech, 2015-05-19:
By the way - `uname -m` is probably the better way of detecting what the architecture is:

    Linuxx 64 = x86_64
    Linux 32 = i686
    Linux ARM = armv6l

igrr, 2015-05-19:
If you can change the makefile to use ```uname -m``` and test on 86 and ARM, then please do add this to your pull request.

majenkotech, 2015-05-19:
Works nicely - just built packages on all three platforms as Linux in tar.gz

majenkotech, 2015-05-19:
ARM compiler now up: http://cdn.majenko.co.uk/esp/xtensa-lx106-elf-linux-arm.tgz

igrr, 2015-05-19:
```
$ wget http://cdn.majenko.co.uk/esp/xtensa-lx106-elf-linux-arm.tgz
--2015-05-19 14:43:59--  http://cdn.majenko.co.uk/esp/xtensa-lx106-elf-linux-arm.tgz
Resolving cdn.majenko.co.uk (cdn.majenko.co.uk)... 31.193.139.41, 31.193.139.48
Connecting to cdn.majenko.co.uk (cdn.majenko.co.uk)|31.193.139.41|:80... connected.
HTTP request sent, awaiting response... 404 Not Found
2015-05-19 14:43:59 ERROR 404: Not Found.
```

majenkotech, 2015-05-19:
Press F5 and use the real URL ;) You grabbed that in the 3 seconds it was wrong.

ficeto, 2015-05-22:
cudos!!! Now I can run this on all kinds of PIs :)

majenkotech, 2015-05-22:
It's fully supported by UECIDE on the Pi now.

igrr, 2015-05-29:
@majenkotech Could you please build the new esptool version for linux x86?

majenkotech, 2015-05-29:
* http://cdn.majenko.co.uk/esp/esptool-0.4.5-linux32.tar.gz
* http://cdn.majenko.co.uk/esp/esptool-0.4.5-linux-armhf.tar.gz

danbicks, 2015-05-30:
Hi igrr,

I know it is slightly of topic, pretty please can you update the build so it uses the latest SDK 1.1.0 really would be fantastic if we could use the rf power feature to save some battery life.
Amazing work you guys are doing.

Thanks Dans

igrr, 2015-05-30:
The version based on 1.1 is not very stable yet, but if you wish to try it
there is a staging package which gets updated more frequently:
http://arduino.esp8266.com/staging/package_esp8266com_index.json

danbicks, 2015-05-30:
Thanks Igrr,

I will give this a try and let you know.

Thanks buddy Dans

danbicks, 2015-05-31:
Igrr,

Thanks buddy I have downloaded the staging version and  the sketch with the latest version is now 67% FROM 46% but does recognize the rf power API call : system_phy_set_max_tpw(VarPWR)
The sketch remains the same quite a big difference in code usage using the staging variant, will this come down in the final build?

Also and I know slightly of subject for this forum but in the later release is there any chance of separating web pages html from the actual code files. You may have a method already in mind for this. A lot of is on the forum would really like this as an option if possible.

I have lots of modules here in my lab, mainly ESP01 and ESP12 I have the E version of the 12 on its way so would be more than happy to be a human Ginny pig for you guys :)

Cheers again

Dans

igrr, 2016-02-29:
@majenkotech how did you go about building ARM versions of the tools? I sorted out linux32 versions by running a 32-bit linux system with Vagrant and virtualbox, but not sure how to go about building the same for ARM. Could you please give me a hint?

majenkotech, 2016-02-29:
IIRC I may have done it on an arm system.

igrr, 2016-04-14:
ARM tools are now available in git version (via tools/get.py).

closed 2016-04-18 by igrr