esp8266/Arduino #270: can not compile

hephaestus9, 2015-05-18:
This is the error I get when compiling the AdvancedWebServer sketch (or any other with the wifi imports)

Arduino: 1.6.5 (Linux), Board: "Generic ESP8266 Module, 80 MHz, 57600, 512K (64K SPIFFS)"

Build options changed, rebuilding all
Using library ESP8266WiFi in folder: /Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266WiFi 
Using library ESP8266WebServer in folder: /Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266WebServer 
Using library ESP8266mDNS in folder: /Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266mDNS (legacy)

/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/tools/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/cores/esp8266 -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/variants/generic -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266mDNS /Arduino/HEX/AdvancedWebServer.cpp -o /Arduino/HEX/AdvancedWebServer.cpp.o 
/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/tools/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/cores/esp8266 -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/variants/generic -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266WebServer/src -I/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266mDNS /Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp -o /Arduino/HEX/ESP8266WiFi/ESP8266WiFiMulti.cpp.o 
In file included from /Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:24:0,
                 from /Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32,
                 from /Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.h:30,
                 from /Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp:26:
/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_vector.h: In member function 'std::vector<_Tp, _Alloc>::size_type std::vector<_Tp, _Alloc>::_M_check_len(std::vector<_Tp, _Alloc>::size_type, const char*) const':
/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:217:18: error: expected unqualified-id before '(' token
 #define max(a,b) ((a)>(b)?(a):(b))
                  ^
/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/xtensa-lx106-elf/include/c++/4.8.2/bits/stl_bvector.h: In member function 'std::vector<bool, _Alloc>::size_type std::vector<bool, _Alloc>::_M_check_len(std::vector<bool, _Alloc>::size_type, const char*) const':
/Arduino-ESP/build/linux/work/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:217:18: error: expected unqualified-id before '(' token
 #define max(a,b) ((a)>(b)?(a):(b))
                  ^
Error compiling.

hephaestus9, 2015-05-19:
Saw there was an update, fresh download and build and it worked.

closed 2015-05-19 by igrr

andyjwang, 2015-05-24:
Please tell me what kind of update you mentioned. Thanks.