esp8266/Arduino #271: Consider adding my Adafruit 1.8" LCD support.

nzmichaelh, 2015-05-19:
Hi there.  I ported the Adafruit 1.8" LCD library to your excellent ESP8266 Arduino support.  The code is available here:

https://github.com/nzmichaelh/Adafruit-ST7735-Library

with performance numbers here:

https://juju.net.nz/michaelh/blog/2015/05/1-8-lcd-support-for-the-esp8266/

I've sent a pull request to upstream, but Adafruit don't seem to actively merge improvements.

igrr, 2015-05-22:
Hi, thanks for the great work on that library! Unfortunately I don't have that lcd myself, but will definitely add the library to the bundle.

duncan-a, 2015-05-22:
Does this mean that the Adafruit_ST7735 *and* Adafruit_GFX library are
ported to the ESP8266? Both are required according to the Adafruit Learning
page.

Is anybody working on the larger displays?

nzmichaelh, 2015-05-22:
I didn't have to change the GFX library, just the driver.

duncan-a, 2015-05-22:
OK, thanks for that - and for porting the driver

Makuna, 2015-07-16:
I can confirm this works, I added a reference for your library to the readme.

hruskozrout, 2015-08-14:
Hi,
anyone get this error while compiling?
``` 
Adafruit_ST7735.cpp:39:3: error: '_cs' was not declared in this scope
   _cs   = cs;
   ^
```

Im using this constructor:
```
#define TFT_CS     15
#define TFT_RST    5                       
#define TFT_DC     2
Adafruit_ST7735 tft = Adafruit_ST7735(TFT_CS, TFT_DC, TFT_RST);
```

Thanks for any hint.

closed 2015-11-06 by Links2004