esp8266/Arduino #273: Crash after 354 UDP packets sent

edufschmidt, 2015-05-19:
Hi,
I'm trying to stream data from the ESP8266 to a server using UDP, and for some reason there is a crash after I send 354 packets.
The code below is more or less what I'm using to transmit the packets:

void loop() {
 count++;
  if ( r = client.beginPacket("192.168.0.103", 7777) ){
    String str = String(count);
    client.write(str.c_str(), str.length());
    r = client.endPacket();
  }
  delay(50);
}

For testing purposes I'm just sending the value of a counter, which is incremented until 354. Right after the ESP sends this value, it crashes, and I get the following message on the UART:

ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 30984, room 16 
tail 8
chksum 0xa9
load 0x3ffe8000, len 2540, room 0 
tail 12
chksum 0x11
ho 0 tail 12 room 4
load 0x3ffe89f0, len 2156, room 12 
tail 0
chksum 0xfb
csum 0xfb

The number of packets sent before the crash is always the same, and does not depend on the data being transmitted.
Any help will be much appreciated!
Thanks in advance!

igrr, 2015-05-19:
I am not able to reproduce the issue based on your sample. Here's the full sketch that I use:
```C++
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

WiFiUDP client;

const char* ssid = "  ";
const char* password = "        ";

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  Serial.println("");
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  while ( WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  
  Serial.println("");
  Serial.print("IP: ");
  Serial.println(WiFi.localIP());
}

int count = 0;
void loop() {
  count++;
  int r;
  if ( r = client.beginPacket("192.168.0.103", 7777) ){
    String str = String(count);
    client.write(str.c_str(), str.length());
    r = client.endPacket();
  }
  Serial.printf("Count: %04d\tHeap: %d\r\n", count, ESP.getFreeHeap());
  delay(50);
}
```

And every time I run it, it goes far beyond 354 packets without any memory leaks:
```
Count: 0779	Heap: 31432
Count: 0780	Heap: 31432
Count: 0781	Heap: 31432
Count: 0782	Heap: 31432
Count: 0783	Heap: 31432
```

I tried this with the latest board manager package release.

edufschmidt, 2015-05-19:
My code looks almost exactly like yours, so I downloaded the latest Arduino version, installed the board manager package within the IDE, and now everything is working. Thank you for the prompt reply!

closed 2015-05-19 by edufschmidt