esp8266/Arduino #275: xtensa-lx106-elf-g++ returned 139 on MAC OSX

meg9468, 2015-05-19:
Hi,

I have tried to compile the IDE for my Mac OSX (because I have OSX ver 10.6.8 and the precompiled version was for 10.7 !).
I have performed the steps as described, and the arduino ver 1.6.5 was build without problem :-)

But when I compile an *.ino file for ESP8266 hardware, I get that error :

Arduino: 1.6.5 (Mac OS X), Board: "Generic ESP8266 Module, 80 MHz, 115200, 512K (64K SPIFFS)"

/Applications/Programmation/Arduino_1_6_5.app/Contents/Resources/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ returned 139
Error compiling.

????? 

What can I do ???? 

Many thanks
Greg

igrr, 2015-05-19:
Hi Greg,
just checked the toolchain binaries and they don't seem to be compatible with 10.6.5:
```
$ otool -l xtensa-lx106-elf-g++
....
     cmd LC_VERSION_MIN_MACOSX
  cmdsize 16
  version 10.10
      sdk 10.10
```
The toolchain needs to be rebuilt with an appropriate ```-mmacosx-version-min``` option. I'll do that, will upload the new toolchain.

claybar, 2015-05-20:
I am getting the same compile error, except with OSX 10.7.5.  I presume the `version 10.10` means it will currently only work on OSX 10.10 upwards.

I am really keen to get the Arduino platform going, I have been using NodeMCU with lua but it seems rather unstable and unpredictable.  Lua isn't exactly my favourite language either.

igrr, 2015-05-20:
Got the toolchain built already. Will upload today.

igrr, 2015-05-22:
Updated package has been deployed. Please try updating your boards manager package.
Note that there seems to be a bug in Arduino 1.6.4, in some cases it's not able to delete the tools directory when uninstalling the package. You may need to delete ```~/Library/Arduino15/packages/esp8266``` manually, close the IDE, and reinstall the package again, if you have troubles compiling.

igrr, 2015-05-23:
@meg9468 @claybar Did you have a chance to update and see if this solves the issue?

meg9468, 2015-05-23:
I Igor,

Thank you for your help.

My side, I get always the same error.
May be I am doing wrong... This is what I try (on Mac OS 10.6.8):
First, I have cleaned the arduino folder.
Then I git your directory,
Then, ant dist.  There I get no error
I then launch arduino IDE (1.6.5) and try to compile a small piece of code, and Java crash with the same error (xtensa-lx106-elf-g++ returned 139)

Thank you

Greg

igrr, 2015-05-23:
I added the new toolchain to the board manager package, so you don't need
to ant dist, just follow the instructions on the front page. Ant build
script hasn't been updated to use the new toolchain, sorry about that.

Ivan

meg9468, 2015-05-23:
I think I do something wrong....
I add "http://arduino.esp8266.com/package_esp8266com_index.json" then I go to the board manager and click on Install button next to "ESP8266",
But I get "Tool esptool is not available for your operating system"

and the trace :
 Tool esptool is not available for your operating system.
java.lang.RuntimeException: java.lang.Exception: Tool esptool is not available for your operating system.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$4.run(ContributionManagerUI.java:179)
	at java.lang.Thread.run(Thread.java:695)
Caused by: java.lang.Exception: Tool esptool is not available for your operating system.
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:84)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$4.run(ContributionManagerUI.java:173)
	... 1 more
 Tool esptool is not available for your operating system.
java.lang.RuntimeException: java.lang.Exception: Tool esptool is not available for your operating system.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$4.run(ContributionManagerUI.java:179)
	at java.lang.Thread.run(Thread.java:695)
Caused by: java.lang.Exception: Tool esptool is not available for your operating system.
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:84)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI$4.run(ContributionManagerUI.java:173)
	... 1 more

igrr, 2015-05-23:
I might have broken something. Will check in a few hours...

arthead, 2015-05-24:
Thanks for all your work igrr! I am having similar problems. I followed the directions on the front page to install and ended up getting the: xtensa-lx106-elf-g++ returned 139
Error compiling. I found this thread and deleted: package_esp8266com_index.json and tried to reinstall the ESP8266 boards manager in the first section of the front page but it still doesn't work. When I go to the boards manager and try to re-install esp8266 platform it says "Tool esptool is not available for your operating system" even though it let me earlier. If I try to compile WiFIWebServer I get a compiling error: "Tool esptool is not available" Should I try starting  from scratch and deleting the Arduino15 directory? I'm on a Mac OS 10.7.5 Thanks in advance and looking forward to digging in to the ESP8266!

claybar, 2015-05-24:
@igrr Working for me, but not directly as you predicted in one of your posts earlier.  The steps I took (may not all be needed)
- Uninstall old boards package from the Boards Manager
- Shutdown Arduino
- Delete directory `~/Library/Arduino15/packages/esp8266`
- Start Arduino, install new package from the Boards Manager
- Compile

Other info:
- MacBook Pro with OSX 10.7.5
- Arduino 1.6.4
- esp8266 package version 1.6.4-673-g8cd3697

arthead, 2015-05-24:
Still not working for me. I tried claybar's suggestions but when I look in my Arduino15 directory I do not have a packages or esp8266 directory. I guess because of the "Tool esptool is not available for your operating system" error i get in the Board Manager when trying to install.
• Macbook with 10.7.5
• Arduino 1.6.4

gerardwr, 2015-05-25:
After a Boards Manager "update" of the ESP8266 package, I also had the `xtensa-lx106-elf-g++` compile error.

I followed the steps from the message from @claybar and this fixes it. It compiles OK now, thanks.

I'm on Arduino 1.6.4 and Mac OSX 10.10.3.

arthead, 2015-05-29:
Any progress on the :"Tool esptool is not available for your operating system" error during Board Manager esp8266 package install?
• Macbook with 10.7.5
• Arduino 1.6.4

pbrs, 2015-06-05:
Got the same exact issue as described by @arthead. Running Mac OS 10.7.5 with 1.6.4 IDE and I'm getting "esptool is not avaliable for your operating system". Is there any fix for this?

meg9468, 2015-06-15:
Hi,

Still the problem :-( 
Has anyone a solution ?

Thanks

igrr, 2015-06-15:
Might be related: https://github.com/arduino/Arduino/commit/e646ca252583e1cddf679fa6208df28d5e204ace

Arduino 1.6.5 is out, can you try with it?

arthead, 2015-06-30:
I will try later as I haven't been able to use it yet.

bmweller, 2015-07-22:
I've tried with both stable and 1.6.5-939-g033463a and geting the same result on i386-apple-darwin 10.7.5.  That host key is missing from the list 'package_esp8266com_index.json' , hence I repeated the contents of x86_64-apple-darwin from the 'package_esp8266com_index.json' setting the host key to i386-apple-darwin. This allows the board manager to download and install the package into the user Library.  Compile works fine however esptool falls over seems it's only been compiled for x86_64 arch.

/Users/mac/Library/Arduino15/packages/esp8266/tools/esptool/0.4.5/esptool returned 139

igrr, 2015-07-22:
Oh, so not all Macs are 64-bit? That explains why this issue happens.
Thanks for the tip, will add `i386-apple-darwin` section to boards manager package.
Regarding esptool, could you please try the following binary and see if it works on your system?
http://arduino.esp8266.com/esptool-universal/esptool
Thanks.

bmweller, 2015-07-22:
Not so. The hardware is Core 2 Duo and capable of running 64bit however for reasons I guess of backward support and overall system performance apple held back on supporting this kit on 10.8 in the final release. 10.8 Mountain lion was the first full 64bit OS. 10.7 seems to include  64bit program addressiblity however the kernel and extensions ran in 32 bit mode.

http://apple.stackexchange.com/questions/12666/how-to-check-whether-my-intel-based-mac-is-32-bit-or-64-bit

So trying your universal binary:

./esptool 
Segmentation fault: 11

file ~/Downloads/esptool 
/Users/mac/Downloads/esptool: Mach-O universal binary with 2 architectures
/Users/mac/Downloads/esptool (for architecture i386):	Mach-O executable i386
/Users/mac/Downloads/esptool (for architecture x86_64):	Mach-O 64-bit executable x86_64

does the package in the esp8266/Arduino.git repo offer universal build support? I can try building on this host?

igrr, 2015-07-22:
esptool is built from this repository:
https://github.com/igrr/esptool-ck.git

I've pushed a change into `osx-universal` branch. You can try building it by running `make all`.

bmweller, 2015-07-22:
looks promising
./esptool 
error: No arguments given

bmweller, 2015-07-22:
Sweet success! Blinkly blue light on pin 2 of NodeMCU 1.0...
Now to compile an actual wifi sample!
BTW any view on the two changes being pushed up to staging? Will test the e-2-e when ready

Thanks for your help and all the great work you're doing here!

igrr, 2015-07-22:
Looks good. Now what's left is to figure out how to achieve the same on a 10.10.
I have already made the change related to `i386-apple-darwin` section, will push new staging package shortly.

pbrs, 2015-07-31:
I got the same problem getting 

/Users/mac/Library/Arduino15/packages/esp8266/tools/esptool/0.4.5/esptool returned 139

when I try to compile on 10.7.5

pbrs, 2015-07-31:
Worked it out after finding the right branch of esptool. Can also confirm it works now!

igrr, 2015-08-01:
I have merged osx-universal branch into master. However it appears that I'm still unable to build a binary on 10.10 which will work on 10.7.5, even with these changes. If there is anyone who runs 10.7.5 and can help debugging this issue (gdb or otherwise), I would appreciate some help.

meg9468, 2015-08-02:
Hello,

I am also unable unable to build the binaries :-(((
But I have a (very) old OS (10.6.8)

When O try to build, I get :

compile:
    [javac] Compiling 118 source files to /abc/temp/Arduino-esp8266/arduino-core/bin
    [javac] javac: invalid target release: 1.8
    [javac] Usage: javac <options> <source files>
    [javac] use -help for a list of possible options

BUILD FAILED
/users/abc/temp/Arduino-esp8266/build/build.xml:97: The following error occurred while executing this line:
/users/abc/temp/Arduino-esp8266/build/build.xml:119: The following error occurred while executing this line:
/users/abc/temp/Arduino-esp8266/arduino-core/build.xml:41: Compile failed; see the compiler error output for details.

where can I find the log ?

Another question, will it be an option to only compile without download (i.e. without using esptool) ???

thewiep, 2015-08-02:
you need to install JDK 8
see here: https://github.com/esp8266/Arduino/issues/631

sgerevini, 2015-08-20:
@igrr 

I am facing these issues ( 139 error when compiling ) too, and i can help to debug, but maybe need some directions.

I am using mac os 10.7.5 and esptool 0.4.3 works like a charm. I used it to reflash my esp board to nodemcu firmware, but 0.4.5 and 0.4.6 always return a "Illegal instruction: 4" error or 139 error under Arduino.

i have tried to replace esptool on Arduino/esp8266 folder, but all i got is another error (255):

[image]

Links2004, 2015-11-06:
please retry in last release, I it still a problem feel free to reopen.

agtabesh, 2015-12-13:
I install esp requirement and I have this problem too and got this error:
```
/Users/iC0d3R/Library/Arduino15/packages/esp8266/tools/esptool/0.4.6/esptool returned 139
Error compiling.
```
Mac OS X 10.7.5
Macbook Pro 2006 ( Core 2 Due 64bit )
Arduino 1.6.5
Please help

igrr, 2015-12-13:
Need to make new release of esptool which will include [the fix](https://github.com/igrr/esptool-ck/commit/36e2fd964b5ca467e8d261b60086ef123af1fc28).

agtabesh, 2015-12-13:
Thanks you but what I should do exactly?
How to change that file?

agtabesh, 2015-12-13:
I try to download and build by running:
```
./Makefile
```
but I get this error:
```
./Makefile: line 1: CFLAGS: command not found
./Makefile: line 2: CXXFLAGS: command not found
./Makefile: line 5: syntax error near unexpected token `$(OS),Windows_NT'
./Makefile: line 5: `ifeq ($(OS),Windows_NT)'
```

agtabesh, 2015-12-13:
Ok Its work!
Thank you
But I have a new problem, I miss esptool in Arduino > Tools > Programmer!
Please help

igrr, 2016-02-27:
esptool 0.4.7 has been released, includes 32-bit OS X compatibility.

closed 2016-02-27 by igrr

arthead, 2016-02-28:
Thank you so much, igrr! As of this evening it is now working on my Macbook. The WiFiWebServer sketch works as expected. Looking forward to getting the ESP-12 working next.
• Macbook late 2007
• Mac OS X 10.7.5
• Arduino 1.6.7
• ESP8266 original 8 pin

Thanks again for your hard work.