esp8266/Arduino #276: Problem with packet install

WoodsterDK, 2015-05-19:
Hi there,
the packet install in arduino has a problem downloading the esptool for windows...

igrr, 2015-05-19:
Can you please provide more context? Operating system, exact error messages, anything else that will help me help you solve this problem?

WoodsterDK, 2015-05-19:
Yes of course...
Win7 64bit - arduino 1.6.4
I press, boards -> Boards manager -> scroll down to "esp8266 by esp8266 community" and press install.
It begins downloading, but stops and says "error downloading https://github.com/igrr/esptool-ck/releases/download/0.4.4/esptool-0.4.4-win32.zip"....

Strange thing is that I can manually download the file using a browser....

igrr, 2015-05-19:
Strange indeed. Perhaps some files were only partially downloaded.
Can you please try deleting the archives found in ```%APPDATA%\Arduino15\staging\packages\``` and install again?

WoodsterDK, 2015-05-19:
The "esp8266-1.6.4-628-g545ffde.zip" is created and downloaded, but "esptool-0.4.4-win32.zip" only gets created, but with the size of 0, and then the installer fails.

igrr, 2015-05-19:
What happens if you download the file using the browser into this folder, and try installing the board again?

WoodsterDK, 2015-05-19:
It actually begins downloading the xtensa toolkit... so it seems to be progressing...It takes a while... I will be back... ;)

WoodsterDK, 2015-05-19:
The installation completes, and the board can be selected.
So the actual installation works - only the error with the esptool....

igrr, 2015-05-19:
Can you post the full error message? Does it happen when you verify the sketch or when you upload it?

WoodsterDK, 2015-05-19:
Sorry - it works now. The package has been installed, and everything seems to be working.
I was just referring to the fact, that the actual package install is broken.
If I manually copy the "esptool-0.4.4-win32.zip" into the %APPDATA%\Arduino15\staging\packages\ folder, and run the package install, then the installation completes.

igrr, 2015-05-19:
Okay, got it. Other users have reported successful installs on Windows so I don't know what to make of it. Perhaps running arduino_debug.exe can help (it shows some error messages on the command line).

WoodsterDK, 2015-05-19:
OK - I might make a fresh install later on, let's see how that goes....
Thanks for your help.
Have a nice day.

closed 2015-05-19 by WoodsterDK

PaulyLarry, 2016-07-12:
I encountered the same problem, on two separate machines, running win 7 64 bit OS.  The third machine I installed it on win 7 32 bit OS installed fine.  I also installed it on win xp sp3 and that was ok.  Any thoughts?

[image]