esp8266/Arduino #277: The read me instructions still wont work on 32 bit

tytower, 2015-05-19:
Install Arduino 1.6.4 from the Arduino website.
Start Arduino and open Perferences window.

Arduino will not start! 
You can go no further 
Same error reported

tytower, 2015-05-19:
Well done Igrr - Thats a big change isn't it . That board system works well

Small problem from your read me notes I was trying
Error on compile

/tmp/build6729920515858897918.tmp/Phant_EthernetOriginalMinewithPressureESP8266Version.cpp.o 
Phant_EthernetOriginalMinewithPressureESP8266Version.ino:43:1: error: 'Wire' does not name a type
'Wire' does not name a type

igrr, 2015-05-19:
Can you please post the sketch that triggers this problem?

tytower, 2015-05-19:
See edit above -thats the only change . Compiles without it

igrr, 2015-05-19:
Since you didn't post the complete sketch, I tried to compile this:
```C++
#include "Wire.h"

void setup() {
  Wire.begin(3,4);
}

void loop() {
}
```
and it compiles successfully.

tytower, 2015-05-19:
Yes ,too much,too late at night ,Thank You

closed 2015-05-19 by tytower