esp8266/Arduino #279: EEPROM unstable again

holgerlembke, 2015-05-19:
I really, really don't understand what goes wrong.

My simple test program write/reads the EEPROM and results are simply strange.... Program first:

#include <EEPROM.h>

void setup()
{
  // initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(5000);

  test();
}

void DumpEEpromdata(String msg) {
  EEPROM.begin(500);

  Serial.println();
  Serial.println(msg);
  Serial.println();
  for (int i = 0; i < 240; i++) {
    yield();
    byte p = EEPROM.read(i);

    if (p < 16) {
      Serial.print("0");
    }

    Serial.print(p, HEX);
    Serial.print(" ");

    if (i % 16 == 15) {
      Serial.println();
    }

    yield();
  }

  yield();
  EEPROM.end();
  Serial.println();
  Serial.println();
}

void test(void) {
  DumpEEpromdata("Test0");

  EEPROM.begin(500);
  for (int i=0;i<10;i++) {
    EEPROM.write(i+20,i);
  }
  EEPROM.end();

  DumpEEpromdata("Test1");
  
  EEPROM.begin(512);
  for (int i=0;i<10;i++) {
    Serial.print(EEPROM.read(i+20));
    Serial.print(" ");
  }
  EEPROM.end();
  
  DumpEEpromdata("Test2");

  EEPROM.begin(512);
  for (int i=0;i<10;i++) {
    Serial.print(EEPROM.read(i+20));
    Serial.print(" ");
  }
  EEPROM.end();
  
  DumpEEpromdata("Test3");
}

void loop()
{
  delay(100);
}

And here the result for the first two tests, only first lines displayed:

Test0

D6 4C D2 F2 FA 25 46 8C 75 B0 3B 70 9E CA B0 C3 
D6 A0 19 E3 15 DF 1F 76 8A 87 4E 7F FD B3 D4 03 

Test1

04 05 06 07 08 09 46 8C 75 B0 3B 70 9E CA B0 C3 
D6 A0 19 E3 15 DF 1F 76 8A 87 4E 7F FD B3 D4 03 

If I reboot I get:

Test0

D6 4C D2 F2 FA 25 46 8C 75 B0 3B 70 9E CA B0 C3 
D6 A0 19 E3 15 DF 1F 76 8A 87 4E 7F FD B3 D4 03 

Both results (first: wrong positions of bytes, second; data gone after reboot) is so wrong on so many levels... Am I crazy or made a stupid error?

holgerlembke, 2015-05-19:
Suggested change:

bool EEPROMClass::end()
{
    if (!_size) {
        return false;
	}

    bool ret = commit();
    if(_data) {
        delete[] _data;
    }
    _data = 0;
    _size = 0;
	
	return ret;
}

holgerlembke, 2015-05-19:
And to my problem:

It seems that

    if (spi_flash_erase_sector(CONFIG_SECTOR) == SPI_FLASH_RESULT_OK) {

never gives SPI_FLASH_RESULT_OK, so COMMIT is never written....

igrr, 2015-05-19:
Which flash size do you select from the Tools menu?

holgerlembke, 2015-05-19:
all. :-)

First the 4M that fits my esp-12, then rolled back to 512k (first on). Both don't work.

I just rolled back to old version, everything is good again. So I'm not insane... :-)

sticilface, 2015-05-19:
My EEPROM code no longer works.  (Using the build using boards manager, and the latest 1.6.4 i think arduino ide..) 

I've been using the same code for all versions, to write SSID, IPs for MQTT, device name, states of various variables etc and never had a problem.  This latest code saves nothing, every reboot it is gone. 

here is a dump of my EEPROM before saving anything

```````````````
HTTP/1.1 200 OK Content-Type: HTTP/1.1 200 OK Content-Type: HTTP/1.1 200 OK Content-Type:
	Ç										g	ÿ	?		é	Ý	ü	´	Ú	B	T	31	
H	u	b	5	-	R	G	N	K																							½	63	
								h	g	ÿ	?		À	o		±	0	T	A	L	K	T	A	L	K	-	0	8	B	1	2	95	
8																			§									°	g	ÿ	?	127	
b	é	Ý	ü	´	Ú	B	T	W	i	f	i	-	X																			159	
	º									ø	g	ÿ	?	,	9	–	y	“	g	B	T	W	i	f	i	191	
-	w	i	t	h	-	F	O	N																			¦					223	
@	h	ÿ	?	,	9	–	y	“	i	B	T	W	i	f	i	-	X											255	
	¨									ˆ	h	ÿ	?	,	9	–	y	287	
“	d	B	T	H	u	b	4	-	F	K	C	3																				319	
	¦									Ð	h	ÿ	?	|	L	¥	˜	d	å	S	K	Y	0	E	E	E	0			351	
¥									383	
À	?		¿	…	‘	V	M																					415	
¨									”	¶	ã	¥	ð	¿	ÿ	?	ˆ				447	
h							h	ÿ	?																					479	
				`	Å	G		C	F			511	
```````````````````

dump of my serial shows that the info has been received, and saved to EEPROM...
`````````````````````````
*SSID recieved....New SSID recieved: fyffest
Saving new... ..success...Check bit written
*Password recieved....New Password recieved: HELLOWWORLD
Saving new... ..success...Check bit written
*Deviceid recieved....New Deviceid recieved: TESTESP
Saving new... ..success...Check bit written
*MQTTServer recieved....192.168.1.24
MQTTserver IPAddress CHANGED: 192.168.1.24
SAVED MqqttServer: 192.168.1.24

Joining Wifi Network.......................................Failed.
Setting up Access Point....TESTESP
Disconnecting MQTT Client:..... 
Disconnecting MQTT wifiClient:..... 
Creating new MQTT Client:..... 
Calling pubsubclient:..... 
Initiating MQTT Connection: Connecting: (192.168.1.24)....Failed
``````````````````````

here is a dump of the EEPROM after save (interestingly it has in it contents of wifi networks surrounding me... this does not seem right... is this now heap space? my stuff is written to EEPROM, the ? are buffer characters i use to overwrite the old contents...

````````````````````
HTTP/1.1 200 OK Content-Type: HTTP/1.1 200 OK Content-Type: HTTP/1.1 200 OK Content-Type:
Ç	Ç	Ç	Ç	Ç						Ç										g	ÿ	?		é	Ý	ü	´	Ú	B	T	31	
T	E	S	T	E	S	P		?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?		63	
f	y	f	f	e	s	t		?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?		95	
H	E	L	L	O	W	W	O	R	L	D		?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?		127	
À	¨			´	Ú	B	T	W	i	f	i	-	X																			159	
	º									ø	g	ÿ	?	,	9	–	y	“	g	B	T	W	i	f	i	191	
-	w	i	t	h	-	F	O	N																			¦					223	
@	h	ÿ	?	,	9	–	y	“	i	B	T	W	i	f	i	-	X											255	
	¨									ˆ	h	ÿ	?	,	9	–	y	287	
“	d	B	T	H	u	b	4	-	F	K	C	3																				319	
	¦									Ð	h	ÿ	?	|	L	¥	˜	d	å	S	K	Y	0	E	E	E	0			351	
¥									383	
À	?		¿	…	‘	V	M																					415	
¨									”	¶	ã	¥	ð	¿	ÿ	?	ˆ				447	
h							h	ÿ	?																					479	
				`	Å	G		C	F			511	
`````````````````````````

here is a dump after reboot.... ok... it saved the details... now I'm confused... I did this twice and each time on reboot none of the saved info... name, access point, password were retrieved... 

now it has worked...

quite why the EEPROM is full of the access points from around me, is weird... i may have a bug somewhere... this is new behaviour though, did not see this with previous side versions.

sticilface, 2015-05-20:
I've made another discovery.  You need to power cycle the ESP..  it looks to me like things are over written on power cycle...  my esp gives me an EEPROM dump that looks ok, but reboot it by removing power, not just with the reset... and boom its all gone.  mine gets replaced by gibberish, and what look like variables from an array of nearby access points... 

heres what it looks like before... 

``````````````````
HTTP/1.1 200 OK Content-Type: HTTP/1.1 200 OK Content-Type: HTTP/1.1 200 OK Content-Type:
•	Ç	Ç	Ç	Ç	Ç	Å	U	u	t	Õ		Œ	¹	™	Ü	Ú		¡	H		S	®	¦		B	Ý	ú		3	²	£	31	
t	e	s	t	e	s	t	e	t	s	t		?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?		63	
f	y	f	f	e	s	t		?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?		95	
w	e	l	l	c	o	m	e	t	r	u	s	t		?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?	?		127	
À	¨			ÿ	ƒ	´	y	²	¾		à	N	b	D	\	×	f	&	Ó		–	ß	¥	ù		˜	€		0	O	j	159	
—	J	[	}	Ê		ã		s	ð	ó	‰	F	H	~	þ	=	×		Ð	ˆ		Ý	†	m	?	µ	Ü	Y	é	Ù	‚	191	
Ù	:	´	b	Ã		I		¸	Û	”	?	€	º	ß	i	³	¢		Ì	¦	^	¥	æ	Ø	´	‚	£	Ç	î	5	à	223	
·	M	4	¼	‘		ú	}	–	{		Î	ý	”	ý	ù	ë	f		=		P	¦		_	Þ		~	ä	å	®	¶	255	
ß	&	|	,	š	:	B	´	‹	ó	Ú		B	c	j	^	%	«		Ö	í		s	ã	Õ			>	Î	n	ý		287	
×	ª	@			à	/	y	Z	Ó	P	E		±	v	.	0	}	E	þ	é	G	û	M		C	f	ô	v	V	÷		319	
	*	ê		”	e	é	/	&	“	³	&	ù	³	˜	n	ú	ø	X	J	@	¬	x	Í	î	¼	'	i	 	ô		¨	351	
	'	x		2			í	e	ú	<		ï		—	4	Ã	w	Q	l	l		G	i		q	X	‚	¯	µ	œ	Û	383	
€	_		æ	¡		5	Ù	c		6	Á	4	ƒ	o	º	M	w	X	`			T		¿	Å				ž	¨	&	415	
G			¥	.	S		c	ú		>	¦	S	÷	‹	¬	r	ü	>	x	C	©	Ö	®	ž	Ê	Ì	Û	·	÷	§	{	447	
b	ï	d	y	x	O	½	#	œ	ç	î	J	°	J	g	<	{	Å	[	¸	À	Ð	î	'	/		ƒ		Í	Œ	Å	›	479	
$	‚		R	ì	v	ú	}	ó		L		ç	Ä	B	¦	Þ	º	7	Œ		Ï	~	Á	V		)	ß	-		n		511	
``````````````````````

Power cycle... then.... 

```````````````````
HTTP/1.1 200 OK Content-Type: HTTP/1.1 200 OK Content-Type: HTTP/1.1 200 OK Content-Type:
•	I	O		!	û	Å	U	u	t	Õ		Œ	¹	™	Ü	Ú		¡	H		S	®	¦		B	Ý	ú		3	²	£	31	
	ú	d		1	°	¤	ƒ	Ñ	7	É	€			Î	á	‰	N		ï	×	à	Þ	£	K	Š	U	$	Õ		ø	s	63	
Ÿ	K	u	Ì	ñ	#	g		ß	¢	#	W		M	ç	è	ë	Î	…		ò	b	¥		_	M		i	:	U		×	95	
	[	!	n	[	ž	n	_	—	¾		Û	Î	—	Õ	º	B	û	(		ú	P	‰	ã	q	L	©			L			127	
Ý	©			ÿ	ƒ	´	y	²	¾		à	N	b	D	\	×	f	&	Ó		–	ß	¥	ù		˜	€		0	O	j	159	
—	J	[	}	Ê		ã		s	ð	ó	‰	F	H	~	þ	=	×		Ð	ˆ		Ý	†	m	?	µ	Ü	Y	é	Ù	‚	191	
Ù	:	´	b	Ã		I		¸	Û	”	?	€	º	ß	i	³	¢		Ì	¦	^	¥	æ	Ø	´	‚	£	Ç	î	5	à	223	
·	M	4	¼	‘		ú	}	–	{		Î	ý	”	ý	ù	ë	f		=		P	¦		_	Þ		~	ä	å	®	¶	255	
ß	&	|	,	š	:	B	´	‹	ó	Ú		B	c	j	^	%	«		Ö	í		s	ã	Õ			>	Î	n	ý		287	
×	ª	@			à	/	y	Z	Ó	P	E		±	v	.	0	}	E	þ	é	G	û	M		C	f	ô	v	V	÷		319	
	*	ê		”	e	é	/	&	“	³	&	ù	³	˜	n	ú	ø	X	J	@	¬	x	Í	î	¼	'	i	 	ô		¨	351	
	'	x		2			í	e	ú	<		ï		—	4	Ã	w	Q	l	l		G	i		q	X	‚	¯	µ	œ	Û	383	
€	_		æ	¡		5	Ù	c		6	Á	4	ƒ	o	º	M	w	X	`			T		¿	Å				ž	¨	&	415	
G			¥	.	S		c	ú		>	¦	S	÷	‹	¬	r	ü	>	x	C	©	Ö	®	ž	Ê	Ì	Û	·	÷	§	{	447	
b	ï	d	y	x	O	½	#	œ	ç	î	J	°	J	g	<	{	Å	[	¸	À	Ð	î	'	/		ƒ		Í	Œ	Å	›	479	
$	‚		R	ì	v	ú	}	ó		L		ç	Ä	B	¦	Þ	º	7	Œ		Ï	~	Á	V		)	ß	-		n		511	
`````````````````

I'm definitely calling commit.. this has been working flawlessly up until now.

igrr, 2015-05-21:
This was an easy one:
```#define CONFIG_START_SECTOR (((uint32_t)_SPIFFS_end - 0x40200000) / 4096)```
An address-of operator missing before ```_SPIFFS_end```.
[source](https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/libraries/EEPROM/EEPROM.cpp#L34)

holgerlembke, 2015-05-21:
Current EEPROM looks good. Write works, survives Reset and Powerdown, so at least on my side it seems a winner.

Thank you!

closed 2015-05-21 by holgerlembke

sticilface, 2015-05-21:
Mine is still not working!  

done git pull, compiled the latest version using ant dist.  Flashed to node-mcu... 
if i write to EEPROM it is there until i power down!

igrr, 2015-05-21:
Which memory config did you try?

sticilface, 2015-05-21:
the 512K, 64K SPIFFS....  however that is correct for my module.  when i get flash size it is 512K

sticilface, 2015-05-21:
If i load the EEPROM.clear sketch then the EEPROM.read it is still full of numbers, not 0s... so it is not my sketch doing it!

holgerlembke, 2015-05-21:
In my understanding, 512k should work on all platforms...

I did a Arduino 1.6.4-IDE download (arduino cc) and then via Config/JSON/Boardthingy. 

Worked.

sticilface, 2015-05-21:
now if i use the write sketch example, I changed it to write 41 to all address... it still fails.. on reboot i get random numbers.  

The curious thing is that when i dump my main sketch EEPROM it is full of local AP... this is happening because  a variable from my sketch is over writing the EEPROM area of memory, i.e. part of heap?  I mean at no point to i write the array containing the local wifi stations to EEPROM?

sticilface, 2015-05-21:
What is interesting is if i compile with a pre-SPIFFS version ( the one where the EEPROM works), 
even after uploading loads of sketches, the EEPROM wipe, EEPROM.clear.. etc etc... from the IDE where EEPROM doesn't work.... the EEPROM is still there from before and my sketch boots and works just fine.... (just helping the debug effort, or trying to)

igrr, 2015-05-21:
@sticilface just a word of warning — if you have the board manager package installed in your system, and then run the compiled version, the board manager package will take precedence over the version of the core bundled with the compiled version. So if you compile anything, be sure to delete the package installed via the board manager.

Older versions were 1.6.1 so they are not affected by the core installed via board manager.

sticilface, 2015-05-21:
Ah. I thought that might be the case. I saw some debug lines that suggested
that. In fact I deleted the package. But the location gave an area in my
lib! Will let you know.

Cheers

ikbelkirasan, 2016-01-07:
I had a problem saving to EEPROM, it was losing its content at every reboot and every byte of its storage was reset to 0xFF. I solved this problem by clearing its content before I upload the main sketch. Now it seems working fine. I don't know what was causing that behaviour.

asetyde, 2016-01-07:
Version ?

Mb used ? ota ? is a bitt generic

ikbelkirasan, 2016-01-07:
I'm using Generic ESP8266 ( ESP-01 ) which has 512KB of Flash. I'm flashing using Serial port.

ricardojlrufino, 2016-10-29:
I had same problem.

Using this bellow sketch, has reporting a wrong size:
https://github.com/esp8266/Arduino/tree/master/libraries/esp8266/examples/CheckFlashConfig

Then i cleaned the EEPROM and works
```arduino
    int SIZE = 512;
    EEPROM.begin(SIZE);
    for (int i = 0 ; i < SIZE; i++) {
      EEPROM.write(i, 0);
      Serial.print(0); delay(10);
    }
    EEPROM.commit();
    Serial.print("[done]");
```