esp8266/Arduino #280: Fatal exception (28)

jeffryr, 2015-05-19:
The onboarding application from https://github.com/jeffryr/ESP8266-Sketches/tree/master/Onboard compiled and worked with the git version from around May 6 running on an ESP-01.  Using this morning's (May 19) git version compiled on Linux it crashes after attempting to launch an AP.  With debugging enabled it logs the following:

Waiting for Wifi to connect...
00000000000000000000
Connect timed out, opening AP
scandone
scan done
3 networks found
1: Tomato (-43)*
2: FiOS (-61)*
3: wrt54gl (-71)*

mode : sta(18:fe:34:9e:3c:4b) + softAP(1a:fe:34:9e:3c:4b)
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
add if1
bcn 100

WiFi connected
Local IP: 0.0.0.0
SoftAP IP: 192.168.4.1
Server type 1 started
Mode: STA+AP
PHY mode: N
Channel: 6
AP id: 0
Status: 0
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
Fatal exception (28):
epc1=0x40214139, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000014, depc=0x000000del if0
bcn 0
del if1
usl
sul 0 0

anesthetized, 2015-05-19:
It's an issue i'm experiencing too. It happens randomly but it hangs always before establishing a wifi connection. Resetting it (with the same firmware!) always "solves" the problem.
I started having this problems today with the new version, updated after a week. So it must be a recently introduced regression!

igrr, 2015-05-19:
I think this is power supply related issue. I used to have it when I
powered my module from USB-UART adapter.

Try adding some capacitors on the supply line.

anesthetized, 2015-05-19:
@igrr yes it can be. I'm using the usb-uart supply but i never had a similar problem before. I have to try blocking the signal and see if i can cause this exception more frequently! I have a thing for breaking things.

jeffryr, 2015-05-19:
In looking at my code I'm thinking it might be an application bug.  Because of the issues with mDNS over softAP, the onboard app only calls mdns.begin() when it has successfully connected to an AP.  The main loop calls both mdns.update() as well as server.handleClient().  In configuration mode, this means mdns.update() is called without ever having called mdns.begin().  That was always incorrect behavior and with the "implement async replies in mDNS library" commit it is no longer necessary in any event.  When I get home I will comment out the call to mdns.update() and see if that solves the problem.

jeffryr, 2015-05-20:
You can go ahead and close this.  I removed the call to mdns.update() and the sketch works in both configuration mode as well as application mode. If you download the updated version and still have the same problem, there may very well be some power supply issues going on.  I have been using a generic MB102 to power my ESP8266 independent of the USB-UART (CP2102) and have had no issues with that configuration.  BTW the rational behind the onboarding sketch was that I helped a coworker design a OneWire monitoring system for a solar hot water system he put together about five years ago.  That was taken out by a lightning strike some time back and he is looking to replace it.  I wanted to send him a ESP8266-based example to try out that he could add to his wireless network without having to futz around with setting up a whole development environment just to configure the WiFi portion.

closed 2015-05-20 by igrr

koltegirish, 2015-09-28:
hello, Can anybody guide me how to resolve this error.
Fatal exception (28):
epc1=0x4000bf70, epc2=0x00000000, epc3=0x00000000, excvaddr=0x000000d4, depc=0x00000000

ets Jan 8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 30820, room 16
tail 4
chksum 0x15
load 0x3ffe8000, len 876, room 4
tail 8
chksum 0x21
load 0x3ffe8370, len 5172, room 0
tail 4
chksum 0xaa
csum 0xaa
rS
load ...

suxilong, 2015-11-14:
Fatal exception (28): 
epc1=0x40249694
epc2=0x00000000
epc3=0x4023450b
epcvaddr=0x00000000
depc=0x00000000
rtn_add=0x40234666
?
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 25992, room 16 
tail 8
chksum 0xee
load 0x3ffe8000, len 924, room 0 
tail 12
chksum 0xf2
ho 0 tail 12 room 4
load 0x3ffe83a0, len 2036, room 12 
tail 8
chksum 0xcb
csum 0xcb
don't use rtc mem data

renatotavares, 2015-12-14:
same here

Exception (28):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000083 depc=0x00000000

ctx: cont 
sp: 3ffef800 end: 3ffefa00 offset: 01a0

>>>stack>>>
3ffef9a0:  3ffe8498 3ffee9a4 0000037a 402027a2  
3ffef9b0:  00000083 3ffee9d0 0000037a 4020254c  
3ffef9c0:  00000000 00000000 3ffee9a4 402025fd  
3ffef9d0:  40202221 00000000 3ffee9c8 40201ab6  
3ffef9e0:  00000000 00000000 00000000 40202249  
3ffef9f0:  00000000 00000000 3ffee9e0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

schizobovine, 2016-01-17:
I got the same exception when attempting to overflow the WiFiClient::readStringUntil(char) routine as part of an Arduino sketch, so my guess would be some kind of memory allocation error. The code below didn't cause an exception on 8K of test data (but did drop the connection on the server side), while 16K caused the fatal error.

```arduino
void loop() {

  const uint16_t port = 80;
  const char *hostname = "www.example.com";

  Serial.print("Connecting to ");
  Serial.println(hostname);
  WiFiClient client;

  if (client.connect(hostname, port)) {

    // GREETINGS EARTHLING
    Serial.print("> HAI");
    client.println("HAI");

    // Wait for response with a 30s timeout
    bool failure = false;
    uint32_t timeout = 30 * 1000 + millis(); // 30 second timeout
    while (!client.available()) {
      if (timeout - millis() < 0) {
        Serial.println("Timeout waiting for client");
        failure = true;
      }
    }

    // If client responded in time, echo string back and tee to Serial
      while (!failure && client.available()) {
        String stuff = client.readStringUntil('\n'); // OVERFLOW HERE
        client.println(stuff);
        Serial.print("> ");
        Serial.println(stuff);
      }

    Serial.println("Closing...");
    client.stop();

  }

  delay(5000);

}
```
"Server" code:
```bash
echo "butts" | tr -d $'\n' | head -c 16K | nc -l -p 80
```

The stack didn't look to have any my testing data on it, so probably getting allocated on the heap somewhere (and no stack smash, drat!).

Further evidence would be that errno 28 (ENOSPC) is "No space left on device" (at least according to POSIX.1 spec for C library stuff). Odd that it isn't ENOMEM (12), but I've seen stranger things.

TL;DR look for memory allocation failures/buffers too big for RAM/files exceeding available flash.

frippe75, 2016-02-02:
Any ideas for this one?
V+ is 2cell 7.4v Lipo, fully charged (to my knowledge)...

```
Fatal exception (28):
epc1=0x4000bf0e, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (28):
epc1=0x4000bf0e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff45b0 end: 3fff47f0 offset: 01a0

>>>stack>>>
3fff4750:  3fff47a8 3fff3700 3fff4790 4021627c  
3fff4760:  3ffe9dad 3fff3700 3fff3148 4021636c  
3fff4770:  3ffe9dad 3fff3700 3fff3148 40209561  
3fff4780:  00000000 00000000 00000000 00000000  
3fff4790:  3fff75e8 0000001f 00000018 3fff75c8  
3fff47a0:  0000000f 0000000d 3fff7518 0000000f  
3fff47b0:  00000007 00000000 00000000 00000000  
3fff47c0:  00000000 00000000 00000000 3fff37c4  
3fff47d0:  3fffdc20 00000000 3fff37bd 402126fe  
3fff47e0:  00000000 00000000 3fff37d0 40100114  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

```

[image]