esp8266/Arduino #282: SD library spi pins declaration error [Arduino 1.6.4 + ESP8266 installed via Boards Manager]

reaper7, 2015-05-19:
when I try any example from SD library or SDWebServer for esp I got this errors:
```
In file included from C:\PROGRAMY\Arduino\libraries\SD\src/utility/Sd2Card.h:26:0,
                 from C:\PROGRAMY\Arduino\libraries\SD\src/utility/SdFat.h:29,
                 from C:\PROGRAMY\Arduino\libraries\SD\src/SD.h:20,
                 from SDWebServer.ino:35:
C:\PROGRAMY\Arduino\libraries\SD\src/utility/Sd2PinMap.h:371:2: error: #error Architecture or board not supported.
 #error Architecture or board not supported.
  ^
In file included from C:\PROGRAMY\Arduino\libraries\SD\src/utility/SdFat.h:29:0,
                 from C:\PROGRAMY\Arduino\libraries\SD\src/SD.h:20,
                 from SDWebServer.ino:35:
C:\PROGRAMY\Arduino\libraries\SD\src/utility/Sd2Card.h:65:37: error: 'SS_PIN' was not declared in this scope
 uint8_t const  SD_CHIP_SELECT_PIN = SS_PIN;
                                     ^
C:\PROGRAMY\Arduino\libraries\SD\src/utility/Sd2Card.h:68:31: error: 'MOSI_PIN' was not declared in this scope
 uint8_t const  SPI_MOSI_PIN = MOSI_PIN;
                               ^
C:\PROGRAMY\Arduino\libraries\SD\src/utility/Sd2Card.h:70:31: error: 'MISO_PIN' was not declared in this scope
 uint8_t const  SPI_MISO_PIN = MISO_PIN;
                               ^
C:\PROGRAMY\Arduino\libraries\SD\src/utility/Sd2Card.h:72:30: error: 'SCK_PIN' was not declared in this scope
 uint8_t const  SPI_SCK_PIN = SCK_PIN;
                              ^
Błąd kompilacji.
```

igrr, 2015-05-19:
Yeah... looks like the customized SD library was not bundled with our core, so it is not included in the board manager package.
Will fix.

igrr, 2015-05-22:
Updated package has been deployed.

closed 2015-05-23 by igrr