esp8266/Arduino #284: Bricked my ESP8266

santiagopoli, 2015-05-19:
Hi!

Yesterday I was experimenting with the ESP8266 and the Arduino IDE. I've made a program which connects to my Wifi and lets you turn a pin on and off. The problem is that I'm being unable to program my module again. I think the issue happened when I deleted the Serial.begin line in my code. The serial doesn't begin, so I'm unable to program it via FTDA.

Is there a way to reset the ESP to its factory settings or i'm doomed?

Thanks!

PS: I've tried to flash my firmware with both esptool and pyesptool and neither of them were able to connect to my ESP.

anesthetized, 2015-05-19:
What kind of error are you experiencing?
Did you try turning it on and off again?

santiagopoli, 2015-05-19:
I tried everything. With pyesptool I get a "Failed to connect" error and with esptool I get a sync error.
I was able to flash the module with the same wiring. The problem started when I deleted the "Serial.begin(9600)" line in my code (in setup());

santiagopoli, 2015-05-19:
I will try putting RST to ground. I didn't try that

Links2004, 2015-05-19:
try this setup:
https://github.com/esp8266/Arduino#minimal-hardware-setup-for-bootloading-only
then the esp never load you code and the boot-loading shut work.

igrr, 2015-05-19:
As I understand, it's not possible to brick an esp by uploading some
firmware. The bootloader starts without even talking to the SPI flash. If
you have trouble communicating with the bootloader, it means the
bootstrapping pins are not in the correct states. Check that you apply
correct pull up/downs to GPIO 0, 2, 15, CH_PD, RST.

igrr, 2015-05-19:
That said, it's entirely possible to brick an esp by applying voltages that
exceed its absolute maximum ratings.

santiagopoli, 2015-05-19:
Thanks,
Can someone clarify on this? 

if no RTS is used a manual power toggle is needed

holgerlembke, 2015-05-19:
I work with the following setup: I manually reset with first setting a jumper pulling GPIO0 down and then press and release a switch pulling RESET down. TXD/RXD and RXD/TXD connected, nothing more.

As far as I understand it, if you pull GPIO0 down your program never starts, ESP goes straight into program mode.

If you do a RESET you should at least see the boot gibberish.

duncan-a, 2015-05-19:
When are you removing the jumper from GPIO0?

holgerlembke, 2015-05-19:
Sometimes never... :-)

It works fine: Set jumper, upload. ESP reboots into my code after upload, make changes, press reset, upload. Repeat...

Normally I remove jumper after upload done. Due to my unsynchronized human hands,

duncan-a, 2015-05-19:
The procedure is:

   - Connect GPIO to GND (via a pulldown) - in your case using a jumper
   - Press RESET (or toggle power) Blue LED flashes
   - Upload the code to the ESP
   - Remove the jumper
   - Press Reset (or toggle power)

There *was* a problem where the Flash needed to be erased using PyTools
before uploading again - but I think that has been fixed...

holgerlembke, 2015-05-19:
For me my method works reliable with ESP-12 & 
[image]
ai-thinker.

kdschlosser, 2015-06-05:
use a better set of cables between whatever you are using for TTL and the ESP (and i hope it's not 5v) 3.3 volt only. i had the exact same problem. do not use a breadboard and as small or short a wire to connect from the ttl to the esp.2 wires you have to use for programming and you have to hold the GPIO0 (i think  it's zero been a minute since i flashed one) to ground, it has to stay there. so ground from TTL to ESP and to the GPIO. and then tap the RST to ground, CHPD has the be pulled high, 5K should be fine, and if all else fails, shoot me a message i have a fix, but will require software. i have all the stuff necessary. I am pretty sure its the connection from the TTL to the esp is not up to snuff. took me about 8 hours of messin around before i pulled my project apart and connected it directly. sometimes i also have found that you may have to tap the RST (definition of tap: half a second 2 times usually works) to ground more than once for it to take. couldn't tell ya why just is. seems to be a timing thing on how long it needs to be held at ground for. and also not sure as to why and not sure if this is the IDE but the auto reset work on my 2 dollar ESP01's things actually reboot into terminal mode without even having to disconnect the GPIO

closed 2015-09-30 by igrr

aaroniscrazy, 2019-07-29:
A little late but thank you. I managed to brick it by uploading some code that made some pins used for an input. By taking gpio0 to low I was able to reset it with esptool.py erase_flash