esp8266/Arduino #286: TLS support?

webworxshop, 2015-05-20:
Any possibility of adding TLS support to the Wifi library? It is my understanding that the ESP8288 SDK supports TLS connections with 1024 bit keys.

igrr, 2015-05-20:
#43 
That can be done, but I haven't started working on this yet.

marvinroger, 2015-05-20:
Great, this will be a good thing as a lot of APIs now requires https.

antelder, 2015-05-21:
So excited to read this! Would fab to have SSL support in WiFiClient.

AndyKirby, 2015-07-23:
+1

MQTT is not terribly secure without it. A sniffer lets you open it up
real easy. SSL/TLS make sit a lot harder to do.

tim-in-oakton, 2015-08-26:
+100

asetyde, 2015-08-26:
+100

marvinroger, 2015-08-27:
I don't want to be tough, but could you please stop publishing `+1` as this is not really constructive and this is a feature everyone wants, so that's probably in the roadmap. These +1s won't make any difference about when it's going to be ready. Don't forget that people subscribed to this issue receive a mail for each comment. Thanks :)

tim-in-oakton, 2015-08-29:
+1 marvin, great comment!  Feedback is clearly unnecessary!

marvinroger, 2015-08-29:
Oh come on, the goal was not to offend anyone! Feedbacks are great, `+1`s only are not. Let's not make a debate, GitHub is one of the rare platforms where there is no fuss. :)

igrr, 2015-08-31:
Closing this one, please follow #43 for updates.

closed 2015-08-31 by igrr