esp8266/Arduino #287: New SDK includes ip_addr.h which conflicts with ip_addr.h in ESP8266WiFi

Imroy, 2015-05-20:
The files are different but both try to define several important preprocessor defines, structs, etc. Compiling any of the ESP8266WiFi examples results in errors.

Probably the conflicting things should be removed from the ESP8266WiFi file?

igrr, 2015-05-20:
So why is this a problem? Are you getting some compilation errors? In that case, can you please post the code that triggers the error?

Imroy, 2015-05-20:
Here's compiling the WiFiClient example sketch:
````
/opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp -o /tmp/build2124191603910360001.tmp/ESP8266WiFi/WiFiClient.cpp.o
In file included from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/ip.h:39:0,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/tcp.h:42,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp:37:
/opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/ip_addr.h:96:0: warning: "IPADDR_NONE" redefined [enabled by default]
 #define IPADDR_NONE         ((u32_t)0xffffffffUL)
 ^
In file included from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/tools/sdk//include/user_interface.h:13:0,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/cores/esp8266/Esp.h:25,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:213,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:24,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp:33:
/opt/arduino-1.6.5/hardware/esp8266com/esp8266/tools/sdk//include/ip_addr.h:57:0: note: this is the location of the previous definition
 #define IPADDR_NONE         ((uint32)0xffffffffUL)
 ^
In file included from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/ip.h:39:0,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/tcp.h:42,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp:37:
/opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/ip_addr.h:100:0: warning: "IPADDR_ANY" redefined [enabled by default]
 #define IPADDR_ANY          ((u32_t)0x00000000UL)
 ^
In file included from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/tools/sdk//include/user_interface.h:13:0,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/cores/esp8266/Esp.h:25,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:213,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:24,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp:33:
/opt/arduino-1.6.5/hardware/esp8266com/esp8266/tools/sdk//include/ip_addr.h:59:0: note: this is the location of the previous definition
 #define IPADDR_ANY          ((uint32)0x00000000UL)
 ^
In file included from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/ip.h:39:0,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/tcp.h:42,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp:37:
/opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/ip_addr.h:147:0: warning: "IP4_ADDR" redefined [enabled by default]
 #define IP4_ADDR(ipaddr, a,b,c,d) \
 ^
In file included from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/tools/sdk//include/user_interface.h:13:0,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/cores/esp8266/Esp.h:25,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:213,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:24,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp:33:
/opt/arduino-1.6.5/hardware/esp8266com/esp8266/tools/sdk//include/ip_addr.h:39:0: note: this is the location of the previous definition
 #define IP4_ADDR(ipaddr, a,b,c,d) \
 ^
In file included from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/ip.h:39:0,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/tcp.h:42,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp:37:
/opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/lwip/ip_addr.h:220:0: warning: "ip4_addr1" redefined [enabled by default]
 #define ip4_addr1(ipaddr) (((u8_t*)(ipaddr))[0])
 ^
In file included from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/tools/sdk//include/user_interface.h:13:0,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/cores/esp8266/Esp.h:25,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h:213,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:24,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp:33:
[... cut ...]
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.h:24,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.h:32,
                 from /opt/arduino-1.6.5/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp:33:
/opt/arduino-1.6.5/hardware/esp8266com/esp8266/tools/sdk//include/ip_addr.h:12:8: error: previous definition of 'struct ip_info'
 struct ip_info {
        ^
Error compiling.
````

igrr, 2015-05-20:
Hmm, doesn't happen here:
```
Build options changed, rebuilding all
Using library ESP8266WiFi in folder: /Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi 

/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/variants/generic -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build6273361321883836173.tmp/WiFiClient.cpp -o /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build6273361321883836173.tmp/WiFiClient.cpp.o 
/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/variants/generic -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src /Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFi.cpp -o /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build6273361321883836173.tmp/ESP8266WiFi/ESP8266WiFi.cpp.o 
/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/variants/generic -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src /Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp -o /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build6273361321883836173.tmp/ESP8266WiFi/ESP8266WiFiMulti.cpp.o 
/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -DF_CPU=80000000L -DARDUINO=10605 -DARDUINO_ESP8266_ESP01 -DARDUINO_ARCH_ESP8266 -DESP8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/cores/esp8266 -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/variants/generic -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src /Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/WiFiClient.cpp -o /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build6273361321883836173.tmp/ESP8266WiFi/WiFiClient.cpp.o 
/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/Users/igrokhotkov/projects/esp8266/arduino/build/macosx/work/Arduino.app/Contents/Java/hardware/esp8266com/esp8266/tools/sdk//include -c -Os -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++ [... cut ...] iClient.cpp.elf -bo /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build6273361321883836173.tmp/WiFiClient.cpp_00000.bin -bm qio -bf 40 -bz 512K -bs .text -bs .data -bs .rodata -bc -ec -eo /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build6273361321883836173.tmp/WiFiClient.cpp.elf -es .irom0.text /var/folders/15/ybtbgzpj7636vv4wp92l2c700000gn/T/build6273361321883836173.tmp/WiFiClient.cpp_10000.bin -ec 

Sketch uses 238,508 bytes (45%) of program storage space. Maximum is 524,288 bytes.
```

jeffryr, 2015-05-20:
Doesn't happen here, either (building on Linux).

Imroy, 2015-05-20:
Ok, the issue seems to be with my installation of the new 1.6.5-based software over an old 1.6.1-based installation. I've done a new clean install and everything is compiling properly.

I've diff'd the known-bad install and this new good install and for the life of me can't tell what's wrong! None of the involved files are different, even the command lines are the same. Odd.

closed 2015-05-20 by Imroy

IMAN4K, 2017-12-06:
In case of anyone else failure, i resolve the problem by changing `lwIP ` Variant to `Espressif (xcc)` in `tools` menu.(`Arduino1.8.5` + version `2.4.0-rc2`)
```
#include <ESP8266WiFi.h>

#ifdef __cplusplus

extern "C" {
#include <mesh.h>
}

#endif // __cplusplus

void setup() {
	espconn_mesh_enable(NULL, MESH_LOCAL);
}

void loop() {

}
```

d-a-v, 2017-12-06:
Yes, espconn* is not yet forward-ported to lwip-v2.

IMAN4K, 2017-12-07:
I have changed 'lwIP Variant' from `v2` to `Espressif (xcc)`, but the same !
Is that suppose to make changes ?