esp8266/Arduino #290: Dallas DS18B20 Hangs ESP [Arduino 1.6.4 + ESP8266 installed via Boards Manager]

juanpintom, 2015-05-20:
I succesfully installed a Dallas DS18B20 on a ESP12 Using the 1.6.1 ESP-IDE (Obviously I added the OneWire and DallasTemperature libraries)

On the lastest Arduino 1.6.4 and installing the ESP via Boards I load the same sketch and the ESP hangs and reset everytime.

If I load the sketch with the previous IDE it loads and works OK.

Anyone has the same problem?

Links2004, 2015-05-20:
may linked to the problem #283 and #282. the OneWire libary is not part of the "hardware" is a generic library.  
you can try to use:
https://github.com/esp8266/Arduino/tree/esp8266/libraries/OneWire

juanpintom, 2015-05-20:
I'll test this night! but If I remember well I downloaded and copied the Onewire library from your link.

Regards!

juanpintom, 2015-05-20:
@Links2004 You're awesome! It's working, I copied the library from the github, but a 5 days ago version... xD Issue Solved just copy the library from 
https://github.com/esp8266/Arduino/tree/esp8266/libraries/OneWire

Thanks @Links2004

igrr, 2015-05-21:
Should be fixed now.

closed 2015-05-23 by igrr