esp8266/Arduino #292: NodeMCU v2 dev board not working

lordp1, 2015-05-20:
Hello esp8266 team,

I can upload firmware to the NodeMCU v2 board succesfully, but the board wont work after the upload of the  arduino sketch (even after reset or fully disconnecting & reconnecting the board).
I've tried a external 3,3 voltage source as well as connecting an FDTI to the serial ports, but noting works (my other esp-03 module does work).

I use the latest mac osx & windows 1.6.4 arduino IDE with the url in the board manager (have tried your url & adafruit's esp8266 url).

Here a simple sketch that i uploaded to the NodeMCU V2 board, as you can see the serial function doesnt work:
[image]

igrr, 2015-05-20:
Can't find any specifications for this board. What flash size and interface does it have?

lordp1, 2015-05-20:
i have orderdered the board from: https://www.antratek.nl/nodemcu-v2-lua-based-esp8266-development-kit they refer to NodeMCU for information, here is there github about the board: https://github.com/nodemcu/nodemcu-devkit-v1.0

robotiko, 2015-05-20:
nodemcu v2 is the confusing commercial name some sellers are using for the devkit 1.0.

Main changes:
"It is the 5th design of NodeMCU devkit. It uses CP2102 as UART bridge, and can flash firmware automatically by using nodemcu-flasher. It support apple's MAC OS. "

It seems to have different pinout and also included buttons have different wiring and purpose (user button is now reset). 
Also based on ESP12 with 4MB.  Announced as 32Mb (32Mbits == 4MBytes), pure marketing noise.

In case that it helps :D

jeffryr, 2015-05-20:
Does it work with Ivan's full IDE package (not the official Arduino IDE + board manager package)?  The DTR/RTS patch hasn't made it into the official Arduino codebase.  Without that patch I found that I could compile and upload sketches to an ESP-01 using a CP2102 USB-UART via the board manager technique but the serial monitor would hold the ESP8266 in a perpetual state of reset, see http://www.esp8266.com/viewtopic.php?f=26&t=3094 for some discussion.

lordp1, 2015-05-20:
where can i find the full ide package (compiled), link?? (could find it here anymore)..

igrr, 2015-05-21:
NodeMCU board shouldn't need the DTR/RTS patch because of the clever wiring of Reset and GPIO0 signals. When both DTR and RTS are asserted, reset and gpio0 remain high. At least this was the case for the original board (v0.9).

tprochazka, 2015-05-21:
So where is problem if it doesn't need the DTR/RTS patch?

lordp1, 2015-05-21:
uploading to the board is not the problem, and if I select NodeMCU as te device I don't have to press reset + flash button *(works automaticly, as mentioned above).
the problem is that after the succesful flasing of the board, the arduino sketch's doesn't work..

jeffryr, 2015-05-22:
Can you power that module independently from the USB connection?  If so, try loading a simple sketch that does nothing but joins your WiFi network and writes a bunch of stuff to the serial port or toggles a led. Disconnect the USB port and power it up.  If you can ping the module and/or and leds are blinking, it is probably some interaction between the Arduino IDE serial monitor and the module.  If not, something else is going on.  In that case, try loading different firmware like the Lua interpreter and see what happens.

lordp1, 2015-05-22:
I already have tried running the board of a 5v powersupply (it has an LDO converter to turn it in the 3,3 volt that the board uses), with the independent powersupply the arduino sketch still doens load and doesnt connect to my wifi (can't see a new device in device manager of my router).

when i reupload the NodeMCU firmware (lua) everything works normally (it start and puts a message via te serial connecton (cant test further because I dont know lua, only C(++)), this is while powered from th usb jack (so this is not the problem).

tprochazka, 2015-05-23:
According the 2A-ESP8266__IOT_SDK_User_Manual__EN_v1.0.1.pdf documentation, flash memory with 1024kB and more kB has different addressation, you can see it here

http://bbs.espressif.com/viewtopic.php?f=10&t=175

lordp1, 2015-05-23:
so the problem with the arduino IDE is that the boot file @ 0X00000000 doesn't load the rest (sketch I presume) that the compiler uploads to 0X00010000 ? could this not be fixed with a little change in the IDE (only upload location of the second file, or is a change in the compiler needed so that the boot-file adresses the correct memory-location (that also needs to change?))?

igrr, 2015-05-23:
Oh, so you have DIO flash? Then can you try modifying boards.txt, changing
```nodemcu.build.flash_mode=qio```
to
```nodemcu.build.flash_mode=dio```?
Don't forget to restart the IDE after this change.

andysmithfal, 2015-05-23:
Setting flash mode to DIO is all you need to do - here's a board definition you can add to boards.txt for the NodeMCU dev kit v2 
https://gist.github.com/andysmithfal/c8d08898ee6ef146355c

igrr, 2015-05-23:
@andysmithfal would you be comfortable sending this addition as a pull request? if not, would you let me add it to the Arduino package?

lordp1, 2015-05-23:
confirmed igrr 's methode, the board works now perfectly (thanks)!!.
I tried andysmithfal 's methode by editing the boards.txt in the esp8266 hardware folder (on osx with arduino 1.6.5 (latest hourly build, because the url for the normal osx version on the arduino.cc website give now the windows version (will likely be fixed soon)) but this methode gave me error code's will compiling (probably because the v2 folder isn't in the hardware folder).

andysmithfal, 2015-05-23:
@igrr submitted a pull request. Works for me on OS X.

Makuna, 2015-05-24:
At first it didn't seem to work for me (just modified the existing entry).  I restarted Arduino several times, even changed the line endings to be lf/cr that Windows Notepad loves, it just didn't work.  The last thing I did was switch away from the NodeMcu to generic and back, and it worked?  Maybe the IDE is caching?

igrr, 2015-05-24:
@andysmithfal thanks, merged and updated a bit.

dicka55, 2015-07-29:
I had the same problem addressed in this issue and this issue solved it. I am making this comment in the hope that it will help someone.

The post by igrr on May 23 says that, if you are running DIO flash, change boards.txt as follows:
    nodemcu.build.flash_mode=dio  (not "qio" as I had).

(I now notice  that Flasher was showing SPI = DIO rather than "QIO"in the Advanced menu.)

andysmithfal's post below igrr's post gives a link to an addition to board.txt which I tried first after restarting the Arduino IDE. This addition uses "dio" but did not work for me.

So I made igrr's change, and it worked!

(Note: I found board.txt quickly when I ran a file search. It was buried in User/AppData.)

Makuna, 2015-07-29:
If you grab the latest, I don't think you need to do any of this anymore.   There is a board entry specifically for this board and it jsu works. 
@lordp1 you should confirm and close this issue.

asetyde, 2015-07-29:
I confirm the last soft (Json ) work perfectly

lordp1, 2015-07-29:
the latest version seems to work for me..
so I close this issue.

closed 2015-07-29 by lordp1

mbajlo, 2015-11-10:
Hi, I'm using nodemcu dev board 0.9(ESP8266-12) and arduino ide 1.6.5, but every time when I try to program my board I get:
"warning: espcomm_sync failed
error: espcomm_open failed"

is there any solution for this? I tried holding flash button at the time of programming but didn't work either.

dicka55, 2015-11-10:
Check to see if you have the correct board called up from Boards Manager.Then try other upload speeds.Other than that, it may be a problem in the board.I'm no expert so that's all I can offer.

lordp1, 2015-11-10:
sometimes i have this problem with the current version of the IDE, I than have to reset en press the flash button once (not holding on to it) and than press the upload button in the IDE.
I find this stange because in earlier version of the ide with the mod (DTR/RTS patch) the automatic reset worked 100% off the time, but this manual reset works for me..

mbajlo, 2015-11-10:
Hi guys thank you for your responses, I've found out that I have set the esp module in programming mode by pushing and holding the flash button, and pushing the reset button once. That worked for me.
 Here is the link where it is described: 
http://www.esp8266.nu/index.php/Tutorial_Arduino_Firmware_Upload