esp8266/Arduino #293: Updates and the board manager...

sticilface, 2015-05-20:
How are updates handled using the board manager?  

ie... how do you get the latest commits etc....

tzapu, 2015-05-21:
I am also interested in this
Cheers

igrr, 2015-05-21:
When the update will be published, you will see "Install" button in the board manager instead of "uninstall". At the moment IDE doesn't notify you when an update is available.

tzapu, 2015-05-21:
hi, thanks for that. i assume this will happen only for releases, or whenever there is a new commit ?

igrr, 2015-05-21:
Only for releases. We break stuff too often to deploy new version after each commit :)

duncan-a, 2015-05-21:
Yeayyyy!!! This bodes very well................

Thanks guys

sticilface, 2015-05-21:
Worth adding here what you mentioned. 

Boards manager installs are independent of the ide! They will persist and override what is in the ide folder. 

For example.... To useatest commits with your own compiled ide uninstall the esp8266 in boards manager. 

I'm sure you could also overwrite the hardware folder in the boards manager location but that will vary by operating system and user! (I'm guessing)

closed 2015-05-21 by sticilface