esp8266/Arduino #295: Max value for analogWrite(pin, value) ?

redflag2k, 2015-05-21:
First of all, My environment is Arduino 1.6.4 with ESP8266 xxx.json (Latest release), ESP-12 version
and  I am trying to test analogWrite for PWM. 

I found my led strip is so dark at following code.

analogWrite(GPIO12, 255); 

So, I changed the value 255 to 999. then the led strip became more brighter.

My question is that...

What is the maximum value of PWM? (In arduino uno, 255 is maximun value, because arduino is 8bit MCU)

My guess is that... In ESP8266, the maximum value of analogWrite is 0xFFFF (=65,535), because ESP8266 has 32bit cpu core.

Is my guessing right?

igrr, 2015-05-21:
Actually it's defined as PWMRANGE:
```#define PWMRANGE 1023```

https://github.com/esp8266/Arduino/blob/HEAD/hardware/esp8266com/esp8266/cores/esp8266/Arduino.h#L47

redflag2k, 2015-05-21:
Thank you for your clear reply.

closed 2015-05-21 by redflag2k