esp8266/Arduino #299: SPIFFS docs folder missing?

holgerlembke, 2015-05-21:
I didn't find the SPIFFS docs folder as an subfolder of SPIFFS.

I think this folder should exist. Or at least a link where to find it.

https://github.com/pellepl/spiffs/tree/master/docs

ficeto, 2015-05-22:
those files are in the spiffs folder

holgerlembke, 2015-05-23:
Why change folder structure without need?

I think imports from other projects should be kept as original as possible.

ficeto, 2015-05-23:
are you really? so it's not ok the files to be there, they have to be in the same folder structure? Maybe leave the Makefile there too? Why? Did you forget about the tests? They are also not there...

holgerlembke, 2015-05-23:
yes.

Costs nothing and you can't be accused of "illegal modification".

ficeto, 2015-05-23:
:D what modification? spiffs files are there as they were (well small adjustments were required, but they do not really count as such) We are talking about 3 files here... (they were 4 but one was empty). I see nothing illegal in reorganizing the folder to my liking. As I said I also removed the Makefile and the tests as those will build with the core every time...
I do not think that he will mind that at all... his code is there, his name is there, docs and whatnot as well.
Same goes for that other dude... shame on him.
And btw I did post a request for some additions in SPIFFS as I saw some problems trying to get everything to work. Sad I will not be around if he makes it.
Bye now :) Me has nothing to do with SPIFFS no moar

closed 2015-07-28 by igrr