esp8266/Arduino #305: Doesn't upload

Pablotis, 2015-05-22:
Since the last version it doesn't upload sketches to my ESP8266:

warning: espcomm_sync failed
error: espcomm_open failed

I the old version and the same cables it works, it also works with the ESP8266 Flash Downloader Tool and an official firmware

igrr, 2015-05-22:
Which OS are you running?

Pablotis, 2015-05-22:
Windows, thanks for helping =)

Pablotis, 2015-05-22:
I added the log on the other comment

Pablotis, 2015-05-22:
How can I install it on ubuntu and try if it works? I have installed the IDE but there isn't any option for adding a source as in Windows

Pablotis, 2015-05-23:
The only difference I see from the other version that uploads is that now it uses 0x00000 and 0x10000 and before it used 0x00000 and 0x40000

igrr, 2015-05-23:
Looking at the log, there seems to be no reply from the ESP...
Do you reset the board manually?

Pablotis, 2015-05-23:
During the procces?

closed 2015-09-07 by Pablotis