esp8266/Arduino #307: TCP socket functionality

RussMathis, 2015-05-22:
Is there any support for TCP socket functionality? Where, or how to use. Examples?

ficeto, 2015-05-22:
Simple telnet to serial server
```c_cpp
#include <ESP8266WiFi.h>

#define MAX_SRV_CLIENTS 3
const char* ssid = "*************";
const char* password = "************";

WiFiServer server(21);
WiFiClient serverClients[MAX_SRV_CLIENTS];

void setup() {
  Serial1.begin(115200);
  //Serial1.setDebugOutput(true);
  WiFi.begin(ssid, password);
  Serial1.print("\nConnecting to "); Serial1.println(ssid);
  uint8_t i = 0;
  while (WiFi.status() != WL_CONNECTED && i++ < 20) delay(500);
  if(i == 21){
    Serial1.print("Could not connect to"); Serial1.println(ssid);
    while(1) delay(500);
  }
  Serial.begin(115200);
  server.begin();
  server.setNoDelay(true);
  Serial1.print("Ready! Use 'telnet ");
  Serial1.print(WiFi.localIP());
  Serial1.println(" 21' to connect");
}

void loop() {
  uint8_t i;
  if (server.hasClient()){
    for(i = 0; i < MAX_SRV_CLIENTS; i++){
      if (!serverClients[i] || !serverClients[i].connected()){
        if(serverClients[i]) serverClients[i].stop();
        serverClients[i] = server.available();
        continue;
      }
    }
    //no free spot
    WiFiClient serverClient = server.available();
    serverClient.stop();
  }
  for(i = 0; i < MAX_SRV_CLIENTS; i++){
    if (serverClients[i] && serverClients[i].connected()){
      if(serverClients[i].available()){
        while(serverClients[i].available()) Serial.write(serverClients[i].read());
        //you can reply to the client here
      }
    }
  }
  if(Serial.available()){
    size_t len = Serial.available();
    uint8_t sbuf[len];
    Serial.readBytes(sbuf, len);
    //bello is a broadcast to all clients
    for(i = 0; i < MAX_SRV_CLIENTS; i++){
      if (serverClients[i] && serverClients[i].connected()){
        serverClients[i].write(sbuf, len);
        delay(1);
      }
    }
  }
}
```

igrr, 2015-05-23:
WiFiClient class is basically your TCP socket. There's even a sample bundled which shows how to use it. Otherwise it's compatible with the WiFiClient class of standard WiFi Shield library.

RussMathis, 2015-05-23:
Wow @ficeto, thanks, looks great, I'll give it a try.
@igrr, I'm lost on the sample bundled? Where can I find it? Please...

igrr, 2015-05-23:
in the IDE look under Examples > ESP8266WiFi > WiFiClient.

RussMathis, 2015-05-23:
Having issue understanding @ficeto's tcp code above. Don't understand Serial1 usage?

What is it connected to? With my limited knowledge I think it is a wifi connection thru the ESP8266 but how is it actually/physically connected? My physical board is connected and powered via a 3.3v FTDI.

Confused a lot, about Serial1.
Any help resolving my understanding would be greatly appreciated.
Thanks!

ficeto, 2015-05-23:
you know there is a serial TX on pi 2
That is what Serial1 is and is used here to show the debug messages since Serial is used for the bridge
meaning there should be a device, computer, GPS ... something that communicates through serial
thus using Serial1 to show the status and any errors

RussMathis, 2015-05-23:
Thanks @ficeto, Serial1 connected to pin2 output only and used for debug messages.

Do you think this will work in AP mode? A single module would be the AP server and all others would connect to it? There is no WAN!

Thanks

ficeto, 2015-05-23:
it should not matter what mode you are in, you connect to the IP of the ESP on the port that runs the server (21 in this case), this is implemented as Serial to Telnet bridge, but if you remove the Serial code, you can use the socket to pass whatever you want.
pin 2 is Output Only Serial1

RussMathis, 2015-05-23:
Hmmm, so all modules would simply have a static IP and Net Mask within the same net. Then a message could be sent between devices(back and forth) simply by using their IP and port.

The goal will be to have them work in a campsite where the can all be placed 100' or so apart around the camp. Then have PIR sensors and if any sensor gets fired that module would send commands to all other sensors in the network to turn on their switch/LED for x time. I could even put one on my fishing pole that would fire off when a fish nibbles or the pole gets a strike.

Thanks @ficeto you've been a great help!

ficeto, 2015-05-23:
nice idea overall, but TCP socket is not the way to go, i think.
you should instead broadcast an UDP message that all nodes will be listening to and have them turn on, without having to call each node and turn it on

RussMathis, 2015-05-23:
Tried UDP @ficeto, but was losing packets. About every 1 out of 4 packets never got across. :(
So thought TCP/IP would work better?

Thanks

ficeto, 2015-05-23:
it is better, but it's also slower... having average 30ms round trip and 30 lights, it will take one second for the last to light up (nice effect btw:) )
you did not get them because UDP has no error correction (ACK/SEQ) so the other side can not know if you got it or not
Why not try to push 10 packets one after another? It will not require the round trip as you will not be awaiting reply and can ensure that the nodes will get the message

ficeto, 2015-05-23:
And since your mentioned PIRs, have you tried using one outdoors? they are indoor sensors usually

RussMathis, 2015-05-23:
So would you suggest a better sensor for motion? Have some 5 milli wat red laser and reflective tape maybe?

ficeto, 2015-05-23:
what do you want to guard against? People or animals? PIRs might work, I just have not seen them used outside for other than turn the light when you pass by. But those are relatively short range setups, and you talk outdoors. What would be the distance between the sensors? How will you power them? I'm a camper myself :)

RussMathis, 2015-05-23:
Hmmm, did not think about range, I get about 10' feet in the office. But the wind, as you suggest, might drastically affect them.

igrr, 2015-05-23:
I had this "lasers guarding perimeter" setup working at one point. Wished they were 5W instead of 5mW. And some sharks of course...

ficeto, 2015-05-23:
:D hahahaha

RussMathis, 2015-05-23:
Maybe some kind of paintball gun too. LOL

ficeto, 2015-05-23:
can I please throw in a quadcopter to carry the gun? Would be so much fun...

igrr, 2015-05-23:
How about an esp-driven quad project, huh?

ficeto, 2015-05-23:
i was too busy getting all of that ready, I have everything needed now but one thing... range :D ESPs are really low power, but we can start without WiFi and see from there.
I have RC output lib as well...

RussMathis, 2015-05-23:
I'm trying to put together a quick test of your TCP socket example but cant figure out how to create a quick client connect on the other module. Have server module running with flashing LED on pin 5 using your sample. What would be a quick example of getting a simple client to connect? Do you have time?

ficeto, 2015-05-23:
get any Client example and connect to it... not a big deal ;)

ficeto, 2015-05-23:
WiFiClient example will suffice

RussMathis, 2015-05-23:
OK thanks WiFiClient OK, Hey, how do I get the ESP to turn off the wifi so to save power? I'm always getting current draw of about 70mA.

ficeto, 2015-05-23:
and just to be clear, to do it TCP way, each client must be a server and then one to be the client that connects to all

RussMathis, 2015-05-23:
So I create both client and server instances on each module?

ficeto, 2015-05-23:
yes, that is why I said to use broadcast instead

RussMathis, 2015-05-23:
wow, it is raining crazy here. West Oklahoma.

ficeto, 2015-05-23:
so about noon?

RussMathis, 2015-05-23:
Yes, power keeps going out. LOL, Thankful for my UPS. LOL

ficeto, 2015-05-23:
have to say the biggest storms i've been in are in the states... stay safe

igrr, 2015-05-23:
Guys, let's move this chat to Gitter please and not explode people's
inboxes :) Thanks.

RussMathis, 2015-05-23:
Cool, found the newest examples. Have been using VisualMicro for Visual Studio (Need my auto complete LOL). And just now got the directories move to the correct folder for the "Arduino IDE" to find them. They were buried in the User roaming packages directory of the Arduino install folders.

RussMathis, 2015-05-23:
Not familiar with Giiter? Have link?

igrr, 2015-05-23:
https://gitter.im/esp8266/Arduino

closed 2015-05-24 by igrr

mariobello, 2016-05-28:
Sorry for my bad English.
I did some tests using sockets but also by setting the maximum value to 6 clients can not be more than five concurrent users .
I can have more the 5 Client ?.
Thank you .

ByTorrez, 2016-07-28:
Dear igrr thanks for be in this forum.
I need to improve an AP tcp ip server, but (not a web server as WiFiAccessPoint library). can you give me some guide?
Thanks a lot.

adoher2000, 2016-10-09:
Pleasure,

At the time I need to implement the next application.

With the PIC18F4550 I capture signals about us motion sensors, by B and D. These events ports, I need to send them to a socket that is listening on a Raspberry Pi operating system with pill.

For this purpose I have the need to communicate ESP8266 module and the PIC SPI with this module, The PIC sends information the ESP8266 and this should send data according to a Socket IP and port.

At the moment I am salivating as setting "Integrated TCP / IP protocol stack" in this module ESP8266 to receive data from PIC SPI and even send them this socket.

Email: adoher2000@yahoo.com

adoher2000, 2017-06-07:
Thank you.

When I complete the code Simple telnet to serial server, with Arduino 1.6.11, the following error

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

adoher2000, 2017-06-07:
Thank you.

I relly need the following, I need to make an ESP8266 code like this:

1. Socket Client: That when a data arrives to the Tx pointer, the data is sent to an IP, Port, data, the indicated IP has a listening socket and stores the data to a database.

2. Socket Server: In other case I need the ESP8266 module to be configured as a server, with its IP, port and receive a data and send it by terminal Rx

Relient the example code may be clear, but I have to interpret and understand.

I am very grateful, if based on my request there is an example that I can make my request.

adoher2000, 2017-06-10:
Estoy muy agradecido, con los integrantes del foro. ya solucione mi requerimiento anterior, con solo leer todos los comentarios y en especial:

igrr commented on May 23 2015
in the IDE look under Examples > ESP8266WiFi > WiFiClient.

Ye logre realiazar mis socket Servidor y Cliente.

fntheirs, 2017-06-29:
Hi @adoher2000 could you tell what is the frequency of your messages for  the 2nd case? I'd like to share my high precision GPS signal from my base that uses GPSD. So, if I read GPSD socket and send it to Serial port then I could share my GPS to a remote node. However, the frequency that I need is at least 10Hz (2 messages at that freq) and I don't know if the delay would be or not a problem there.
I'm from Argentina but I think that is better to write in English to share knowledge or experience with the rest of the world.

Thanks!

adoher2000, 2017-08-05:
Thank you.

At the moment I am developing a project, and it consists, that when I pass from low to high the RB0 input of a PIC16F877A microcontroller, it sends by RxD the following data "L01Z01C01P01B01" and the data is received by an industrial server and sends the data to a Socket server that is listening on the net and stores it in a data table in MySQL. This process is established with a serial cable and a UTP cable.

Now I need to change the serial cable and TCP module for an ESP8266 module.

For this purpose, take two examples and join them SerialEventro ClienteBasic, record and compile without novelty, power the ESP8266 module and respond ping, the module has updated ESP_8266_BIN0.92.bin Firmware.

The following steps are required:

1. When a data arrives by the terminal RxD of the module ESP8266-01, this data should be sent WiFi to a socket that is listening on a Linux computer.

With this code does not work, I do not know what settings to perform.

/*
  Serial Event example

 When new serial data arrives, this sketch adds it to a String.
 When a newline is received, the loop prints the string and
 clears it.

 A good test for this is to try it with a GPS receiver
 that sends out NMEA 0183 sentences.

 Created 9 May 2011
 by Tom Igoe

 This example code is in the public domain.

 http://www.arduino.cc/en/Tutorial/SerialEvent

 */

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

ESP8266WiFiMulti WiFiMulti;

String inputString = "";         // a string to hold incoming data
boolean stringComplete = false;  // whether the string is complete

void setup() {
  // initialize serial:
  Serial.begin(9600);
  delay(10);
  // reserve 15 bytes for the inputString:
  inputString.reserve(14);

  // We start by connecting to a WiFi network
  WiFiMulti.addAP("dlink", "");

  //Serial.println();
  Serial.print("Wait for WiFi... ");

  while(WiFiMulti.run() != WL_CONNECTED) {
        Serial.print(".");
        delay(500);
  }

  //Serial.println("");
  //Serial.println("WiFi connected");
  //Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  delay(500);
 
}
//Fin Setup

void loop() {
  // print the string when a newline arrives:
  if (stringComplete) {
     Serial.println(inputString);
     Serial.print("Recibio RxD\n");
     //Codigo copia ejemplo ClienteBasic
      const uint16_t port = 1961;
      const char * host = "192.168.0.100"; // ip or dns
     
      Serial.print("connecting to ");
      Serial.println(host);

      // Use WiFiClient class to create TCP connections
      WiFiClient client;

      if (!client.connect(host, port)) {
           Serial.println("connection failed");
           Serial.println("wait 5 sec...");
           delay(5000);
           return;
      }

      // This will send the request to the server
      // "Send this data to server"
      client.print(inputString);

      //read back one line from server
      String line = client.readStringUntil('\r');
      client.println(line);

      Serial.println("closing connection");
      client.stop();
   
      Serial.println("wait 5 sec...");
      delay(5000);
        
     //Fin codigo copia
     // clear the string:
     inputString = "";
     stringComplete = false;
  }
  //Serial.print("Loop\n");
}
//Fin Loop
/*
  SerialEvent occurs whenever a new data comes in the
 hardware serial RX.  This routine is run between each
 time loop() runs, so using delay inside loop can delay
 response.  Multiple bytes of data may be available.
 */
void serialEvent() {
  Serial.print("Entro serial evento");
  while (Serial.available()) {
    // get the new byte:
    Serial.print("Evento ");
    //char inChar = (char)Serial.read();
    char inChar = Serial.read();
    // add it to the inputString:
    inputString += inChar;
    // if the incoming character is a newline, set a flag
    // so the main loop can do something about it:
    if (inChar == '\n') {
      stringComplete = true;
    }
  }
}

adoher2000, 2017-08-06:
Thank you.

Find that the serialEvent () function did not work and the content passed to the loop.

The RxD pin of the ESP8266 module receives several bytes, concatenates them and sends them to an IP, port and data. The socket that is listening on a server, receives the data and stores it in a DBA.

It's already working, see code....

/*
  Serial Event example

 When new serial data arrives, this sketch adds it to a String.
 When a newline is received, the loop prints the string and
 clears it.

 A good test for this is to try it with a GPS receiver
 that sends out NMEA 0183 sentences.

 Created 9 May 2011
 by Tom Igoe

 This example code is in the public domain.

 http://www.arduino.cc/en/Tutorial/SerialEvent

 Documentcion

 https://playground.arduino.cc/ArduinoNotebookTraduccion/Serial

Dato de prueba
------------------------------------------------
L01Z01C01P01B01
L01Z01C01P01B11

 */

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

ESP8266WiFiMulti WiFiMulti;

String inputString = "";         // a string to hold incoming data
boolean stringComplete = false;  // whether the string is complete
boolean newData = false;

void setup() {
  // initialize serial:
  Serial.begin(9600);
  delay(10);
  // reserve 15 bytes for the inputString:
  inputString.reserve(15);

  // We start by connecting to a WiFi network
  WiFiMulti.addAP("dlink", "");

  //Serial.println();
  //Serial.print("Wait for WiFi... ");
  //Serial.print("1 IP : ");
  //Serial.println(WiFi.localIP());
 
  while(WiFiMulti.run() != WL_CONNECTED) {
        Serial.print("2 IP : ");
        Serial.println(WiFi.localIP());
        delay(500);
  }
 
  //Serial.println("");
  //Serial.println("WiFi connected");
  //Serial.println("IP address: ");
  //Serial.println(WiFi.localIP());

  delay(500);
 
}
//Fin Setup

void loop() {
  // print the string when a newline arrives:
  //Serial.print(" Antes While ");
  while (Serial.available() > 0 ) {
    //Serial.print(" Dato available() > 0 ");
     // get the new byte:
    //char inChar = (char)Serial.read();
    char inChar = Serial.read();
    // add it to the inputString:
    inputString += inChar;
    // if the incoming character is a newline, set a flag
    // so the main loop can do something about it:
    if (inChar == '\n') { stringComplete = true;Serial.print(" Evento "); }
  }
 
  if (stringComplete) {
     //Serial.println(inputString);
     //Serial.print("Recibio RxD\n");
     //Codigo copia ejemplo ClienteBasic
      const uint16_t port = 1961;
      const char * host = "192.168.0.100"; // ip or dns
     
      //Serial.print("connecting to ");
      //Serial.println(host);

      // Use WiFiClient class to create TCP connections
      WiFiClient client;

      if (!client.connect(host, port)) {
           Serial.println("connection failed");
           Serial.println("wait 3 sec...");
           delay(3000);
           return;
      }

      // This will send the request to the server
      // "Send this data to server"
      client.print(inputString);

      //read back one line from server
      String line = client.readStringUntil('\r');
      client.println(line);

      //Serial.println("closing connection");
      client.stop();
   
      //Serial.println("wait 5 sec...");
      //delay(5000);
        
     //Fin codigo copia
     // clear the string:
     inputString = "";
     stringComplete = false;
  }
  //Serial.print("Loop\n");
}
//Fin Loop

adoher2000, 2017-08-14:
Thank you,

Eight days ago, notify that the code was working and today is not entering a look.

I really do not understand what happens, I attach the code.

/*
Serial Event example

When new serial data arrives, this sketch adds it to a String.
When a newline is received, the loop prints the string and
clears it.

A good test for this is to try it with a GPS receiver
that sends out NMEA 0183 sentences.

Created 9 May 2011
by Tom Igoe

This example code is in the public domain.

http://www.arduino.cc/en/Tutorial/SerialEvent

Documentcion

https://playground.arduino.cc/ArduinoNotebookTraduccion/Serial
Dato de prueba

L01Z01C01P01B01
L01Z01C01P01B11

*/

#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

ESP8266WiFiMulti WiFiMulti;

String inputString = ""; // a string to hold incoming data
boolean stringComplete = false; // whether the string is complete
boolean newData = false;

void setup() {
// initialize serial:
Serial.begin(9600);
delay(10);
// reserve 15 bytes for the inputString:
inputString.reserve(15);

// We start by connecting to a WiFi network
WiFiMulti.addAP("dlink", "");

//Serial.println();
//Serial.print("Wait for WiFi... ");
//Serial.print("1 IP : ");
//Serial.println(WiFi.localIP());

while(WiFiMulti.run() != WL_CONNECTED) {
Serial.print("2 IP : ");
Serial.println(WiFi.localIP());
delay(500);
}

//Serial.println("");
//Serial.println("WiFi connected");
//Serial.println("IP address: ");
//Serial.println(WiFi.localIP());

delay(500);

}
//Fin Setup

void loop() {
  // print the string when a newline arrives:
  Serial.print(" Antes While ");
  while (Serial.available() > 0 ) {
        //Serial.print(" Dato available() > 0 ");
        // get the new byte:
        //char inChar = (char)Serial.read();
        char inChar = Serial.read();
        // add it to the inputString:
        inputString += inChar;
        // if the incoming character is a newline, set a flag
        // so the main loop can do something about it:
        if (inChar == '\n') { stringComplete = true;Serial.print(" Evento "); }
  }

  if (stringComplete) {
     //Serial.println(inputString);
     //Serial.print("Recibio RxD\n");
     //Codigo copia ejemplo ClienteBasic
     const uint16_t port = 1961;
     const char * host = "192.168.0.100"; // ip or dns

     //Serial.print("connecting to ");
     //Serial.println(host);

     // Use WiFiClient class to create TCP connections
     WiFiClient client;

     if (!client.connect(host, port)) {
        Serial.println("connection failed");
        Serial.println("wait 3 sec...");
        delay(3000);
        return;
     }
     
     // This will send the request to the server
     // "Send this data to server"
     client.print(inputString);

     //read back one line from server
     String line = client.readStringUntil('\r');
     client.println(line);

     //Serial.println("closing connection");
     client.stop();

     //Serial.println("wait 5 sec...");
     //delay(5000);
    
     //Fin codigo copia
     // clear the string:
     inputString = "";
     stringComplete = false;

  }
//Serial.print("Loop\n");
}
//Fin Loop

adoher2000, 2017-08-24:
Thank you very much,
Already solve the case, add the following parameters
#include <SoftwareSerial.h>
SoftwareSerial SerialESP8266(4,5); // RX, TX

adoher2000, 2017-08-24:
At the moment I am developing trying to send a data between a PIC16F877A an ESP8266 module

For pin 25 of PIC16F877A send "L01Z01C01P01B61" like this:

The pin 25 of the PIC, is connected pin 11 of the MAX232 and pin 14 to the pin 2 of an industrial server SENA LS100 and this sends the data to an IP and port, the socket server, reads the data and stores it in a table in MySQL , very good.

I perform the above procedure, but I change the industrial server SENA LS100 by the module ESP8266, the pin 4 of the RxD module to ping 14 of the MAX232 and the data does not reach the database ..

If I execute the procedure, by CMD "type dato1.txt 1> COM4:" connecting the serial port pin 2 with 14 MAX232 and pin 3 with 13 MAX232 and the module ESP8266 pin 4 RxD with the pin 12 MAX232 the data reaches the base of data.

But when I connect the pin 25 of the PIC16F877A to pin 4 RxD of the module the data does not reach the database.

Codigo en PIC16F877A
--------------------------------------------------
#include <16F877a.h>
#use delay(clock=4M)
#fuses XT,NOWDT,HS,NOPROTECT,NOLVP,NODEBUG,NOBROWNOUT,PUT
#use rs232(baud=9600,xmit=pin_c6,rcv=pin_c7,parity=N,bits=8)

#Byte PORTB = 0X06
#Byte PORTC = 0X07

#Byte TRISB = 0X86
#Byte TRISC = 0X87

#Bit  RB0 = PORTB.0

#Bit  RC5 = PORTC.5

int1 zb0=0;

void main(VOID) {
port_b_pullups(FALSE);
TRISB=0B11111111;

TRISC=0B00000000;
RC5=1;

CICLO:
IF(RB0==1){IF(zb0==0){zb0=1;printf("%s","L01Z01C01P01B01");}} delay_ms(5); IF(RB0==0){IF(zb0==1){zb0=0;printf("%s","L01Z01C01P01B02");}};
GOTO CICLO;
}
--------------------------------------------------

Codigo en modulo ESP8266-01
--------------------------------------------------
#include <SoftwareSerial.h>
SoftwareSerial SerialESP8266(4,5); // RX, TX
#include <ESP8266WiFi.h>
#include <ESP8266WiFiMulti.h>

ESP8266WiFiMulti WiFiMulti;

String inputString = ""; // a string to hold incoming data
boolean stringComplete = false; // whether the string is complete
boolean newData = false;

void setup() {
// initialize serial:
Serial.begin(9600);
delay(10);
// reserve 15 bytes for the inputString:
inputString.reserve(15);

// We start by connecting to a WiFi network
WiFiMulti.addAP("dlink", "");

//Serial.println();
//Serial.print("Wait for WiFi... ");
//Serial.print("1 IP : ");
//Serial.println(WiFi.localIP());

while(WiFiMulti.run() != WL_CONNECTED) {
Serial.print("2 IP : ");
Serial.println(WiFi.localIP());
delay(500);
}

//Serial.println("");
//Serial.println("WiFi connected");
//Serial.println("IP address: ");
//Serial.println(WiFi.localIP());

delay(500);

}
//Fin Setup

void loop() {
  // print the string when a newline arrives:
  //Serial.print(" Dios ");
  while (Serial.available() > 0 ) {
        //Serial.print(" Dato available() > 0 ");
        // get the new byte:
        //char inChar = (char)Serial.read();
        char inChar = Serial.read();
        // add it to the inputString:
        inputString += inChar;
        // if the incoming character is a newline, set a flag
        // so the main loop can do something about it:
        if (inChar == '\n') { stringComplete = true;Serial.print(" Evento "); }
  }

  if (stringComplete) {
     //Serial.println(inputString);
     //Serial.print("Recibio RxD\n");
     //Codigo copia ejemplo ClienteBasic
     const uint16_t port = 1961;
     const char * host = "192.168.0.100"; // ip or dns

     //Serial.print("connecting to ");
     //Serial.println(host);

     // Use WiFiClient class to create TCP connections
     WiFiClient client;

     if (!client.connect(host, port)) {
        Serial.println("connection failed");
        Serial.println("wait 3 sec...");
        delay(3000);
        return;
     }
     //http://forum.arduino.cc/index.php?topic=409456.0
     //Serial.print(buffer);

     // This will send the request to the server
     // "Send this data to server"
     client.print(inputString);

     //read back one line from server
     String line = client.readStringUntil('\r');
     client.println(line);

     //Serial.println("closing connection");
     client.stop();

     //Serial.println("wait 5 sec...");
     //delay(5000);
   
     //Fin codigo copia
     // clear the string:
     inputString = "";
     stringComplete = false;

  }
//Serial.print("Loop\n");
}
//Fin Loop
--------------------------------------------------

I am very grateful for the support you can give me.

adoher2000, 2017-09-05:
Great pleasure, I am very grateful for the attention,

The novelty that I presented myself, in my consultation in the forum, to search, I found a problem in the protoboard, and already working, The topic was as follows:

As the circuit has been implemented in a protoboard, connect in the same hole of the pin 25 of the PIC the connection wire of the pin 4 RxD of the module ESP8266. So I already managed to send the pic data to the ESP8266 module and it to the socket on a linux server.

The protoboard help us to test the circuits, but they give us problems

Thank you very much.

jeancode, 2018-01-20:
hola hijos de codigo ami me paso algo similar lo solucione con una chela en mano y un buen porro

la unica razon porque los servidordes publicos rechasan la conexion y cuelgan es esp8266 es porque el buffer manda informacion aunque este basio client.print();  = clien.wirte("")<- solucion es usar client.println("hola");-> client.write("hola + \n")

/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <ESP8266WiFi.h>

 WiFiClient client;

 
void setup() {
  Serial.begin(115200);
  delay(10);

  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  
  
  WiFi.mode(WIFI_STA);
  WiFi.begin("xxxxxxxxx",xxxxxx");
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

    
  if (!client.connect("189.195.56.220", 1337)) {
    Serial.println("connection failed");
    return;
  }
  
}

int value = 0;

void loop() {

  
  delay(1000);

  Serial.print("connecting to ");
  Serial.println("189.195.56.220");
  

  Serial.print("Requesting URL: ");

  client.println("Hols soy espa8266");//-<client.print("");xxxx

               
  unsigned long timeout = millis();
  while (client.available() == 0) {
    if (millis() - timeout > 5000) {
      Serial.println(">>> Client Timeout !");
      client.stop();
      return;
    }
  }
  

  while(client.available()){
    String line = client.readStringUntil('\r');
    Serial.print(line);
  }
  
  Serial.println();
  Serial.println("closing connection");
}
;