esp8266/Arduino #308: EEPROM more memory

RussMathis, 2015-05-22:
Is there a way to extended the EEPROM memory? Use more banks maybe? 
Or support for SPIFFS?

Thanks

holgerlembke, 2015-05-22:
My two cents: if SPIFFS is stable EEPROM becomes mostly useless. 

Too much hassle to write data into it. Just create a struct and write/read it into/from a file.

igrr, 2015-05-23:
Yes it's possible to extend. We're a bit short on free blocks for 512k variant but for larger flash sizes it can be done. @RussMathis  How many blocks do you think are required?

RussMathis, 2015-05-23:
How about 64k?

RussMathis, 2015-05-23:
OK @holgerlembke, I'll bite. I'm a newbie... how do I get SPIFFS to work in the environment and how do I implement it? Seems like a cool feature but my abilities might not be up to snuff?

Thanks!

igrr, 2015-05-23:
@RussMathis just wait a few days, and you'll be able to just fopen your files and fread them. If you are using the webserver library, it will be just like one line of code in your sketch to serve static files.

holgerlembke, 2015-05-23:
@RussMathis There was an example... then something happend... :-)

I only do FS.mount(); and it is there.... To write I use FSFile file = FS.open(filename, FSFILE_OVERWRITE); file.write(buffer, gelesen); and FS.close();

But I really think there needs to be more documentations and some clear examples. Just the basics in small simple snippets: setup the filesystem, create, write, read, delete...

RussMathis, 2015-05-23:
@holgerlembke / @igrr Thanks, What will be the max file size?

ficeto, 2015-05-23:
@holgerlembke all of the FS. stuff is gone

igrr, 2015-05-23:
No include required, but don't use FS.* stuff — it won't be there in the next release. Let's say we are having a transition in API...

ficeto, 2015-05-23:
@igrr it's sad to resort to the core api (fopen, fwrite) just because there is only one way that it can be done the "Arduino" way, which is not "fopen"

igrr, 2015-05-23:
You're right. Better have both. Libs can use fopen and friends, while users
will have FS stuff.

holgerlembke, 2015-05-23:
Not on my disk... :-)

But I'll move to whatever comes after that. Currently I only explore the feature and do my top level routines...

ficeto, 2015-05-23:
You can open the FileSystem cpp and h files and see what is going on, it's just forwarding calls to the core spiffs api which is still there (missing the flash driver, but you have that as well)
It does not matter what the new api would be, it will rely on those same methods, so you can employ them and have your code working

closed 2015-09-30 by igrr